BillyBlaze/OctoPrint-TouchUI #2: [PR] Checklist v0.1.0

BillyBlaze, 2015-09-27:
**Features**
- [X] Printer tab
- [X] Temperature tab
- [X] Controls tab
- [ ] Webcam tab [testing needed]
- [ ] Terminal tab
  - [ ] Highlighter
  - [X] ASCII header
  - [X] Version check
  - [X] Scroll touch (if scrollbar)
- [ ] Timelapse tab [testing needed]
- [X] GCode viewer
- [X] Modals
   - [X] Scroll touch on open
   - [X] Dropdown navigation
   - [X] Leftside orientated full-height modal
   - [X] Fullwidth views
- [X] Notifications
- [X] Dialog screens (i.e. reload needed, offline)
- [X] Write all JS to jQuery plugin
- [X] Compile LESS with Gulp and serve CSS for better performance
- [X] Register to OctoPrint version checking
- [X] Use the latest FontAwesome for more icons

**Known bugs**
- [ ] No-pointer class doesn't get remove from dropdown, causing no navigation possible

**Cross-browser**
- [ ] Test as many devices as possible
- [ ] Ask community for beta testing

**Register Plugin repository**
- [ ] Clone repo
- [ ] Add content
  - [ ] Screenshots from devices
- [ ] Request merge

Akex2, 2015-10-05:
The plugin won't install on my raspberry
[image]

[image]

Thx

BillyBlaze, 2015-10-05:
Awesome thanks, I just merged a merge-request, could you try it again?

Akex2, 2015-10-06:
Hy
In the temperature tab can you make just number keyboard, now there are azerty keyboard
Thx
And the temperature of raspberry and bed and extruder, can you make two line because i can click on terminal tab i think is that
Thanks billy work fine ;)

closed 2015-10-10 by BillyBlaze

satoer, 2015-10-17:
Hi Billy,

Thanks for sharing this plugin. I’m looking if I can use it to make a printer completely standalone with a nifty LCD. But I think the plugin has some bugs:
There’s no way I can go to the settings menu. On every client I try the configuration icon is not where it supposed to be (according to screenshots). If I press the upper right corner only the shutdown and restart options appear.
[image]

This is with the latest versions of Firefox, IE, Chrome, Iceweasel and safari on the ipad (this is an ipad screenshot)

And bug number 2: On the (non TouchUI) configuration page the “Open the user settings” button does nothing. The only way to activate TouchUI is: octopi.local/#touch And for now the only way to disable it is to remove cookies.

Regards, Michel