BillyBlaze/OctoPrint-TouchUI #4: No configuration menu button and dead "open user settings" button

satoer, 2015-10-17:
Hi Billy,

Thanks for sharing this plugin. I’m looking if I can use it to make a printer completely standalone with a nifty LCD. But I think the plugin has some bugs:
There’s no way I can go to the settings menu. On every client I try the configuration icon is not where it supposed to be (according to screenshots). If I press the upper right corner only the shutdown and restart options appear.
[image]
This is with the latest versions of Firefox, IE, Chrome, Iceweasel and safari on the ipad (this is an ipad screenshot)

And bug number 2: On the (non TouchUI) configuration page the “Open the user settings” button does nothing. The only way to activate TouchUI is: octopi.local/#touch And for now the only way to disable it is to remove cookies.

Regards, Michel

BillyBlaze, 2015-10-17:
Hi Michel,

Thanks for reporting the bug, I suspect both bugs have to do with a javascript not loading correctly. Could you perhaps open this plugin in Firefox or Chrome with the developers tool open, and share any errors generating in the console?

satoer, 2015-10-17:
This came up:
```
Reconnect trial #0, waiting 0s packed_app.js:11227:13
Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience. For more help http://xhr.spec.whatwg.org/ less.min.js:8:0
Password fields present on an insecure (http://) page. This is a security risk that allows user login credentials to be stolen.[Learn More] <unknown>
Starting dependency resolution... packed_app.js:12070:9
... dependency resolution done packed_app.js:12127:9
Use of getPreventDefault() is deprecated.  Use defaultPrevented instead. packed_libs.js:4:0
Going to bind 22 view models... packed_app.js:12346:13
... binding done packed_app.js:12414:13
Firefox can't establish a connection to the server at ws://octopi.local/sockjs/420/w6bkp7qq/websocket. packed_libs.js:10176:0
The connection to ws://octopi.local/sockjs/420/w6bkp7qq/websocket was interrupted while the page was loading.
```

BillyBlaze, 2015-10-17:
hmm that looks pretty normal, not what I was hoping for. Did you first open the developers tool and then the plugin? so you might not have missed any errors?

Do you also have any plugins installed beside the bundled ones?

satoer, 2015-10-18:
started with the non touch UI octoprint:
```
The connection to ws://octopi.local/sockjs/757/mddydb2s/websocket was interrupted while the page was loading. packed_libs.js:10176:0
Reconnect trial #0, waiting 0s packed_app.js:11227:13
Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience. For more help http://xhr.spec.whatwg.org/ less.min.js:8:0
Starting dependency resolution... packed_app.js:12070:9
... dependency resolution done packed_app.js:12127:9
Password fields present on an insecure (http://) page. This is a security risk that allows user login credentials to be stolen.[Learn More] <unknown>
Use of getPreventDefault() is deprecated.  Use defaultPrevented instead. packed_libs.js:4:0
Going to bind 22 view models... packed_app.js:12346:13
... binding done packed_app.js:12414:13
```
went to: octopi.local/#touch:
```
The connection to ws://octopi.local/sockjs/900/goqqduhz/websocket was interrupted while the page was loading. packed_libs.js:10176:0
Reconnect trial #0, waiting 0s packed_app.js:11227:13
Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience. For more help http://xhr.spec.whatwg.org/ less.min.js:8:0
Starting dependency resolution... packed_app.js:12070:9
... dependency resolution done packed_app.js:12127:9
Use of getPreventDefault() is deprecated.  Use defaultPrevented instead. packed_libs.js:4:0
Going to bind 22 view models... packed_app.js:12346:13
... binding done
```

I havent installed octoprint plugins besides touchUI.

Steps I've done:

installed 015-07-02_2015-05-05-octopi-wheezy-0.12.0.zip	from: http://mirror.tsone.net.uk/octopi/ on SD card

in the startup menu:
expanded the filesystem
and boot to desktop

Did an update in the console:
sudo apt-get update
sudo apt-get upgrade

Had some problems getting a wifi adapter to work (RTL8191SU, Wi-Fi 300Mbps). The "Wifi config" utiliti failed to recognise the wifi dongle, and I eventually used the console to manually add the wifi using this tutorial:
http://raspberrypihq.com/how-to-add-wifi-to-the-raspberry-pi/

Installed touchUI using the octopi webinterface on a PC:
Settings => Plugin Manager => More => TouchUI =>install

Used your tutorial "Automaticly boot into fullscreen mode with iceweasel".

That's all the steps I did.
The problem does not seem to link to a specific browser. All browsers have the same problem (over network: IE, FireFox and Chrome. And local on the Pi: Iceweasel)

Shall I try to create an diskimage from the SD card and send it to you?

satoer, 2015-10-18:
You can download a complete diskimage here (1GB file):
http://we.tl/IGTT2m4W0b 
It's a RAR compressed 16GB diskimage. I don't know how to schrink the filesystem after expanding it.

satoer, 2015-10-21:
I have cornered the bug ready to smash it!
If you disable Access Control, then there are no User settings. So the button “Open User settings” can’t work. Also the topright menu in TouchUI is malfunctioning. If you enable user settings by changing the config.yaml:
```
cd  ~/.octoprint
sudo nano config.yaml
```
change:
```accessControl:
  enabled: true
...
server:
  firstRun: true
```
Than everything works fine and dandy.

BillyBlaze, 2015-10-21:
I was still trying to get your image on an 16gb sd but was failing miserly due to the larger size.

With this info I can easily reproduce the problem. I think I will make a dedicated modal and menu item for TouchUI so that your situation won't break the plugin :+1: I expect to release this, this weekend.

En uiteraard hartelijk bedankt voor alle info en voor het uitendelijk vinden van de oorzaak :smile:

satoer, 2015-10-23:
> En uiteraard hartelijk bedankt voor alle info en voor het uitendelijk vinden van de oorzaak

:flushed: WTF... haha, en ik maar mijn best doen het in begrijpelijk Engels te posten :wink:

BillyBlaze, 2015-10-28:
Hehehe :smile: op deze manier begrijpen iedereen die het zelfde probleem hebben wat de oorzaak is. Bedankt.

I just released v0.2.0 and have add'd your suggestion, thanks alot for your feedback, if you experience any problems just add it to this ticket so I will reopen it!

closed 2015-10-28 by BillyBlaze