BillyBlaze/OctoPrint-TouchUI #5: User Settings Unavailable

jneilliii, 2015-10-21:
I love the concept of a cleaner UI for mobile devices and was excited to see your plugin in the repository, but there is one caveat that I don't know if there is a way around or not.  I've decided to utilize haproxy's configuration for authentication utilizing certificates and username/passwords and as a result disabled OctoPrint's built in user authentication mechanism.  Because of that there is no way to modify your plugin to turn on the streamlined UI.  Is there any possibility of having it just automatically detect the device and/or browser and do the interface swap automatically vs having to turn it on?

BillyBlaze, 2015-10-21:
Thanks :smile: To be honest I never thought about turning off octoprint's authentication and assumed it was required. This is also why issue #4 has problems.

I think your suggestion is a great addition, I could check a few parameters; (i.e. resolution, touch device, iceweasel vendor string) and automatically turn it on. I think I will add one notification if the user wants to turn it off again. Also I might add a config where you can disable it, but by default it should be enabled :+1: 

Also in addition with the other issue, I think I also will make a dedicated modal for the TouchUI user settings.

BillyBlaze, 2015-10-28:
I just released v0.2.0 and have add'd your suggestion, thanks alot for your feedback, if you experience any problems just add it to this ticket so I will reopen it!

closed 2015-10-28 by BillyBlaze