BillyBlaze/OctoPrint-TouchUI #6: "toggle touchUI"-bug

derSkimmy, 2015-10-29:
Hey, since updating to 0.2.0 it's no longer possible for me to toggle manually between touchUI and the standard-Interface (Display is the Raspberry 7" LCD). Tried everything I know.

BillyBlaze, 2015-10-30:
Hmm, can't you see the ``TouchUI Settings`` in the dropdown when TouchUI is active? You should be able to toggle the interface there.

If you want to toggle within the standard interface then you can active the ``TouchUI settings`` menu items in ``Settings`` under ``TouchUI``

derSkimmy, 2015-10-30:
I can see the button as always. It is highlighted to state "on". When I press the button to toggle it off, the webpage reloads, but still in touch-mode.
In The addressbar is no "/#touchui", so I can't delete it this way. This happens only on the 7" LCD, all other (bigger) Monitors it works just as expected. My guess is, that it somehow has to do with the automatic switch to touchUI if the resolution is below a certain tresholdpoint. Is there a way to disable the automatic switch?

BillyBlaze, 2015-10-30:
Ahh that's not good, are you using Iceweavel? 

You can disable the autoload with the checkbox: `Start TouchUI automatically onload based on resolution or touch device.` in the `Settings` (remember to Save the settings they won't be stored instantly)

BillyBlaze, 2015-10-31:
I have identified the problem, this should be fixed in 0.2.1

Thanks for reporting :+1:

BillyBlaze, 2015-11-01:
This problem is resolved in v0.2.1

closed 2015-11-01 by BillyBlaze