BillyBlaze/OctoPrint-TouchUI #7: Camera shortly shows a picture and then stops

fraternl, 2015-10-29:
I just started using it on my Nexus 6 and most of it looks great.

Camera shortly shows a picture and then stops
If a go to another tab and return it doesn't even show that and it stays black.

lars-danielis, 2015-10-31:
Same here. I use Silk browser on a fire hdx.

BillyBlaze, 2015-10-31:
Thanks both for reporting, a fix has been pushed to develop. I'm planning to release it tomorrow after some more testing.

BillyBlaze, 2015-11-01:
This problem is resolved in v0.2.1

closed 2015-11-01 by BillyBlaze

lars-danielis, 2015-11-03:
Now the stream is there and it stays there.
The only thing what I am wondering is the temperature graph. It seems to have the perfect size to fit the screen, but it slips out to the sides of the screens by approximately the half of the screen. You can take and drag it to the middle. But it will not stay there it will slide to the other side and again the half is outside of the window. I did try it on Chrome on my laptop and on the fire HDX. Both the same.

BillyBlaze, 2015-12-19:
Sorry @lars-danielis, I forgot about your comment. I just finished some work in v0.3.0 where I fixed the table width. If your browser size is larger than 800pixels it will show the full table.

I will release this version soon after some more testing.