BillyBlaze/OctoPrint-TouchUI #8: Full screen mode is not looking as intended on my Nexus 6

fraternl, 2015-10-29:
I set my Nexus 6 in full-screen and I believe it isn't displaying as you intended...
Some stuff should be on 1 screen without the need for scrolling.
I made several screenshots to make this clear.

Great stuff.... 
I hope you can fix this.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

BillyBlaze, 2015-10-30:
Thanks for reporting, I'll look into this a bit more.

BillyBlaze, 2015-11-01:
This problem is resolved in v0.2.1

closed 2015-11-01 by BillyBlaze