BillyBlaze/OctoPrint-TouchUI #9: Removal of URL status messages

satoer, 2015-11-02:
Hi Paul

One step at the time I’m getting closer to fully implement a touch UI for my 3D printer. I have a small touch screen tft connected to my Rasberry Pi, and configured it’s auto starting with full screen Iceweasel and TouchUI. But there’s a small thing that’s bugging me. Every time I select a menu button, the virtual mouse pointer stay’s hovered above the button (because you press and remove your hand). Because of this, the URL status display’s constantly in the bottom left corner. This is breaking the illusion of a touch screen controlled printer a bit. 
[image]

I’ve been looking if it’s possible to remove tis status line in Iceweasel directly, but I’m afraid it’s not possible (don’t know for sure). What I did find was a javascript workaround for the buttons:
See:
http://jsfiddle.net/5KYLx/1/
Do you think this could be implemented?

Cheers Michel

BillyBlaze, 2015-11-02:
Hi Michel,

Thanks for reporting :+1: 

Your code is going to be hard to implement, some functionality of the core of OctoPrint relies on the href attribute. However there should be a workaround for this issue, I'll dive into this issue a bit more to get a better solution.

Cheers Paul

satoer, 2015-11-02:
Great!

Sidenote: I'ts not my code ;) . Found it over here:
http://stackoverflow.com/questions/4315029/how-can-i-hide-the-status-bar-in-the-browser
Credits to Mahn

BillyBlaze, 2016-01-19:
Sorry that it took a bit of time but I found a nice solution for this issues. I have added a new page to the Wiki which explains all the steps to hide the 'statusbar'. [See that here](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Tweaks:-Iceweasel).

Also if you would like to hide the cursor after a short period, then I'll recommend you follow all the steps  on [this Wiki page](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser). Regards if you already followed them, I just updated a startup script so that 'unclutter' will hide the cursor.

closed 2016-01-19 by BillyBlaze