BillyBlaze/OctoPrint-TouchUI #10: [Bug] The Menu link doesn't work in iPhone 6S+

IAmOrion, 2015-11-07:
The menu button doesn't appear correctly on my iPhone 6S+ and therefore I can not login etc.  Landscape or Portrait makes no difference.  

It appears to work correctly on my iPad Air 2 though so not entirely sure why it's not working as expected on the iPhone

BillyBlaze, 2015-11-08:
Thanks for reporting.

BillyBlaze, 2015-11-08:
hmm, it seems I cannot reproduce your problem, could provide me some more information about your os version and the plugins you have installed? (That white box might be part of the problem)

[image]

IAmOrion, 2015-11-08:
The iOS Simulator is not the best environment to debug browser based sites.  I'm on iOS 9.0.2

I have these plugins installed:

Automatic Shutdown (0.1.0)
CuraEngine
Discovery
DisplayProgress (0.1.0)
Email Notifier (0.1)
Navbar Temperature Plugin (0.7)
Plugin Manager
Print History Plugin (0.7)
Printer Stats (1.0.0)
ScreenSquish (0.1)
Software Update
Status Line (0.1.0)
STL Viewer (0.3.0)
TouchUI (0.2.1)
Virtual Printer

IAmOrion, 2015-11-09:
Update: At the time I replied, I was printing so could not disable plugins.  However, My print has just finished, I disabled "ScreenSquish" and it now works :)

So it would appear there is a conflict between the two plugins - however since TouchUI effectivley negates the need so ScreenSquish I've just removed it.

BillyBlaze, 2015-11-09:
Yeah I know the iOS simulator isn't the best tool but as I don't own any Apple products and since Apple made it mandatory to own an OSX device to remote inspect an iOS device it's the closest thing for me to check the compatibility. Unless you have any better suggestions? :+1: 

Good to hear the problem has been resolved. I will leave this ticket open with an `enhancement` tag. I think I will add a 'notice' if these two plugins are enabled.

IAmOrion, 2015-11-09:
Unfortuantely I don't.  You're right that apple make it difficult some times.  I wonder if there is a mode in Safari that enables "Mobile View" similar to the function in FireFox (I don't use Safari on my Mac)

Anyways... I imagine a simple banner notice if ScreenSquish is enabled would suffice :)

Glad we resolved it

BillyBlaze, 2015-12-24:
I have add'd a message in 0.3.0.

Again thanks for the issue report :)

closed 2015-12-24 by BillyBlaze