BillyBlaze/OctoPrint-TouchUI #16: No camera in V0.3.0

pcfriend, 2015-12-31:
Hi there,
I like your plugin very much and I used it on my Raspberry Pi B. But then I wanted to have more power for the grafical UI and I bought a Raspberry Pi 2B. After the installation of your plugin I couldn't see the image of the webcam on the Touch UI. It is still there on the normal Octoprint interface. What's wrong here. Do you have an idea?

Thank you! It's an amazing project! =D

Raphael

BillyBlaze, 2015-12-31:
Hi Raphael,

Thanks for the compliment and for reporting the bug, could you enlighten me a bit with the OctoPrint version you're running? And can you see any error's in the console of the developer toolbar?

Paul

pcfriend, 2015-12-31:
Hi Paul
I've running Octoprint 1.2.8. How do I enable the developer toolbar to watch the errors?

Happy new Year! ;D
Raphael

BillyBlaze, 2015-12-31:
Happy new year :smile: 

Checkout this link on how to open your developer toolbar. https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers

Then open the console and check if you can see any errors after opening your webcam tab.

pcfriend, 2016-01-01:
I can't see any errors. Maybe the warning is important, but I really don't know a lot about web applications.
[image]

BillyBlaze, 2016-01-01:
Awesome, that was exactly what I needed, sadly it doesn't give that much info about your problem, but it does give my information that all scripts are running perfectly fine.

I'll dive some more into this problem and try to fix it in 0.3.1

BillyBlaze, 2016-01-01:
Ok, it looks like I identified the problem. Apparently there was a refactor of code in OctoPrint 1.2.8 that I missed, I will make a fix for this in 0.3.1 and plan to release it soon.

BillyBlaze, 2016-01-03:
I just launched v0.3.1, feel free to open any new tickets if you have any issue's.

closed 2016-01-03 by BillyBlaze

pcfriend, 2016-01-09:
Thank you Paul. It work's!