BillyBlaze/OctoPrint-TouchUI #17: Not really an issue more a qustion

Bigfellas, 2016-01-01:
I am a bit unclear as to the use of this on a Raspi touch screen. 
I use this plugin for my mobile web browser and think it is amazing. I have a display on my Raspi that is running Octoprint. Can I run this on that same Raspi and use it as a to control interface. I tried installing as per you wiki. But I can't get past "Open Iceweavel" not sure how to run this or get any further. I amy be totally wrong and this is not the intention at all.
Thanks
Matt

Bigfellas, 2016-01-01:
Ok I think I get it now.. Didn't realise Ice weasel was a browser. it is pretty much un usable on a 2.8in display. Sad.

BillyBlaze, 2016-01-01:
Hi Matt,

Good you figured this part out (I updated the wiki on the part you got stuck on), could please explain or show me what part is unusable? If I'm correct your display will have a resolution of 320x240, it is a very small resolution but my plugin in fullscreen mode is workable on this resolution. Or do you have problems getting into fullscreen mode due to the small size?

If you are running Raspian Jessie, you could try installing the newest Chromium browser and edit the configuration to open it in "Kiosk" mode. I haven't tried this because my TouchScreen is a cheap TouchScreen that don't have updated driver for Jessie, so I couldn't test this method.

Paul

pcfriend, 2016-01-01:
Hi Matt
I found your post. I've tried today to use a Raspberry Pi 2B as a printer host and as a interface via Iceweasel, exactly like in the description of Paul. My problem is, that Iceweasel need so much cpu usage that the printer stopps during a 3D-print. Sometimes the hole Raspberry crashes! Maybe it works better with Midori. I'll try that soon.

Raphael

BillyBlaze, 2016-01-01:
Hmm that doesn't sound good at all, I haven't tried Midori but it does use the webkit engine, so that might be a very good alternative next to running Chromium 45. I will take some more time for this issue soon. It might be useful if we could disable the g-code viewer or webcam on a touchscreen during printing.

Bigfellas, 2016-01-02:
Hey Paul. The issue for me was there appears to be parts of the screen I couldn't get to. Like the login.

BillyBlaze, 2016-01-02:
Ouchy that sounds like a serious bug, could you perhaps give me some information about what OctoPrint you're running and with which browser you tried TouchUI with?

BillyBlaze, 2016-01-03:
Hi Matt,

It looks like I have identified the problem together with @satoer on IRC. I presume you're using any Apple product, if so, I will fix this bug in v0.3.1

BillyBlaze, 2016-01-03:
Ok, I released v0.3.1 and I'll hope this would work like it should be for you. Please feel free to open any new tickets if you stumble upon a bug.

closed 2016-01-03 by BillyBlaze

BillyBlaze, 2016-01-07:
@pcfriend I have also researched a bit more, but it really seems Iceweasel give the best performance. Sofar I have tried midori, ephinay-browser and chromium 22, but it wasn't a success, the overall interaction with the touchscreen is sofar supreme in Iceweasel.. (This test excluded Chromium 45, but I can't install that one...)

However I did came to the conclusion that we could ditch the "Boot to desktop" and move to a X.org window. Which might lower iceweasel's CPU usage.

If you find anything or want to try this new method then checkout ticket #18

Bigfellas, 2016-01-07:
Thanks Will give it a go when I have some time. Nice work

pcfriend, 2016-01-09:
Hi Paul
At the moment I have some problems with the power supply of the touch screen (Touchscreen draws to much current). So I think, that I won't use a touch screen on my 3D-Printer in the future. But anyway, thank you for your great work. Maybe I give a try to your new methode. Othervise I have to swich to a external tablet. That definitly will work! ;)

BillyBlaze, 2016-01-09:
I had similar problems and I had bought a 12v to 5v step down converter on ebay (couple of bucks) and used that to power an USB hub (e.g. dedicated power for webcam and the pi). I also blocked the USB power leading back from the Pi to Arduino. (e.g. USB cable from Pi to Arduino only let through the serial, and the power USB cable on the Pi blocked serial)

BillyBlaze, 2016-01-28:
Just a heads up, we have identified the problem with Iceweasel and there will be an update that addressing this issue in OctoPrint 1.3.0.

After upgradiong to 1.3.0, the CPU stays stable low (-40%) during printing. (currently 1.3.0 is available on the devel branch of OctoPrint)

Bigfellas, 2016-02-14:
Hey Billy. I decided to change over to a pi 2 today. I am not sure if it is the updated browser or the speed of the pi 2. But the touch UI is awesome now. So much more responsive. Gone from me not using it to. I think I want a bigger screen now. Amazing work Thanks.

BillyBlaze, 2016-02-14:
Thanks and glad to hear it works :smile: 

I think its a combination of 4 things: 1) your RPi2, 2) OcotPrint 1.2.9 update (which adressed some terminal issues with slow browsers), 3) TouchUI update with touchscreen optimization and 4) The kweb3 browser