BillyBlaze/OctoPrint-TouchUI #18: Touchscreen performance improvements

BillyBlaze, 2016-01-07:
Some experimental methods for trying to improve TouchUI on a touchscreen

### Method 1: Disable Desktop and boot to an X window instance
- __Disable "Boot to Desktop" (e.g. sudo raspi-config)__
- __Be up 2 date__
````
sudo apt-get update
sudo apt-get upgrade
````
- __Install X.org, iceweasel and stuff__
````
sudo apt-get install matchbox x11-xserver-utils unclutter iceweasel
````
- __Create /home/pi/start_iceweasel.xinit with this contents:__
````
# run this script with 'xinit ./start_iceweasel.sh
#!/bin/sh
unclutter &
matchbox-window-manager &
iceweasel --url "http://localhost:5000/"
````
- __Edit /etc/rc.local__  
 - Run ``sudo nano /etc/rc.local``  
 - Add ``sudo -H -u pi bash -c 'xinit ~/start_iceweasel.xinit'`` before ``exit 0``
 - Save

Reboot

BillyBlaze, 2016-01-25:
After some more investigation it looks like Iceweasel is unable to keep up with OctoPrint's Terminal and is therefor slowly killing CPU and Memory. The only currently solution is to disable the terminal.

This issue will be addressed in the upcoming release.

BillyBlaze, 2016-01-27:
This issue will be addressed within a new release of OctoPrint. If you would like to have a fix now then you can checkout the ``devel`` branch of OctoPrint and it will be released as 1.3.0

BillyBlaze, 2016-01-30:
I will reopen this issue for a while, so people can find it easier.

pcfriend, 2016-02-14:
Hallo Paul
I installed now Octoprint 1.2.9 and TouchUI 0.3.2. I also setup X.org and the Minimal Kiosk Browser but  I have bad news: The TouchUI works pretty well on touchscreen. Everything is ok, but when I start a print it immediately begins to lag. The printer can't finish the first layer, because the Raspi will crash before. In this state, I was unable to connect via ssh! Sometimes I was lucky and just the Kiosk Browser crashed. Then I could print without problems. Do you know how to deactivate X.org, so that I can activate it again when you changed something? Do you think it would help to reinstall the whole os?
Best Regards
Raphael

BillyBlaze, 2016-02-14:
:sob: I did some stress testing with kweb3 and printing, and for me kweb3 didn't take more then 40% CPU. Could you perhaps log into your SSH, run `top` and start printing to see what exactly is happening? (excessive CPU or memory?)

Could you perhaps also tell me your setup? Rpi1/2, OS? So I can do some stress testing on that.

For now you could turn off X.org by editing `/etc/default/touchui` and change `START=yes` into `START=no`. You can then manually turn on touch with `sudo service touchui start`.

BillyBlaze, 2016-02-14:
btw. would mind that the terminal is disabled during print?

BillyBlaze, 2016-02-16:
Do you have any feedback @pcfriend? I really would like to resolve this issue for you, would you mind installing the develop branch once I tried disabling features to see if it works for you? You could run a dryrun in order not to waste any filament.

pcfriend, 2016-02-16:
Hello Paul
Sorry for my late answer. I run the latest Octopi image on a raspberry pi 2b. I will test the "top" command soon. I also want to test your develop branch. How do I install this package? Maybe it will take a few days, maybe until this weekend. I'm working on my final exam. =D ... so that's why I'm a little bit busy..
Best Regards
Raphael

BillyBlaze, 2016-02-16:
Thanks for the feedback :) I'll try to write some codes this weekend, so take your time with your final exams!! all the best with that :D

I will post some information here for you on how to install it once I pushed some new codes to the develop branch. I think I will add a checkbox where you can enable/disable the terminal while printing. Heck I might even add an option to disable the temperature graph, so we could also see if that has any influence.

Thanks,
Paul

pcfriend, 2016-02-20:
Hello Paul
I have tried the top command now. It's not the CPU usage, it's the memory usage. The cpu usage stays under 40% but the memory usage rises higher and higher. At 70% in the gcode terminal appeared the message "Client to slow, log output disabled while printing..." Until know the Raspi didn't crash, but the memory usage still rises.
A Version in which i can disable unused fuction would be great. Not just for debugging, also for customizations =D.

Thanks for your engagement
Raphael

BillyBlaze, 2016-02-20:
Hi Raphael,

Thanks for the report! This gives me alot of help. Could you perhaps tell me if you have the G-code viewer enabled? And if you disable this, does the memory leak go away?

Thanks,
Paul

pcfriend, 2016-02-20:
Hi Paul
To disable the G-code viewer solved the problem. The memory usage is now at 17.2%!

Raphael

BillyBlaze, 2016-02-21:
Awesome :) as posted by @luokj in issue #30, if you would like to enable the G-code viewer then you can add the following lines in your ``config.yaml`` file:
````
gcodeViewer:
  mobileSizeThreshold: 1097152
  sizeThreshold: 1097152
````

I will add the ability to disable the G-code viewer while printing in the next release. If you would like to test that before the release, then ofcourse I will still post that info in here how todo that.

Thanks,
Paul

BillyBlaze, 2016-02-28:
Small update, I pushed some new codes to ``develop`` to fix this problem. If you want you could try it out but since we already figured out the problem, it is properly best to wait for the official release, since it can be unstable atm.

luokj, 2016-03-02:
I got it, wait release, thank you for hard work!-------- 原始邮件 --------主题：Re: [OctoPrint-TouchUI] Touchscreen: Browser crash and no response during print (#18)发件人：Paul de Vries 收件人：BillyBlaze/OctoPrint-TouchUI 抄送：luokj Small update, I pushed some new codes to develop to fix this problem. If you want you could try it out but since we already figured out the problem, it is properly best to wait for the official release, since it can be unstable atm.

BillyBlaze, 2016-04-23:
This problem should be totally resolved in version 0.3.3.

closed 2016-04-23 by BillyBlaze

luokj, 2016-04-23:
Great, thank you!

GeeF, 2017-01-18:
Still have this with TouchUI on mit piTFT. With or without gcode viewer