BillyBlaze/OctoPrint-TouchUI #19: bug : touchui not starting automatically on official raspberry pi 7" screen

korneelb, 2016-01-15:
Hi;

i follow the instructions on your site to the letter, but i get a 
FAILED : failed to start /etc/rc.local compatibility

i had to install xinit by hand with apt-get install xinit

when i ran the bash cmdlet
sudo -H -u pi bash -c 'xinit ~/start_iceweasel.xinit' 
it gave me an error that user pi was not authorized to run xinit
if fixed that with sudo dpkg-reconfigure x11-common.
now when i run the sudo cmdlet it will run the touchui. however, i still get the same error during boot and it does not automatically start into the touchui.

BillyBlaze, 2016-01-16:
Hi Korneel,

Thanks for reporting. I have updated the process by adding a service script and updated the Wiki, please take a look again @ https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser and continue from step 4.

I'll mark this issue now as closed, if you have any issues/problems then feel free to add comments. (I would still get notified)

BillyBlaze, 2016-01-16:
Wait... I did find a small issue in the new service script, please wait.

korneelb, 2016-01-16:
cool, will test later. this will not interfere with the autostart scripts from OctoPi right?

BillyBlaze, 2016-01-16:
No not at all, I created an independent startup script in the Debian LSBInitscript format, the previous method with the edit of ``rc.local`` was really a bad practice.

BillyBlaze, 2016-01-16:
(sorry for the open/close, accidentally pressed the wrong button)

I have updated the script init file so that the TouchUI will now start after OctoPrint has started.

closed 2016-01-16 by BillyBlaze