BillyBlaze/OctoPrint-TouchUI #21: Does not complete initialization if Software Update plugin not enabled

CapnBry, 2016-01-20:
I don't know how this project doesn't have more people cheering for it, it is 10x better than any of the paid Android apps for controlling Octoprint and it just uses web controls! I use a Nexus 7 2012 that docks to my printer to be an always available remote control that's so much better than using the dang character LCD and encoder interface that sucks so much ice. None of this sort of kudos should be in a problem report.

I OctoPrint enabled a second printer and couldn't access it with the TouchUI. I could get to the main page, but couldn't login using the TouchUI. On a desktop browser, the TouchUI settings were all blank and wouldn't update when saved.

Looks like the initialization code tries to register itself with the built-in OctoPrint Software Update model to be able to auto update. This printer had that plugin disabled because I do development on it and don't want anything to be messing with my source. When this call fails, it aborts initialization early it seems and leaves TouchUI in a weird state where the menus don't appear, and login is impossible. I didn't look at your source but if there's a way to be defensive when trying to bind to the Software Update plugin so TouchUI can carry on without it that would be great.

As a workaround I just enabled Software Update on that printer and reloaded the octoprint service / page reload and it worked like a champ.

Inside joke: I love cookies!

BillyBlaze, 2016-01-21:
Thanks for the compliment! that is always a good motive for keeping the project alive. I felt sometimes lonely because I have zero insight in how much this plugin is being used. (I can't even see how many times it have been downloaded), So even if it's a problem report, it's still my only insight that the plugin is actually downloaded and used... the irony..

Also thanks for the reporting the issue, I was able to easily reproduce this. I'm currently in doubt what the nicest solution is; I could easily disable the dependency call of the software plugin, but your problem did raise the question to me if I could make the dependency optional so that OctoPrint would still execute the rest of the plugin. I will need to discuss this with @foosel when I can get hold of her and if she is willing to hear me out about another dependency topic :smile: 

I :cookie: love!

BillyBlaze, 2016-02-20:
Just to give you an update on this bug. I have created a Pull-Request [#1232](https://github.com/foosel/OctoPrint/pull/1232) at OctoPrint, which will give me the ability that the rest of my plugin is executed without it stopping on an optional missing dependency. Once this is published, I do need to release another TouchUI update where these 'optional flags' is enabled.

BillyBlaze, 2016-06-18:
Ok, update, the Pull-request has been accepted, and is merged, therefor it will be fixed with TouchUI 0.3.3+ and OctoPrint 1.3.0+

I will close this issue now, if you have issues, then please let me know. Thanks for the report!

closed 2016-06-18 by BillyBlaze