BillyBlaze/OctoPrint-TouchUI #22: Boots to black screen

Bigfellas, 2016-01-21:
I thought I would try and use you new easy to follow wiki page. to get the pi display to boot into touch display. Unfortunately the display is booting to a blank screen with a cursor at the top.

I get the system dialogue while booting but then blank. I know the touch ui is working, it is fine on my iphone.  
Any ideas.
Thanks Matt

BillyBlaze, 2016-01-21:
Hi Matt, 

Thanks for reporting, do you have an error log located @ ``/home/pi/.xsession-errors`` you can share??

It seems like everything is working untill it tries to open Iceweasel, but maybe it fails to open unclutter. Hopefully the error log has some more info. If you do not have the ``.xsession-errors``, there is also an error log located @ ``/var/log/Xorg.0.log``

~Paul

Bigfellas, 2016-01-21:
Hey Paul sorry literally about to get into a plain. I will get back to you in a week

BillyBlaze, 2016-01-21:
Okay &  Fly safe

Bigfellas, 2016-01-27:
Here we go....

Xsession: X session started for pi at Thu Jan 21 17:44:43 AEDT 2016
localuser:pi being added to access control list
Obt-Message: Xinerama extension is not present on the server
Openbox-Message: Requested key "F13" does not exist on the display
Openbox-Message: Unable to find a valid menu file "/usr/share/lxde/openbox/menu.xml"
** Message: x-terminal-emulator has very limited support, consider choose another terminal

** (pcmanfm:2601): WARNING **: The directory '~/Templates' doesn't exist, ignoring it
** Message: x-terminal-emulator has very limited support, consider choose another terminal

** (lxpanel:2600): WARNING **: The directory '~/Templates' doesn't exist, ignoring it

(lxpanel:2600): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-can-change-accels after class was initialised

(lxpanel:2600): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popup-delay after class was initialised

(lxpanel:2600): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popdown-delay after class was initialised

(lxpanel:2600): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-images after class was initialised

(lxpanel:2600): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-label-select-on-focus after class was initialised

(lxpanel:2600): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-button-images after class was initialised

(lxpanel:2600): Gtk-CRITICAL **: IA__gtk_misc_set_alignment: assertion 'GTK_IS_MISC (misc)' failed

(lxpanel:2600): Gtk-CRITICAL **: IA__gtk_misc_set_alignment: assertion 'GTK_IS_MISC (misc)' failed

(pcmanfm:2601): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-can-change-accels after class was initialised

(pcmanfm:2601): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popup-delay after class was initialised

(pcmanfm:2601): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popdown-delay after class was initialised

(pcmanfm:2601): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-label-select-on-focus after class was initialised

(pcmanfm:2601): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-images after class was initialised

** (lxpanel:2600): WARNING **: launchbar: desktop entry does not exist

** (lxpanel:2600): WARNING **: launchtaskbar: can't init button

** (lxpanel:2600): WARNING **: launchbar: desktop entry does not exist

** (lxpanel:2600): WARNING **: launchtaskbar: can't init button

** Message: Connecting ...
** Message: Status changed to down

(lxpanel:2600): Gtk-CRITICAL **: IA__gtk_dialog_response: assertion 'GTK_IS_DIALOG (dialog)' failed
** Message: Status changed to opened
** Message: Connected to dhcpcd-6.7.1
** Message: Status changed to connected
** Message: eth0: CARRIER
** Message: eth0: Configured 192.168.1.250/24
ALSA lib control.c:951:(snd_ctl_open_noupdate) Invalid CTL Äbe¶:

** (lxpanel:2600): WARNING **: volumealsa: Default ALSA device not valid - resetting to internal
numid=3,iface=MIXER,name='PCM Playback Route'
  ; type=INTEGER,access=rw------,values=1,min=0,max=2,step=0
  : values=2

(lxpanel:2600): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-scrolled-window-placement after class was initialised
** Message: Status changed to down

(lxpanel:2600): Gtk-CRITICAL **: IA__gtk_dialog_response: assertion 'GTK_IS_DIALOG (dialog)' failed

** (lxpanel:2600): CRITICAL **: dhcpcd_open: No such file or directory
[... 182 lines cut ...]
(leafpad:2746): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popdown-delay after class was initialised

(leafpad:2746): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-images after class was initialised

(leafpad:2746): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-scrolled-window-placement after class was initialised

(lxpanel:2675): Gtk-CRITICAL **: IA__gtk_misc_set_alignment: assertion 'GTK_IS_MISC (misc)' failed

(leafpad:2746): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-button-images after class was initialised

(leafpad:2746): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-entry-select-on-focus after class was initialised

(leafpad:2746): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-entry-password-hint-timeout after class was initialised

(lxpanel:2675): Gtk-CRITICAL **: IA__gtk_misc_set_alignment: assertion 'GTK_IS_MISC (misc)' failed
** Message: Status changed to down

(lxpanel:2675): Gtk-CRITICAL **: IA__gtk_dialog_response: assertion 'GTK_IS_DIALOG (dialog)' failed

** (lxpanel:2675): CRITICAL **: dhcpcd_open: No such file or directory
XIO:  fatal IO error 4 (Interrupted system call) on X server ":1"
      after 33 requests (33 known processed) with 0 events remaining.
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":1"
      after 7907 requests (7907 known processed) with 0 events remaining.
pcmanfm: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.

BillyBlaze, 2016-01-27:
Ahh sorry, this is not the log I have been looking for...

Could you run this commands and copy the output?
1. ``sudo service touchui stop``
2 ``xinit /home/pi/OctoPrint-TouchUI-autostart/start_iceweasel.xinit``

Bigfellas, 2016-01-29:
Try this one

pi@octopi ~ $ sudo service touchui stop
[ ok ] Stopping TouchUI Daemon: TouchUI.
pi@octopi ~ $ xinit /home/pi/OctoPrint-TouchUI-autostart/start_iceweasel.xinit

_XSERVTransSocketOpenCOTSServer: Unable to open socket for inet6
_XSERVTransOpen: transport open failed for inet6/octopi:0
_XSERVTransMakeAllCOTSServerListeners: failed to open listener for inet6

X.Org X Server 1.12.4
Release Date: 2012-08-27
X Protocol Version 11, Revision 0
Build Operating System: Linux 3.2.0-2-mx5 armv7l Debian
Current Operating System: Linux octopi 4.1.6+ #2 PREEMPT Fri Sep 4 16:36:22 UTC                              2015 armv6l
Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbhei                             ght=416 bcm2708.boardrev=0x10 bcm2708.serial=0x181f2f08 smsc95xx.macaddr=B8:27:E                             B:1F:2F:08 bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_                             low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.                             mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 ro                             ot=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=                             font:VGA8x8
Build Date: 11 February 2015  09:31:17PM
xorg-server 2:1.12.4-6+deb7u6 (Julien Cristau <jcristau@debian.org>)
Current version of pixman: 0.33.1
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Fri Jan 29 20:08:08 2016
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
/home/pi/OctoPrint-TouchUI-autostart/start_iceweasel.xinit: 2: /home/pi/OctoPrin                             : not foundutostart/start_iceweasel.xinit:
/home/pi/OctoPrint-TouchUI-autostart/start_iceweasel.xinit: 3: /home/pi/OctoPrin                             : not foundutostart/start_iceweasel.xinit:
wm_set_cursor_visibility: visible using XCreateFontCursor

(process:2578): GLib-CRITICAL **: g_slice_set_config: assertion 'sys_page_size =                             = 0' failed

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-label-select-on-focus after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-can-change-accels after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-menu-popup-delay after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-menu-popdown-delay after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-button-images after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-menu-bar-popup-delay after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-entry-select-on-focus after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-entry-password-hint-timeout after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-images after class was initialised
console.error:
  [CustomizableUI]
  Custom widget with id loop-button does not return a valid node
console.error:
  [CustomizableUI]
  Custom widget with id loop-button does not return a valid node

pi@octopi ~ $ sudo service touchui stop
[ ok ] Stopping TouchUI Daemon: TouchUI.
pi@octopi ~ $ xinit /home/pi/OctoPrint-TouchUI-autostart/start_iceweasel.xinit

_XSERVTransSocketOpenCOTSServer: Unable to open socket for inet6
_XSERVTransOpen: transport open failed for inet6/octopi:0
_XSERVTransMakeAllCOTSServerListeners: failed to open listener for inet6

X.Org X Server 1.12.4
Release Date: 2012-08-27
X Protocol Version 11, Revision 0
Build Operating System: Linux 3.2.0-2-mx5 armv7l Debian
Current Operating System: Linux octopi 4.1.6+ #2 PREEMPT Fri Sep 4 16:36:22 UTC                              2015 armv6l
[... 16 lines cut ...]
(process:2578): GLib-CRITICAL **: g_slice_set_config: assertion 'sys_page_size =                             = 0' failed

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-label-select-on-focus after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-can-change-accels after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-menu-popup-delay after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-menu-popdown-delay after class was initialised

[... cut ...]
(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-entry-select-on-focus after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::                             gtk-entry-password-hint-timeout after class was initialised

(iceweasel:2578): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-images after class was initialised
console.error:
  [CustomizableUI]
  Custom widget with id loop-button does not return a valid node
console.error:
  [CustomizableUI]
  Custom widget with id loop-button does not return a valid node

BillyBlaze, 2016-01-30:
Thanks for the log, I have compared yours with mine and although it's not clean, they're exactly the same as mine that does start. I also looked with a co-worker at the logs and we couldn't spot anything ordinary. So we need to dig a bit deeper.

Can you share the logs ``/var/log/messages`` and ``/var/log/Xorg.0.log``, perhaps it's wise to put that in a pastebin.

Bigfellas, 2016-01-31:
Thanks for Looking.

http://pastebin.com/PYqLB8D4
http://pastebin.com/jYB7tKtw

BillyBlaze, 2016-01-31:
Ahh, maybe I found it, I think you're using a wrong driver in the X.org configuration. And therefor its failing to load X.org. If we run the calibration, then I hope this problem will be fixed.

You can run these steps to calibrate: https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/23#issuecomment-173679830

Bigfellas, 2016-02-01:
No good.
There wasn't a 99-calibration.conf is that normal. So I just added the text an created it.

BillyBlaze, 2016-02-01:
Yeah, that can be normal. Did the calibration work? (i.e. could you see the crosshairs?)

Bigfellas, 2016-02-01:
yeah I could.. But boots up the same now

BillyBlaze, 2016-02-02:
Ok :+1: I think we figured it out, big thanks to @foosel for pointing it out.

We think your Raspberry is attaching TouchUI to the HDMI display port. (You could verify that but if not just follow below steps)

1. Run ``sudo nano /etc/X11/xorg.conf.d/99-fbdev.conf``
1. Paste the following code:

    ```
    Section "Device"  
      Identifier "myfb"
      Driver "fbdev"
      Option "fbdev" "/dev/fb1"
    EndSection
    ```
1. Save & Exit (CTR-O, ENTER, CTR-X)
1. Run  ``sudo chmod +x /etc/X11/xorg.conf.d/99-fbdev.conf``
1. Reboot and hopefully it works

Bigfellas, 2016-02-03:
Getting closer yeah I am getting an error when it opens the browser and my touch is inverted may need to re calibrate. and I think I need to get rid of the menu bar somehow.
[image]
[image]

foosel, 2016-02-03:
The "unresponsive script" stuff can definitely be fixed by upping the value for `dom.max_script_run_time` in `about:config`. Alternatively, do the following in a shell

    echo 'user_pref("dom.max_script_run_time", 20);' >> ~pi/.mozilla/firefox/*.default/prefs.js

Might be that you have to stop the x server with the running iceweasel browser first.

The easiest way to get rid of the UI and put the browser in a real kiosk mode would probably be to install something like a FF Fullscreen or a similar fullscreen extension.

BillyBlaze, 2016-02-03:
Awesome command Gina :) Be sure you stopped iceweael browser first (run ``sudo service touchui stop``) else iceweasel will overwrite your addition.

Alternatively for installing the fullscreen extension goto TouchUI > Enable Fullscreen, you can also enable fullscreen mode of firefox in their menu. Once you're in fullscreen mode, iceweasel restores fullscreen mode after turning the Pi on and off again.

BillyBlaze, 2016-02-03:
As for the inverted axis, follow this:

1. Edit the calibration file you made ``sudo nano /etc/X11/xorg.conf.d/99-calibration.conf``
2. Add the following line before the ``EndSection`` that you copied from the calibration:

    ````
    Option "InvertX" "true"
    ````

Bigfellas, 2016-02-03:
Awesome thanks so much. I will give it a go tomorrow night.

Bigfellas, 2016-02-07:
Hey Working now. I am still getting the script error at the start. It is really sluggish and a little hard to use on such a small screen. I think it would be cool on the new pi screen.
Thanks For your help

BillyBlaze, 2016-02-07:
When you get the script error, check the box: "Don't show again" and click the most left button. This should execute all the javascript (now it's partially loaded). Also if you're in fullscreen mode it should be easier to work with the interface.

Also there is a fix in OctoPrint 1.2.9+ that will improve the terminal.

Also in my following release I will address some of the touchscreen 'lag' and try to make it better. (I am also awaiting a browser that supports Hardware Accelleration for better performance)

closed 2016-02-07 by BillyBlaze