BillyBlaze/OctoPrint-TouchUI #23: Touch Screen Calibration.

Stratrascal, 2016-01-21:
What a great job guys. I am 99% there. One last thing.  I can't figure out how to calibrate the screen.
It is very responsive but when I move left the cursor goes right.  I'm pretty much a noob so if this isn't the place for this forgive me and point me in the right direction.

Thanks
Bart

BillyBlaze, 2016-01-21:
Hi Bart,

Thanks for the compliment :) I didn't include a calibration yet because it would require some more testing on multiple devices, however you could try the following method, if it worked for you I will add it to the WIki.

Steps:
1. Stop TouchUI with ``sudo service touchui stop``
2. Run ``sudo apt-get install -y xinput-calibrator``
3. Run ``sudo rm /etc/X11/xorg.conf.d/99-calibration.conf``
4. Run ``FRAMEBUFFER=/dev/fb1 startx & DISPLAY=:0.0 xinput_calibrator``
5. Follow the calibration steps
6. After you completed step 6, pay attention to your command line, because it will show a wall of text with at the bottom instructions, please copy ``Section`` until ``Endsection`` and follow step 7
7. Run ``sudo nano /etc/X11/xorg.conf.d/99-calibration.conf`` and replace ``Section`` until ``Endsection``
8. Save (CTR-O, Enter, CTR-X)
8. Reboot

Stratrascal, 2016-01-21:
Wow, I don't get support like this on our million dollar cad system at
work. Speaking of work that is where I'm at right now. So I'll try it when
I get home. Since I have your ear right now I'll ask this. Will the
calibration fix the problem of moving left and the cursor moving right? And
lastly one of my displays has very dim backlighting. Is that adjusted with
software or is that a hardware problem. I really do want to thank you for
your efforts. It seems like a lot of work just to help us hardware guys
out.
 I work for Denso, my department designs instrument clusters and
infotainment centers for cars. Subsuquently we test bunches of displays.
I'm a pack rat and scrounge them up and I have a bunch of them most are
touch displays. If you would like I'll send you a few. Let me know
Bart

BillyBlaze, 2016-01-21:
Your welcome :) glad to help you out. Sounds like a very awesome job! To be honest I couldn't refuse your offer, sounds like some quality touch displays. However I am a bit worried about shipping costs :smile: it needs to be shipped to the Netherlands. Let me know if that's feasible.

Ontopic to your problems:

1. Fix moving left to right (inverting the x axis):
    After you completed step 7 in my first post (but before saving it in step 8) add the following code before the ``Endsection`` you just inserted:
    ````
    Option "InvertX" "true"
    ````

2. DIM lighting
I believe the back-light is either on or off. So any problems should be external or hardware related. It sounds like the LED requires more AMPS then the pi can deliver. Maybe you need to feed it with his own power? (Even more maybe; is the touchscreen 12v rated?)

Stratrascal, 2016-01-22:
Oh, I am definitely an American. We just assume everyone is here. Sorry.  I
will check into shipping costs.  Perhaps I can find something cheap and
slow.  On the displays, I have not figured them out yet. They have only two
cables on them. One for the video and the other is for the touch display
they are just flat cables with some circuitry on them. We drive them
directly from whatever product they happen to be on so the the video board
is not with them.  I will send you some pics. but we should jump out of
this thread.

As for your instructions I don't have a xorg.conf.d directory. I do have  a
xorg.conf.d.tar.gz file though.  Everything seems to be working so I am not
sure why that is not expanded. I am squeaky new with Linux so I could be
talking out of my @$$.  I don't mind digging around though I am learning a
lot by making mistakes.  But as  written your instructions give me this
error.
when I run  "sudo apt-get install -y xinput-calibrator"
E: Unable to locate package xinput-calibrator

 I am going to keep banging away at it

I think it will actually be calibrated ok I just need to change the
direction of the X axis.  On my other display Both of them are backwards.
So first I need to get it going the right direction then I will get the
calibration working.
I am pretty sure I need to edit 99-calibration.conf  but I cant find that
file outside of the tar.gz. which is
in /usr/share/man/man5/xorg.conf.d.5.gz Do I need to decompress it?

Thanks
Bart

Stratrascal, 2016-01-22:
> displays. However I am a bit worried about shipping costs [image:

Ok, I think I need to know which file my touchscreen is using for config. I
don't have 99-calibration.conf so I wonder which one is it using right
now.  If I can find that out I think I'll be there.  Bear in mind I have
only been using Linux for about a week now.  I used to be a DOS monster
back in the day and I am slowly getting up to speed but some things that
seem obvious to most of you guys is probably beyond my ken. But that is how
I have learned everything else.  Get myself in trouble and wiggle out.  I
have learned quite a bit that way.  So If anyone could let me know how one
would go about finding out What is loading and where it is looking for
config files. I am sorry for being so dense but there it is. I did get the
stuff installed but that was because your instructions were so great. I did
not know what I was doing just blindly running commands from the
instructions. Maybe as just a general question. How do I set up search
paths and how do I see what is in my search path at the moment.

Thanks
Bart

BillyBlaze, 2016-01-22:
Hi Bart,

Just to be safe, you're running Raspbian right? (if not do __not__ follow below instructions, else proceed ahead)

As for your first problem, the ``sudo apt-get install -y xinput-calibrator``, that was a bad from my part. This steps worked for me because I have add _Adafruits_ repository and xinput was bundled inside that repository :smile: 

Just run the following command
````
curl -sLS https://apt.adafruit.com/add | sudo bash
````

After it has installed try reinstalling xinput-calibrator. As for the file, run the calibration and check again if /etc/X11/x.conf.d/ has been generated. If not manually create the directory and file :+1:

Stratrascal, 2016-01-22:
I am actually running Wheezy

Stratrascal, 2016-01-23:
Success! I am starting to get a handle on it. Ok, the files in /etc/X11/xorg.conf.d all start with a number. and the system will execute those files in order? or just the lowest number?....  I really need to find a good tutorial that is not to basic and not to advanced.  Anyway I just found  the xinput-calibrator on "http://engineering-diy.blogspot.co.uk/2013/01/adding-7inch-display-with-touchscreen.html" and followed the instructions there. I did run the curl -sLS https://apt.adafruit.com/add | sudo bash command you suggested and it did load but after that no joy. But the calibration did set up the scroll directions properly. So if a dumb @$$ like me can get this going I guess it's not all that hard. The thing is I'm not that smart but I am tenacious. Thanks a lot.  Billy, Send me an Email outside of this thread and we will talk about the displays.

BillyBlaze, 2016-01-23:
Awesome :) You're right, the numbers indicates which file gets executed first (from low to high). I wanted to keep it simple (without all the build steps) so I tried the Adafruit method, but strange it didn't worked for you... (I also run Wheezy and it worked like a charm.. maybe you needed to run another command, I'll go find that out before I put it in a wiki)

Anyway I am going to close the issue now, feel free to comment if you have a question and we'll talk on the mail!

~Paul

closed 2016-01-23 by BillyBlaze

Stratrascal, 2016-01-24:
Great Job.  Not only did I solve my problem, I learned a lot.