BillyBlaze/OctoPrint-TouchUI #25: Prompts for switch to Full screen every boot even when already full screen

jamesarm97, 2016-02-05:
I am running on a RaspberryPi that auto starts Chromium with the --kiosk --incognito http://localhost:5000 and when Octprint starts and touchUI kicks in it always prompts if I want to go full screen although the browser already is. I see no setting that turns off this prompt and enabling doesn't do anything but pop up another annoying message about being switched to full screen and pressing esc to exit.

BillyBlaze, 2016-02-06:
Your issue is the incognito paramater, not TouchUI.

TouchUI stores your answer in a cookie (if you clicked No, it will not nag anymore) however the incognito parameter removes all cookies every new session, so TouchUI thinks each time you're running this for the first time.

If you clicked Yes, it will also ask it again on startup, since mobile devices will not return in fullscreen mode, so it asks again if the user want fullscreen mode. These are some workarounds since browsers has very strict security on fullscreen modes.

closed 2016-02-06 by BillyBlaze

jamesarm97, 2016-02-06:
Thanks. I'll try with just the kiosk mode when I get a chance.