BillyBlaze/OctoPrint-TouchUI #27: TouchUI Error

Zakattacks, 2016-02-07:
This error pops up the first time I open settings with touchUI enabled after loading the page. I have tested it on multiple android devices, and it even happens on laptops. It doesn't seem to do any harm though. I can post a screenshot if necessary. I'm not sure if it's my setup that's causing the issue or if it is one of my (many) plugins, or  code problem.
------- 
````
Unable to parse bindings.
Message: TypeError: $parent.layout is not a function;
Bindings value: contextMenu: { menuSelector: '#commandContextMenu', menuSelected: $root.controlContextMenu }, css: { 'sortable': $parent.layout() != 'horizontal_grid' }, attr: { 'id': id }
L/
````

BillyBlaze, 2016-02-09:
Thanks for reporting! Sadly this isn't a bug I can fix, as it is not TouchUI related, however it is related to the Custom Control plugin, so I have notified the plugin owner of your error.

Sorry for inconvenience I will remove these error notifications in the next release.

closed 2016-02-09 by BillyBlaze

Salandora, 2016-02-10:
Thanks @BillyBlaze for informing me.
@Zakattacks can you please open a Bug report here: [https://github.com/Salandora/octoprint-customControl](Octoprint-CustomControl) and include your config.yaml.
Thanks