BillyBlaze/OctoPrint-TouchUI #28: Almost finished install, but fell a little short.

waityn4jc, 2016-02-11:
First, let me say thank you for your really cool piece of software and the great walk-through on the wiki.  I am new to Raspberry Pis and also to Octoprint, but your guide was so user friendly, even I was able to get through it.

However, I need a little help crossing the finish line.  After first completing your instructions, my Pi booted to a black screen with a white line in the corner.  From reading your support of another person, I was able to get the Raspberry to stop attaching TouchUI to the HDMI.  (Thanks for that)

Now, as you can see from the attached image, I am booting into Iceweasel, but the touchscreen does not work.  If I attach a mouse I can move around, but the touchscreen is not working.  Can you please help me get that set up?

Again, I am super grateful for your awesome piece of software.  I am also new to Github, so if this is the wrong place to make this request, I am truly sorry.  I saw where you helped other users in this area and assumed this was the proper method.  (FYI, I have no Linux ability.  Thus far, I have followed your recipe on the Wiki.)
[image]

BillyBlaze, 2016-02-12:
Thanks and you're welcome :) I don't mind the help through github issue's atm, since I haven't set up any other communication. It also should help making the Wiki better. (I should really add this as troubleshooting)

Could you first try to [calibrate your touchscreen](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/23). I think you are missing some touchscreen configuration.

If that didn't work, then please attach the file `/var/logXorg.0.log`.

waityn4jc, 2016-02-12:
I am attaching two picture showing the error on the PiTFT and the SSH.

So, my Linux knowledge is basically 0, so I had no idea how to get the log file for you that you requested.  I was able to find a recommendation online to use cat /var/log/Xorg.0.log.old.  His file name was just a little different than the one you requested, so I am submitting them both.  I have also never used pastebin before, so I hope I did it correctly.  :-)

.old:
http://pastebin.com/aJRth3F8

Your request
http://pastebin.com/TdDK70J8

waityn4jc, 2016-02-12:
[image]
[image]

BillyBlaze, 2016-02-13:
Ok, don't follow those calibration steps :smile: 

Can you be sure you followed all the steps of Adafruit instructions page: https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/touchscreen-install-and-calibrate

waityn4jc, 2016-02-13:
I did the easy install image, so it said I did not have to do those steps.  (at the top of the page you linked).  However, I am happy to try it if you think that will help.  I was afraid I would break something by trying it since I already did the easy install.  :-)  That's problem of not knowing the OS you're working in, because I am worried to even tinker as I don't know what will break my install and cause me to start from square one.  Just following recipes here.  :-)

BillyBlaze, 2016-02-13:
Ok that makes alot of sense. The problem is not the OS, but all the different drivers and pre-defined configurations. Please follow the learn page of Adafruit because:

1. We need to be sure that the touchscreen event (the '/dev/input/event0') is bound to touch event. Maybe it is bound to '/dev/input/event1'. The first part of the learn page will check if you have the required files and parameters. The second part let's you test if your touchscreen is giving the OS your touch events.

2. We removed the old calibration file and we need to recreate the calibration file. Just follow after you done the 'evtest' the calibration methods of Adafruit.

waityn4jc, 2016-02-13:
Will do.  I have to go out of town today, so I might not be able to get to this before I leave, but I would like to.  :-)  If not, it will be Monday.  (USA East Coast here)

waityn4jc, 2016-02-15:
So, I started to do as requested and got to this line:
sudo evtest /dev/input/touchscreen

It said I should be able to test the touchscreen in real time.  However, nothing would happen when I would touch the screen.  I started to assume that something must not be right and started to examine the screen physically when I found this broken wire.  I assume that must be the culprit.  I will get this screen returned and a new one ordered.  Here's to hoping the next one arrives in better condition.  :-)

[image]

I will update you when the new screen arrives.

closed 2016-02-15 by BillyBlaze

waityn4jc, 2016-03-09:
> Ouchy, that explains a lot about it not working, it seems exactly like the touch interface connector   <https://assets-cdn.github.com/images/icons/emoji/unicode/1f62d.png>
>
> I will close this issue now, feel free to add any updates in here. I will still help you out, but I presume it will work after you got a new one   <https://assets-cdn.github.com/images/icons/emoji/unicode/1f44d.png>
>
> Just remember after you got your new touchscreen to run the calibration from the Adafruit page. We removed a important file, which the calibration generates in order to work properly. (I think you can easily start calibration with sudo adafruit-pitft-touch-cal)

Paul,

 

It’s me again.  I got overloaded with print jobs and didn’t have the time to work on troubleshooting my PiTFT.  I finally have some relief because I have ordered a 2nd printer and am setting up another instance of Octoprint.  I started a new Pi from scratch and everything appears to be working.  However, the webcam on the touchscreen just shows a blue and white question mark.  I installed snapstream per your troubleshooting, but can’t seem to follow what settings you want configured to make it work.  If I can’t get that working, it’s no big deal because I don’t have any need to view the webcam when standing in front of the machine, but was just trying to figure it out.  Any help would be appreciated.  Thanks.

 

Thank you,

Steven Seeberger

Chief Executive Officer

------------------------------------

Waityn4JC Mobilization, LLC

ToyBurg
1080 Cable Court 

Lawrenceburg, KY 40342

502-517-4649

www.toyburg.com <http://www.toyburg.com/> 

 

IMPORTANT NOTICE:
This message may contain confidential information. If you have received this e-mail in error, do not use, copy, or distribute it. Do not open any attachments. Delete it immediately from your system and notify the sender promptly by e-mail that you have done so. Thank you.

BillyBlaze, 2016-03-10:
Hi Steven,

Good to hear your printers are running overtime :smile:

Have you first verified that the webcam is working in the default interface of OctoPrint? If that is not the case then please be sure this is working properly before you continue on with SnapStream.

If all these settings are correct, then open the following url on your desktop browser with an url like: ``http://yourpi-ip:8080/?action=snapshot``

If that works in the browser then replace ``/webcam/?action=snapshot`` with ``http://yourpi-ip:8080/?action=snapshot`` in the SnapStream settings and save it.

Good luck,
Paul

waityn4jc, 2016-03-30:
Again, sorry for the extremely slow delay in responding.  I have to wait for lulls in orders to test things on my printers.  It’s a good problem to have.  :-)

 

So, the only place the webcam doesn’t work is on the PiTFT’s that I keep attached to the printers.  It works on a browser on a computer, it works on my phone.  It works in regular octoprint or in TouchUI on everything except the Pis with the TFT.

 

I don’t particularly need it there so if the fix is a big deal, then let’s not worry about it, but I thought you might want to know to troubleshoot possible bugs.

 

Thank you,

Steven Seeberger

Chief Executive Officer

------------------------------------

Waityn4JC Mobilization, LLC

ToyBurg
1080 Cable Court 

Lawrenceburg, KY 40342

502-517-4649

www.toyburg.com <http://www.toyburg.com/> 

 

IMPORTANT NOTICE:
This message may contain confidential information. If you have received this e-mail in error, do not use, copy, or distribute it. Do not open any attachments. Delete it immediately from your system and notify the sender promptly by e-mail that you have done so. Thank you.

BillyBlaze, 2016-03-30:
Yes this is a known bug in the browser, if you install SnapStream from OctoPrint's plugin repository, you can have a webcam feed on your PiTFT.

The technical part: The browser fails to load MJPEG and after it fails it doesn't do anything but displaying a broken image, if you install SnapStream, then this plugin will watch if MJPEG fails, if it does, it will load a static image every few seconds rather then streaming. This is also better for the CPU usage of the Pi.

waityn4jc, 2016-03-30:
Yea, I have the snapstream thing set up on both printers, but that didn’t resolve the issue.  If I didn’t take so long to respond it would be easier to remember these things.  :-)  Sorry again for that.

 

Thank you,

Steven Seeberger

Chief Executive Officer

------------------------------------

Waityn4JC Mobilization, LLC

ToyBurg
1080 Cable Court 

Lawrenceburg, KY 40342

502-517-4649

www.toyburg.com <http://www.toyburg.com/> 

 

IMPORTANT NOTICE:
This message may contain confidential information. If you have received this e-mail in error, do not use, copy, or distribute it. Do not open any attachments. Delete it immediately from your system and notify the sender promptly by e-mail that you have done so. Thank you.