BillyBlaze/OctoPrint-TouchUI #30: Out of memory

luokj, 2016-02-19:
1024 X 600 lcd touch
512M ddr
dibian OS,
Epiphany-browser  is killed by kernel when octoprint is printing a 3d model, because  out of memory. 

**So I think TouchUI may be have a memory leak!**

The following is kernle log:

[15532.879547] haproxy invoked oom-killer: gfp_mask=0x201da, order=0, oom_adj=0, oom_score_adj=0
[15532.888519] Backtrace: 
[15532.891113] [<c00181b4>] (dump_backtrace+0x0/0x10c) from [<c051d978>] (dump_stack+0x18/0x1c)
[15532.899963]  r6:000201da r5:00000000 r4:de6ae280 r3:c0744df0
[15532.905914] [<c051d960>] (dump_stack+0x0/0x1c) from [<c007b60c>] (dump_header.isra.14+0x54/0x150)
[15532.915222] [<c007b5b8>] (dump_header.isra.14+0x0/0x150) from [<c007b91c>] (oom_kill_process.constprop.19+0x70/0x22c)
[15532.926300]  r8:de62b780 r7:00000191 r6:000201da r5:00000000 r4:de62b780
[15532.933349] [<c007b8ac>] (oom_kill_process.constprop.19+0x0/0x22c) from [<c007be64>] (out_of_memory+0x23c/0x2d0)
[15532.944000] [<c007bc28>] (out_of_memory+0x0/0x2d0) from [<c007ed54>] (__alloc_pages_nodemask+0x654/0x678)
[15532.954010] [<c007e700>] (__alloc_pages_nodemask+0x0/0x678) from [<c007ac28>] (filemap_fault+0x214/0x400)
[15532.964050] [<c007aa14>] (filemap_fault+0x0/0x400) from [<c008e8d0>] (__do_fault+0x70/0x490)
[15532.972869] [<c008e860>] (__do_fault+0x0/0x490) from [<c00913a4>] (handle_pte_fault+0x78/0x618)
[15532.981994] [<c009132c>] (handle_pte_fault+0x0/0x618) from [<c00919d4>] (handle_mm_fault+0x90/0xc0)
[15532.991455] [<c0091944>] (handle_mm_fault+0x0/0xc0) from [<c001aba4>] (do_page_fault+0x15c/0x1f8)
[15533.000732] [<c001aa48>] (do_page_fault+0x0/0x1f8) from [<c0008454>] (do_PrefetchAbort+0x3c/0xa4)
[15533.010070] [<c0008418>] (do_PrefetchAbort+0x0/0xa4) from [<c00146d8>] (ret_from_exception+0x0/0x10)
[15533.019622] Exception stack(0xde6f1fb0 to 0xde6f1ff8)
[15533.024902] 1fa0:                                     00000064 0010db42 000a00a8 00000000
[15533.033447] 1fc0: 000a04d0 00095a8c 00095d58 000a00a4 00095cf0 00095cf0 000ae350 00095d04
[15533.042022] 1fe0: 0007a11f bec0eb98 0000d63f 0001a758 60000030 ffffffff
[15533.048950]  r8:00095cf0 r7:000a00a4 r6:ffffffff r5:60000030 r4:0001a758
[15533.055969] Mem-info:
[15533.058349] Normal per-cpu:
[15533.061248] CPU    0: hi:  186, btch:  31 usd:  43
[15533.066284] active_anon:120662 inactive_anon:354 isolated_anon:0
[15533.066284]  active_file:163 inactive_file:223 isolated_file:0
[15533.066284]  unevictable:0 dirty:0 writeback:0 unstable:0
[15533.066284]  free:722 slab_reclaimable:854 slab_unreclaimable:1650
[15533.066284]  mapped:477 shmem:397 pagetables:930 bounce:0
[15533.096405] Normal free:2888kB min:2880kB low:3600kB high:4320kB active_anon:482648kB inactive_anon:1416kB active_file:652kB inactive_file:892kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:519168kB mlocked:0kB dirty:0kB writeback:0kB mapped:1908kB shmem:1588kB slab_reclaimable:3416kB slab_unreclaimable:6600kB kernel_stack:1792kB pagetables:3720kB unstable:0kB bounce:0kB writeback_tmp:0kB pages_scanned:2410 all_unreclaimable? yes
[15533.137756] lowmem_reserve[]: 0 0
[15533.141204] Normal: 720*4kB 1*8kB 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB 0*8192kB = 2888kB
[15533.152923] 783 total pagecache pages
[15533.156738] 0 pages in swap cache
[15533.160186] Swap cache stats: add 0, delete 0, find 0/0
[15533.165649] Free swap  = 0kB
[15533.168670] Total swap = 0kB
[15533.189483] 131072 pages of RAM
[15533.192779] 1198 free pages
[15533.195709] 3305 reserved pages
[15533.199005] 2504 slab pages
[15533.201904] 1273 pages shared
[15533.205017] 0 pages swap cached
[15533.208282] [ pid ]   uid  tgid total_vm      rss cpu oom_adj oom_score_adj name
[15533.216064] [  164]     0   164      617      151   0     -17         -1000 udevd
[15533.223876] [ 1596]  1000  1596      799      176   0       0             0 haproxy
[15533.231903] [ 1626]     0  1626     1473      780   0       0             0 dhcpd
[15533.239746] [ 1652]     0  1652    34372     5884   0       0             0 octoprint
[15533.247955] [ 1659]     0  1659     7099      114   0       0             0 rsyslogd
[15533.256042] [ 1669]     0  1669      334       28   0       0             0 acpid
[15533.263885] [ 1693]     0  1693      839       40   0       0             0 cron
[15533.271636] [ 1703]   102  1703      722      119   0       0             0 dbus-daemon
[15533.279998] [ 1710]   104  1710      722       72   0       0             0 avahi-daemon
[15533.288482] [ 1711]   104  1711      691       55   0       0             0 avahi-daemon
[15533.296936] [ 1724]     0  1724     7600      120   0       0             0 lightdm
[15533.304962] [ 1746]     0  1746    13494     5592   0       0             0 Xorg
[15533.312683] [ 1778]     0  1778     1514      106   0     -17         -1000 sshd
[15533.320434] [ 1810]     0  1810     3782      137   0       0             0 lightdm
[15533.328460] [ 1813]     0  1813     6616      270   0       0             0 console-kit-dae
[15533.337188] [ 1880]     0  1880     5429      173   0       0             0 polkitd
[15533.345214] [ 1892]  1000  1892     3005       61   0       0             0 lxsession
[... 23 lines cut ...]
[15533.539642] [ 2011]     0  2011     5468      137   0       0             0 udisks-daemon
[15533.548187] [ 2012]     0  2012     1262       66   0       0             0 udisks-daemon
[15533.556762] [ 2017]  1000  2017     1931      108   0       0             0 gvfs-gphoto2-vo
[15533.565490] [ 2019]  1000  2019     4335      109   0       0             0 gvfs-afc-volume
[15533.574249] [ 2088]     0  2088     1024       80   0       0             0 bash
[15533.581970] [ 3685]  1000  3685    22167      566   0       0             0 lxterminal
[15533.590270] [ 3686]  1000  3686      411       30   0       0             0 gnome-pty-helpe
[15533.598999] [ 3687]  1000  3687      342       19   0       0             0 sh
[15533.606567] [ 3691]  1000  3691     1030       86   0       0             0 bash
[15533.614288] [ 3699]     0  3699      616      157   0     -17         -1000 udevd
[15533.622131] [ 3703]     0  3703      616      157   0     -17         -1000 udevd
[15533.629974] [ 3711]  1000  3711     1147       82   0       0             0 su
[15533.637542] [ 3719]     0  3719     1029       86   0       0             0 bash
[15533.645263] [ 3744]     0  3744     1050      118   0       0             0 top
[15533.652923] [ 3770]     0  3770   145282    54825   0       0             0 epiphany-browse
[15533.661682] [ 3794]     0  3794     6136      162   0       0             0 upowerd
[15533.669677] [ 3849]     0  3849    50260     4251   0       0             0 yelp
[15533.677429] [ 3860]     0  3860    50221     4231   0       0             0 yelp
[15533.685150] [ 3924]     0  3924     2411      558   0       0             0 sshd
[15533.692901] [ 3926]     0  3926      667      212   0       0             0 sftp-server
[15533.701263] [ 3929]     0  3929     2186      133   0       0             0 sshd
[15533.709014] [ 3932]     0  3932     1029       85   0       0             0 bash
[15533.716766] [ 5060]     0  5060    35945    33631   0       0             0 cpan
[15533.724487] Out of memory: Kill process 3770 (epiphany-browse) score 401 or sacrifice child
[15533.733245] Killed process 3770 (epiphany-browse) total-vm:581128kB, anon-rss:219236kB, file-rss:64kB

luokj, 2016-02-19:
More information, 
When open TouchUI on beaglebone, don't connect to printer, memory usage is low, but connect to printer and load gcode file,   memory usage is always high, about 60% and don't reduce.

luokj, 2016-02-20:
I am sorry, It cause by GCode Viewer.  It eat  too much memory.

BillyBlaze, 2016-02-20:
Thanks for reporting & of course no problem :+1: so if you turn off the G-Code viewer you don't have any issue's?

luokj, 2016-02-21:
If turn off, issue will disappear. 
But a tradeoff fix, I have a low memory setting in config.yaml. It will use %20 memory.
gcodeViewer:
  mobileSizeThreshold: 1097152
  sizeThreshold: 1097152

BillyBlaze, 2016-02-21:
Ok awesome :+1: then it will not require an emergency fix. I will add the option to disable the g-code viewer while printing. (I think I will turn that on by default in Touchscreen mode) Of course I will add the option that you can enable the g-code viewer while printing

BillyBlaze, 2016-02-21:
I am going to close this issue due to another open issue with the same problem, for updates follow issue #18 

Once again thanks for the report and your time!

closed 2016-02-21 by BillyBlaze

luokj, 2016-02-22:
No thanks, Touch UI is awesome!