BillyBlaze/OctoPrint-TouchUI #31: No Camera running touch ui on a Raspberry pi.

Bigfellas, 2016-02-24:
Hey Paul. I have moved up to a official pi 7" screen. Mind you this problem existed on the small screen as well. Anyway I am not getting camera feed in the ui. getting a little ? icon on the screen. So I assume something is missing.
Thanks
Matt

BillyBlaze, 2016-02-24:
Hi Matt,

Does the webcam work in the normal interface? And if so, could you perhaps add the .octoprint log in a pastebin??

~Paul

Bigfellas, 2016-02-25:
yup working in the normal interface. 
I think this is what you were after 
http://pastebin.com/8A6DE4nH

Bigfellas, 2016-02-25:
Loads fine on a iPhone as well. So must be something missing with the Pi install.

BillyBlaze, 2016-02-25:
Okay something weird is going on, I shortly had the same issue on my touchscreen but after saving the webcam settings with the touchscreen it resolved itself. (just open Settings > Webcam, press test and if you see in the preview the image feed, just save it and reboot the browser)

BillyBlaze, 2016-02-25:
also it seems the feed stops working after a while (which is a kweb3 bug), if go to another page and you wait a few seconds and go back, then octoprint will reset the feed and it continues.

Bigfellas, 2016-02-27:
How do I press test I can't see anywhere to do that

BillyBlaze, 2016-02-27:
Sorry I forgot I was on OctoPrint 1.3.0 (devel branch) where this test button is a new feature. Please take a look at the [SnapStream plugin](http://plugins.octoprint.org/plugins/snapstream/).

This plugin will fallback if mjpeg fails, and is exactly what you want.

After installing the plugin, save the SnapStream Settings with your config. Be sure you have unchecked ``Only use snapstream when mjpg-streamer fails``. For some reason SnapStream doesn't work when MJPEG has failed in kweb3.

p.s. using SnapStream will lower your CPU usage, so I recommend using it with Touchscreens.

Bigfellas, 2016-02-28:
Unfortunately no go. I even switched to the Devel branch Still get a little ? even in the test screen.

BillyBlaze, 2016-02-28:
Sorry I need more information; Did you install the SnapStream plugin? and what was the SnapStream settings you saved after installing it? Did you verify you have the correct settings?

(Verify settiigs: If you go to a desktop and open the browser with ``http://localhost:0000/cam.jpg``, do you see a static image? (change the url to your settings)

Did you also try to uncheck ``Only use snapstream when mjpg-streamer fails``?

There has to be something with your configuration since I am unable to reproduce the problem with SnapStream.

Bigfellas, 2016-02-28:
Sorry Paul blundering my way through this. the only setting other than "Only use snapstream when mjpg-streamer fails" and I have tried this both ways, and frame rate is the path. I have left this at default  /webcam/?action=snapshot
Do I need to change this.

http://192.168.1.250:8080/cam.jpg
When I put this in I get  404: Not Found!
Could not open file

BillyBlaze, 2016-02-28:
I think the valid URL should be http://192.168.1.250:8080/?action=snapshot for your setup.

If that works in the browser then replace ``/webcam/?action=snapshot`` with ``http://192.168.1.250:8080/?action=snapshot`` and save it.

BillyBlaze, 2016-03-02:
any luck yet?

Bigfellas, 2016-03-02:
Sorry Paul. Been slammed at work no time for play. I'll have a crack tonight and get back to you

BillyBlaze, 2016-03-02:
don't apologize :) I was just curious. Good luck!

Bigfellas, 2016-03-02:
I just put a post on google+ octoprint not sure if you saw it about touch ui not fitting on a pi 7" screen. Not sure if you can shed any light on this.

BillyBlaze, 2016-03-02:
thanks for the head up!

Bigfellas, 2016-03-03:
Legend. That worked. Thanks again

BillyBlaze, 2016-03-03:
Awesome!! Glad it works now. Your welcome!

closed 2016-03-03 by BillyBlaze