BillyBlaze/OctoPrint-TouchUI #33: TouchUI shown in desktop chrome

KriLL3, 2016-03-09:
Chrome on both my desktop and laptop are showing TouchUI, Firefox shows the normal non-touch UI, is the desktop chrome user agent mistaken for mobile chrome?

Would be nice if there was a button in touchUI to disable it if it's accidentally activated, or a URL parameter to force it off etc.

BillyBlaze, 2016-03-09:
Thanks for reporting!

For anyone having this issue and are unable to turn off TouchUI, please be sure you have enabled the checkbox "Display TouchUI settings when the touch interface is inactive", after you enabled this, the ``TouchUI Settings`` will be visible again and you can turn off TouchUI.

BillyBlaze, 2016-06-19:
Again thanks for reporting.

This has been fixed in 0.3.4

closed 2016-06-19 by BillyBlaze

SBachman, 2019-04-07:
Looks like this is happening again. Works fine in Edge and mobile Chrome but on desktop Chrome it defaults to TouchUI, the hamburger menu isn't clickable, and "Display TouchUI settings when the touch interface is inactive" doesn't seem to work.

sciencecentredelft, 2019-06-08:
I have the same problem. Touch UI shows on desktop Chrome. How can I fix this? How do I delete plugins?

BillyBlaze, 2019-06-08:
Hi @sciencecentredelft,

Could you install TouchUI from the maintenance branch; you can uninstrall TouchUI and install it by url with the following url: https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip

Please let me know if that resolved your issue.

DanCrank, 2019-07-13:
I have the same problem, and unfortunately the maintenance branch does not fix it for me. I am getting TouchUI my desktop browser (Chrome 64-bit 75.0.3770.100) any time the TouchUI plugin is active, even with the "based on resolution" and "based on touch device" options turned off.

Oddly, I don't have the same problem on my laptop running the same version of Chrome - it always correctly displays the non-TouchUI interface. The laptop's display resolution is 2560x1440. The desktop has two displays, 1920x1080 and 1440x900. I saw in Issue #280 that having a resolution under 980 was one of the triggers, but I am still seeing TouchUI with that option turned off. Perhaps having two displays of different resolutions is confusing it.

Please let me know if I can get you any more information, or test a fix for you. Thanks!

Randomjimboy, 2019-11-03:
Hi, trying to remove this plugin from my desktop running windows 7 64 browser chrome