BillyBlaze/OctoPrint-TouchUI #34: Plugin fails to install via Plugin Manager

KeiroD, 2016-03-12:
Hi,

I'm running `OctoPrint 1.2.9` for the version.

As for Pip?

```
keiro@emberberry:~$ pip --version
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)
```

This is the following error when the Plugin Manager attempts to install your plugin:

```
Downloading/unpacking https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/master.zip
  Downloading from URL https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/master.zip
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1198, in prepare_files
    do_download,
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1376, in unpack_url
    self.session,
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 572, in unpack_http_url
    download_hash = _download_url(resp, link, temp_location)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 433, in _download_url
    for chunk in resp_read(4096):
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 421, in resp_read
    chunk_size, decode_content=False):
  File "/home/keiro/OctoPrint/venv/lib/python-wheels/urllib3-1.9.1-py2.py3-none-any.whl/urllib3/response.py", line 256, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/home/keiro/OctoPrint/venv/lib/python-wheels/urllib3-1.9.1-py2.py3-none-any.whl/urllib3/response.py", line 186, in read
    data = self._fp.read(amt)
  File "/usr/lib/python2.7/httplib.py", line 573, in read
    s = self.fp.read(amt)
  File "/usr/lib/python2.7/socket.py", line 380, in read
    data = self._sock.recv(left)
  File "/home/keiro/OctoPrint/venv/lib/python-wheels/urllib3-1.9.1-py2.py3-none-any.whl/urllib3/contrib/pyopenssl.py", line 188, in recv
    data = self.connection.recv(*args, **kwargs)
ZeroReturnError
```

How do fix?

BillyBlaze, 2016-03-12:
Hello,

Sorry, I cannot really help you with solving the issue, it seems like it fails to download the zip file over SSL and is either an issue in OctoPrint or of your envoirment. 

One thing you could try is installing with ``http://`` instead of ``https://``

If that didn't work, then you can get the best and quickest help @ [OctoPrint+](https://plus.google.com/communities/102771308349328485741) or [OctoPrint mailinglist](https://groups.google.com/group/octoprint)

Good luck,
Paul

closed 2016-03-12 by BillyBlaze

KeiroD, 2016-03-12:
I managed to get it to install via the Install via URL option. It was odd, but... it worked.