BillyBlaze/OctoPrint-TouchUI #35: Camera not working on Touchscreen

seefew, 2016-03-23:
Hi there,
i'm new to Github so pls be lenient with me if I make a mistake
My env:
Raspy 2 , RaspyCam, 5" TFT w WaveshareTouch, Ethernet
running latest OctoPi, TouchUI, kweb. 
Raspy starts automatically to TouchUI
At this point: Many thanks for your efforts with this Plugin
When connect to Raspy via my iMac or iPhone i get full access to TouchUI incl the Cam.
On the Raspy TFT mostly functions are reachable.
My problem:
At the Cam Tab only a Question mark is shown, no img or stream from the Cam.
I hope you can help me solving
Greets

BillyBlaze, 2016-03-23:
Hi seefew,

Your welcome and thanks :+1: 

Did you try installing the plugin [SnapStream](http://plugins.octoprint.org/plugins/snapstream/)? If not, you need this plugin to get it working; see some more details in issue #31

I will close these issue, if you have any questions just add it here, I will still see your responses.

~Paul

closed 2016-03-23 by BillyBlaze

seefew, 2016-03-23:
Hi Paul,
&thx for your quick reply
After i followed these instruction:

> I think the valid URL should be http://192.168.1.250:8080/?action=snapshot for your setup.
> 
> If that works in the browser then replace /webcam/?action=snapshot with http://192.168.1.250:8080/?action=snapshot and save it.

with my IP it works quite fine 
Thanks a lot for your help
Greetings