BillyBlaze/OctoPrint-TouchUI #36: unable to load page

Bigfellas, 2016-03-27:
This is a problem I have always had but have only just got around to asking the question.
I get this error every time I start my pi and display. more times than not if I hit the onscreen button is will come good. Do you have any idea what could cause this. Thanks Matt.
[image]

BillyBlaze, 2016-03-27:
Basically the browser has already started before OctoPrint was completely started. I don't know what the best solution is, but I think we should add a delay somewhere to make sure OctoPrint's server has started before the browsers starts.

Bigfellas, 2016-03-29:
Do you have any suggestions on how to do that 😜

BillyBlaze, 2016-03-29:
Not at the moment (this needs some R&D + testing), but it should be something like this:

1. Load octoprint default before loading default touchui with:  
   ``[ -r /etc/default/octoprint ] && . /etc/default/octoprint``
1. Add a check if localhost is accessible  
    ``while ! echo exit | nc localhost $PORT; do sleep 10; done``

BillyBlaze, 2016-06-10:
I just finished creating a bootloader, this will ensure that TouchUI is loaded in the right order. If any of the steps are wrong an error will appear and you can retry easily by tapping the screen.

Could you try this out, if this works for you? If so, please follow below instructions:

1. Uninstall TouchUI and install TouchUI by url with the following url:

    ```
    https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip
    ```

1. Login to ssh and get the new bootloader:

    ```
    git clone https://github.com/BillyBlaze/OctoPrint-TouchUI.git ~/TouchUI-autostart && cd ~/TouchUI-autostart && git checkout autostart-dev
    ```

1. Register the new paths:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
    sudo update-rc.d touchui defaults
    ````

reboot and the new screen should appear

Bigfellas, 2016-06-11:
Unfortunately that has stopped my auto login. I set this up such a long time ago I cant even remember what version I am running. I might just go through the setup process again

BillyBlaze, 2016-06-11:
You can also switch back to your old boot method and install everything again when this feature is officially released.

Just run:
````
sudo cp ~/OctoPrint-TouchUI-autostart/touchui.init /etc/init.d/touchui
sudo chmod +x /etc/init.d/touchui
sudo cp ~/OctoPrint-TouchUI-autostart/touchui.default /etc/default/touchui
sudo update-rc.d touchui defaults
````

Bigfellas, 2016-06-11:
Cool So I just did a clean install, then followed your instructions. Works a treat. Love the splash screen.
Thanks for doing that

BillyBlaze, 2016-06-11:
Awesome! Thanks again for checking and your welcome! I guess I can close this issue once this is realeased.

BillyBlaze, 2016-06-19:
Ok, 0.3.4 has just been released. If you installed the maintenance branch then please follow these instructions to prevent TouchUI not working in the future:

1. Remove TouchUI and install the latest 0.3.4 version through the plugin manager

1. Make sure you're using the right bootloader repository and are up-to-date:

    ```
    rm -rf ~/TouchUI-autostart/
    git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/
    ```

1. Register the new paths:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
    sudo update-rc.d touchui defaults
    ````

reboot and it should work :)

closed 2016-06-19 by BillyBlaze