BillyBlaze/OctoPrint-TouchUI #37: Problem installation?

fbarcenas, 2016-04-01:
Hi,
i install everything like the wiki:

> https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser

And not appears error, but when reboot at finish, not show the windows with touchUI, show normal Desktop...

I do step for step 3 times and i don't know that can happen

Some ideas? Things to check?

BillyBlaze, 2016-04-01:
Hello,

It really sounds like your configuration ``sudo raspi-config`` is still set on ``Enable Boot to Desktop`` and should be set to ``Enable Boot to Command line``

fbarcenas, 2016-04-01:
You are the best, 
i test and is very fine.
All solved, i want say also than in another raspberry i have this problem:
https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/36

But in the new version of Octoprint 1.2.10 seem than is solved... I can read a message than say than Octoprint is loading...

Thanks again, you are god.

Bye!

BillyBlaze, 2016-04-02:
Awesome glad to hear it's working.
I'll take a look at #36 as soon as I can.

and Your welcome ofcourse :+1:

Bye!

closed 2016-04-02 by BillyBlaze