BillyBlaze/OctoPrint-TouchUI #38: Touch screen stops working after tapping touch UI Setting.

mrp192, 2016-04-17:
So I am in full screen mode and I need to enable virtual keyboard. When I go to make it active in the settings tab for screen doesnt load. Just kinda grays it is going to load. After I tap a part of the screen touch stops working.

Thanks for the help!

mrp192, 2016-04-17:
I just tried to get to User setting and the same thing happened..

Best of all when I try to remove and reinstall I just get and unknown error.

BillyBlaze, 2016-04-17:
Hi,

Could you enlighten me with some information about which browser/octoprint version you're running?

Also could you check the logs/octoprint.log to see what the unknown error is?

Thanks.

mrp192, 2016-04-17:
Minimal Kiosk Browser 
OctoPrint 1.2.10
Raspberry pi 3 with 7" screen

Does this mane scenes? 
2016-04-16 22:37:02,679 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.0.82 2016-04-16 22:37:37,200 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk 2016-04-16 22:38:13,187 - octoprint.plugin.core - WARNING - Trying to mark an unknown plugin touchui 2016-04-16 22:38:13,188 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins and installed plugin packages... 2016-04-16 22:38:13,630 - octoprint.plugin.core - INFO - Found 6 plugin(s) providing 5 mixin implementations, 2 hook handlers 2016-04-16 22:38:13,754 - octoprint.plugin.core - INFO - 6 plugin(s) registered with the system: | CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/cura | Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/discovery | Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/pluginmanager | Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/softwareupdate | !TouchUI (0.3.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui | Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/virtual_printer 2016-04-16 22:38:21,682 - octoprint.server.api - INFO - Performing command: sudo service octoprint restart 2016-04-16 22:38:21,834 - octoprint.server - INFO - Shutting down... 2016-04-16 22:38:22,362 - octoprint.server - INFO - Goodbye! 2016-04-16 22:38:26,558 - octoprint.server - INFO - Starting OctoPrint 1.2.10 (master branch

mrp192, 2016-04-17:
So I was able to get touchUI to be removed. But when I was in the normal octoprint screen when I would click on any menu it would act like it opening then nothing and touch wouldnt work.

BillyBlaze, 2016-04-18:
Ok, it sounds like you're having OctoPrint issues, could you try to reproduce this problem in the normal octoprint screen and see if your javascript console is giving you any errors?

Some more information about javascript errors: https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers

mrp192, 2016-04-22:
Sorry ive been busy with work. Whats weird is that it worked when I had the screen on a b+ but there was just lad. Now that im on a pi3 theres no lag but I just cant like on menus when if I use a mouse.

BillyBlaze, 2016-04-23:
I see the issue, I was also able to reproduce it.
Currently investigating it.

BillyBlaze, 2016-04-23:
Fixed in the latest 0.3.3 release.

mrp192, 2016-04-23:
Awesome! Thanks you!

mrp192, 2016-04-23:
So... it is still doing the same thing after the date.

BillyBlaze, 2016-04-24:
ok thanks for reporting :(

mrp192, 2016-04-24:
Will it help if I record a video of whats happening?

BillyBlaze, 2016-04-25:
I'll try to look at it as soon as possible, I might have released it too early maybe. If I can't reproduce it I might need a video but since #40 has similar issues I don't think it will be that hard.

BillyBlaze, 2016-04-25:
btw, if you're running Jessie, you might want to consider switching to the latest Chrome version. (it's in the Wiki)

BillyBlaze, 2016-04-27:
Sorry, I forgot to test this in the latest KWEB browser and my fix (0.3.3) worked for 1.6.9 and lower (you have atleast 1.7.0). However since you're running higher then that means you have Jessie. Therefor with this bug I am dropping support for KWEB and add support for Chromium on Jessie.

So sorry for the troubles, but please upgrade your browser with these steps: https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(Chromium-&-Jessie)#upgrade-from-iceweasel--kweb

closed 2016-04-27 by BillyBlaze

mrp192, 2016-04-27:
Is there a way I can message you directly? I upgraded and still having the same problem, I even started from new with the same problem. Could it be that i'm using the offical pi 7" screen?

I'm using the latest nightly build for octoprint. Did everything that was in the write up.

BillyBlaze, 2016-04-27:
You can find in the #OctoPrint channel on irc, else you can contact me here :D 

Since you're running octopi, then that means you have Wheezy. Chromium will only work on Jessie (not Jessie Light). If you run the following command what does it say? ``sudo dpkg -l | grep kweb``

mrp192, 2016-04-27:
pi@octopi:~ $ sudo dpkg -l | grep kweb
ii  kweb
1.7.0-4
arm
hf        Minimal Kiosk Browser

BillyBlaze, 2016-04-27:
ok, thats the wrong version for your Debian version. You need 1.6.8, I have updated the Wiki, to be sure that this wont happen again.

To remove kweb run:
``sudo dpkg -r kweb``

After that run:
````
wget -P /home/pi/ http://steinerdatenbank.de/software/kweb-1.6.8.tar.gz
tar -xzf /home/pi/kweb-1.6.8.tar.gz
cd /home/pi/kweb-1.6.8
./debinstall
````

Restart TouchUI and you it should work!

mrp192, 2016-04-27:
YOU ARE AMAZING!! IT WORKED!!!

BillyBlaze, 2016-04-28:
Awesome to hear :+1: