BillyBlaze/OctoPrint-TouchUI #39: 1.0.0

BillyBlaze, 2016-04-18:
# Develop is highly broken a.t.m.

- [x] Refactor dashboard
- [ ] Refactor files list (use hammerjs)  
      _90% done_
- [ ] Refactor navbar (use hammerjs)  
      _90% done_
- [ ] Refactor dropdowns  
      _20% done_
- [ ] Refactor controls tab  
      _5% done_ (not for up for grabs)
- [ ] Refactor virtual keyboard  
      _0% done **and broken**_
- [ ] Refactor form elements  
      _5% done - material design pattern_
- [ ] Refactor modals  
      _10 done_
- [ ] More touchscreen friendly  
      _0% done_
- [ ] New feature: Themes from repo  
      _0% done_

Alex9779, 2016-04-24:
Hey, just wanted to thank you for the 0.3.3 update and 3ce7d10f304c2234d1673fc62a0ec3182af15cb5!
Thanks, thanks, THANKS, TYVM!!!
Now I finally can use the home screen links to my two printes without the need to relog every time!!!
Keep up the good work...
(Sorry, didn't find an other way to get in touch with you quickly...)

BillyBlaze, 2016-04-25:
:+1: Your welcome! If you have any annoyances or improvement, just let me know!!

Alex9779, 2016-04-25:
Unfortunately I was a bit too early excited. The login persists a bit longer than before. But killing the "web app" on iOS still removes the login. I would accept that.
But it still seems to happen if you just switch to other apps, in my case I spend G+, then a game, then mail, then I switched back and the login was gone again...
Shall I open an issue? Or is it more related to OctoPrint itself than only with your cool addon?

I did some researches myself but I do not find any good information about how Apple handles that stuff, all workarounds I find talk about just to set some meta data to prevent iOS to start web app mode for home screen links...

BillyBlaze, 2016-04-25:
Ugh, this is a real pain in the ass and I feel your pain. Sadly this is not a task TouchUI can help with.

After browsing the codebase of OctoPrint I don't think it would be an easy task. It currently relies on the built-in remember me functionality of Flask, that only support cookies. If I understand correctly everyone use the sessionStorage approach on mobile. Anyway, I also have some intrest in this, so I will try to talk directly with foosel about this.

BillyBlaze, 2016-04-27:
@Alex9779 can we keep this ticket open for 1.0.0 and can we continue on in ticket #41 ?

Alex9779, 2016-04-27:
Sure... I just "used" it to get in contact with you because of lacking alternatives... And then I was not sure, and you agreed at least a bit, that it is not fully up to you... -> #41

maxbots, 2016-05-25:
Hi Billy. A keyboard specific to GCode would be really convenient. Just the M & G keys, spacebar, numbers, and relevant punctuation. Even easier, just add M & G to the numeric keyboard and you don't really need to make anything custom.

BillyBlaze, 2016-07-14:
Add compatibility for (https://github.com/donovan6000/M33-Fio/issues/161)[M33-Fio]

lciscon, 2016-08-21:
I think this is a feature-request rather than a howto: I have an OctoPrint plugin that adds a tab to the web UI.  Under TouchUI this appears in the dropdown.  I'd like to have a way to optionally add an icon for it on the main screen instead.  I'm running the RPI 7" Touchscreen and there's room for a couple more main-screen icons.

BillyBlaze, 2016-08-21:
Thanks for the suggestion, the dropdown have already been removed in 1.0.0 and replaced with new navigation, on touchscreen device it will show an condensed overlay with round buttons and presented with an icon, all items are displayed there. On larger devices it will have a swipe-in menu, where it will fit as many items as it can fit.

lciscon, 2016-08-21:
That sounds perfect!  Thanks!!

closed 2020-12-09 by github-actions[bot]