BillyBlaze/OctoPrint-TouchUI #40: Settings button works just once

pcfriend, 2016-04-23:
Hi Paul
TouchUI works great now. Its very useful to combine it with a touchscreen. Theres just one little problem: I crashed my raspberry last week, so I had to redo all the installation steps for Octoprint. This gave me the chance to update all my plugins. But now I have found a new bug in TouchUI. When I click the button "settings", you know the button with the three bars, and then exit the settings menu I wont be able to click the settings button again. This problem does only appear on the touch screen. On a PC it works just fine. I don't know really the reason for that issue, so I ask you for help. I also have installed the plugin System Command Editor. Maybe it has something to do with that?
Thank you for helping
Raphael

BillyBlaze, 2016-04-24:
Thanks for reporting Raphael!

Just to be sure, you're running KWEB, OctoPrint 1.2.10 and TouchUI 0.3.3 now??

pcfriend, 2016-04-24:
Thats correct.

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/40#issuecomment-213909160

BillyBlaze, 2016-04-25:
Ok, I will investigate this problem a bit more, btw, if you're running Jessie, you might want to consider switching to the latest Chrome version. (it's in the Wiki)

BillyBlaze, 2016-04-27:
Sorry, I forgot to test this in the latest KWEB browser and my fix worked for 1.6.9 and lower (you have atleast 1.7.0). However since you're running higher then that means you have Jessie. Therefor with this bug I am dropping support for KWEB and add support for Chromium on Jessie.

So sorry for the troubles, but please upgrade your browser with these steps: https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(Chromium-&-Jessie)#upgrade-from-iceweasel--kweb

pcfriend, 2016-04-27:
Hi PaulI run the latest Octopi image and as I know it's Raspbian Wheezy. Raphael

BillyBlaze, 2016-04-27:
hmm so what version of kweb are you running then? 1.6.8? you might still have the installation directory in your home directory

to get the version run ``sudo dpkg -l | grep kweb``

BillyBlaze, 2016-04-28:
Read #38 for more information

pcfriend, 2016-04-28:
It says:
`ii  kweb                                  1.6.8-1                                 armhf        Minimal Kiosk Browser`
I think that's the right version, isn't it?

BillyBlaze, 2016-04-28:
Yeah that looks right. I will investigate this a bit more. Maybe also do a fresh reinstall on my Wheezy card.

pcfriend, 2016-04-28:
Hi PaulI just found the reason for this problem. But I haven't a solution. The settings button works fine when I don't use one of my custom system commands made with the plugin available in the plugin manager. When I use one of these commands the Button doesn't work anymore. However this problem does only appear on the touchscreen. On my iPhone  it works fine. So I guess the problem has something to do with kweb.  Maybe you know a solution but thats definitely not your job to fix kweb faults ;). Maybe I shoud change my OS to Jessie, so I can install chromium. Thank you for helping.  Raphael

BillyBlaze, 2016-04-28:
Hi Raphael,

Thanks, that saves alot of figuring out. I do got to say as frontend developer it's required I get websites to work for a specific browser :smiley: 

But yeah, I bet this problem is more an issue in the custom system commands, and I doubt that the author will support kweb. However if you take the effort of installing Jessie and Chromium, then I bet it would be easily resolved. (+ better performance)

Your welcome and lemme know if there is any issues!
~Paul

closed 2016-04-28 by BillyBlaze