BillyBlaze/OctoPrint-TouchUI #41: [Proof of Concept] Persistent login on iOS

BillyBlaze, 2016-04-27:


Alex9779, 2016-04-27:
Sure I help testing...
Is ist just downloading the "maintenance" archive and install this?

BillyBlaze, 2016-04-27:
ok, pushed.

The easiest is indeed to uninstall TouchUI, and reinstall with the maintenance link: `https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip`

When installed you should be running 0.3.4

Currently I implemented the following mechanism:

1. Check onload & onWindowVisilibty if you have a cookie "remember_token"  
   - If not, and it is in the localStorage then it will replicate it into a cookie

1. On login / logout  
  - Add or delete the localStorage item

Alex9779, 2016-04-27:
Ok I might have some positives results.
Testing and comparing is easy for me, I got two printers both with an RPi and OctoPi/OctoPrint with nearly the same addins.
Update only one with the maintenance branch.
I started both as web apps from the home screen, logged in into both and ticked "Remember me".
Then I switched apps, started games and did some stuff for about 2 minutes.
I switched back to the first printer with 0.3.3, I was not logged in.
I switched back to the second printer with 0.3.4, I was logged in!

Next test I closed both "apps" in the app switcher.
I started the first (0.3.3), I was not logged in (what a surprise ;) ).
I started the second (0.3.4), I was logged in!

BillyBlaze, 2016-04-27:
That's awesome news :+1: 

I will release this feature with either 0.3.4 or with 1.0.0, depending if 0.3.3 needs bugfixes.

Alex9779, 2016-04-28:
I didn't open the 'app' since yesterday when I did the first test.
Opened it today first time and I am still logged in! And really. I executed some commands. (Had it a while ago it looked like I was logged in but was not able to execute commands, that was with 0.3.2 or 0.3.3. Can't remember.)

Good work whatever you did :) I had a look at the change set but web development is not really something for me...

Alex9779, 2016-04-28:
Ok now I have again an issue.
I installed the maintenance 0.3.4 on the second printer.
If I now log into the seance one the credentials of the first seem the get invalid.
I am then unable to log out. Nothing happen when doing it. I have to terminate both "apps". Then I can logout or it is logged out, both happened.
Do they share the settings somehow?

BillyBlaze, 2016-04-28:
hmm you access both Pi's with a different IP right? In that case they shouldn't share the localStorage. If both printer's user-interface are on the same domain then they do share localStorage.

But I figured out your problem; it sort have to do with OctoPrint itself, once the backend refuse to provide service (due to an invalid token) it refuses to do any action, even an action like logout. I think this is better to fix inside OctoPrint itself.

Alex9779, 2016-04-28:
Yes different IPs but they are on the same local domain and I use DNS to access them.
So you say it would be fine if I use IPs instead to access?

BillyBlaze, 2016-04-28:
Do you access them like ```http://printer/1/``` and ```http://printer/2/```??

Anyway I should add some more logic to this mechanism.

Alex9779, 2016-04-28:
Nope actually big box.alitecs.local and ultimaker2.alitecs.local...

BillyBlaze, 2016-04-28:
hmm, no they shouldn't share their localStorage with eachother, but we're talking here about iOs, so we should take in consider that they might do share localStorage.

Could you try to reproduce this problem with access them both with IP's?

Alex9779, 2016-04-28:
This is strange, I had a state some tests when it worked with the IP created icons, now I tried to clear everything, logged out every where, controlled that I am logged out, then closed all in the app switcher, then tried again. Now the same issue with the IP created ones.

And I discovered another issue, it seems app switching is handled different than going back to home screen and opening the icon again.
When I am logged into one instance and switch to another app using the switcher and switch back then the progress wheel in the status bar turns infinitely and the controls when being logged in are not loaded. Actually the menu says "Login".
When I quit to home screen and start the icon I am logged in...

BillyBlaze, 2016-04-30:
Ok, first off, I have [pushed a simple fix](https://github.com/foosel/OctoPrint/commit/8e98e9c40c67ae6a7e81625cfa211783256056a5) into OctoPrint, that will allow you to logout when your session is invalid. It's currently on the maintenance branch and will be shipped into the next release.

I will try to figure out some more whats going wrong, but I have a feeling it has todo with the websockets. (i.e. once you open the second one, it loses the socket in the first one)

Alex9779, 2016-05-21:
Wouldn't it be a temp fix to set those metas to disable the app mode on mobile devices?
I have no problems with login persistence when I use Safari so just with the home screen icons. If the app mode would be disabled it would just open another tab in Safari... I would be totally fine with this...

BillyBlaze, 2016-05-21:
Hmm, good question, but let me answer this with a counter questions; it works fine with only one printer right? And what if Apple will fix this in their next release? And what about old iOs devices that don't have this problem, should I kill this feature for them too?

Alex9779, 2016-05-21:
I can't tell for sure but I think it works with both... What about a toggle to let me choose I want it as an app?

Alex9779, 2016-05-21:
I just verified: in Safari two tabs one for each printer both logged in work with out issue...

BillyBlaze, 2016-05-21:
Thanks for the info! but can't you just delete the home screens icons and we can leave this feature for people with one printer?

Alex9779, 2016-05-21:
Oh yes I could but I want to have shortcuts. That's what I think the home screen is for. I don't wanna open Safari. Open a new tab. Navigate through my bookmarks... I wanna click and "status"...

BillyBlaze, 2016-05-21:
crap, I understand the issue now, your not allowed to make a shortcut as long as the home screen feature is enabled?

Btw. wouldn't it be a better and nicer if you could switch between printers within OctoPrint itself?

Alex9779, 2016-05-21:
How should that work? Two printers, two RPis, two OP instances. But yes I'd love to have a master OP maybe a third instance. Possible virtual that acts as a master...

Alex9779, 2016-05-21:
And to you first question. Yes that's it. If I make a shortcut in the home screen this activates the app mode and I have no option to just do it as other simple pages behave that is just open a tab in Safari...

BillyBlaze, 2016-05-21:
1) Well all the data on the interface of OctoPrint is received through AJAX, you can set this end point to any url you like, so you could point it to another IP address and receive/send to that API. It wouldn't be that simple through, it does require some more codes, but since I suspect alot of people have multiple printers, it would be a nice feature for OctoPrint.

2) You can remove the meta line yourself for now:

   1. Find TouchUI
     ``cd /home/pi/oprint`` (or what your main octoprint location is)  

      Find it with:
       ``venv/bin/pip show TouchUI``

   2. Copy the _Location_ path and run
     ``cd /your/path/``
     ``nano octoprint_touchui/static/js/touchui.bundled.js``

   3. Press ``F6`` and search for ``apple-mobile``

   4. Remove the line or add to forward slashes at the beginning of the line

   5. Save (CTR-O, Enter, CTR-X)

   6. Reboot octoprint 
     ``sudo service octoprint restart``

Alex9779, 2016-05-21:
1) sounds like a new plugin project for you! ;) Unfortunately my Python skills are at the very beginning and I don't have time atm to dig in.

2) Ok that's fine for me... Thanks for pointing out where I have to change that...

BillyBlaze, 2016-05-21:
Another thing you could try without needing to edit codes; disable TouchUI, make a shortcut after OctoPrint was reloaded, and then after that re-enable TouchUI.... :+1:

Alex9779, 2016-05-21:
Doesn't work. Even with disabled TouchUI, not only in settings but the complete plugin in the manager does open in app mode form a home screen link...

Alex9779, 2016-05-23:
Ok I tried both variants but editing the code has no effect on this too. I searched everywhere for the file you mentioned and found it one time and commented the code there. Actually I commented the next line too but iOS still switches to app mode from a new created bookmark link on the home screen...

BillyBlaze, 2016-06-19:
This feature landed in 0.3.4

I will leave it there, sorry that I don't have an anwser on your wierd iOs behavior, I think the only solution is to launch an app with the 1.0.0 release.

closed 2016-06-19 by BillyBlaze