BillyBlaze/OctoPrint-TouchUI #42: Cannot log in via touchscreen interface

diverp01, 2016-05-13:
I recently installed TouchUI on my Raspberry Pi 3 running Jessie and the latest Octoprint. I have had a few issues in installation that I am still tracking down. 

1) I can get the camera to show up in the Octoprint page on a laptop, but it will not show up on my 10" touchscreen on my Pi. Everything else does show up (once logged in). I saw a couple of other bugs related to this issue and have not been able to remedy yet.

2) I cannot login for the life of me from the TouchUI interface. I can on a touchscreen laptop using the Octoprint interface, but cannot in the TouchUI interface. This is my main issue now.

The interface looks great and expandable. I look forward to ironing out these last couple of issues and using it as a mainstay on my delta printer.

BillyBlaze, 2016-05-13:
Thanks for reporting!

Could you please provide me the versions of Chromium and TouchUI you're running? and did you perhaps installed the Jessie Light version?

diverp01, 2016-05-13:
Chromium version: 49.0.2623.108
TouchUI: 0.3.3

The version of OS is the latest Jessie (not the light version)

Octoprint is 1.2.11

Thanks

BillyBlaze, 2016-05-13:
Hmm I just did a fresh install with all the same versions and I couldn't reproduce any issue's.

Could you perhaps use a Chrome browser on your laptop and activate TouchUI there and try to reproduce the problem? if it occurs again, then please check the [developers console](http://webmasters.stackexchange.com/a/77337) to see if you have any errors.

If that does work in your Chrome browser, then I have another way of remote inspecting your touchscreen, but that is our backup plan.

diverp01, 2016-05-13:
Interesting. 

I did use Chrome on my laptop and the camera shows up. I was able to login from there as well, but not from the same interface via the touchscreen connected to the RPi. 

When I get home tonight, I'll have a look at the link you provided to see if I can get some debug output.

If that doesn't work, I may try to do a clean install as you did and see how that goes. 

Thanks for the help/ideas.

diverp01, 2016-05-18:
Hi Sorry for taking a few days to get to this. I reinstalled the Full Jessie on my Pi, configured it with MJPG-Streamer, and TouchUI. I can see the interface on my laptop fine. The web came shows up and the touchui interface looks fine. However on my 7" screen that is attached, I cannot see the webcam and I still have issues with logging in from the touchscreen. Are there issues if I access it from a laptop, then still try to access on the touchscreen at the same time? I have tried getting rid of the laptop login but the issue remains. I'll have to dig more into the logs to see if anything else shows up. Oh, I also did install SnapStream as well but it doesnt make a difference for my touchscreen and looks the same for the laptop browser.

Could this be a Chromium issue?

BillyBlaze, 2016-05-18:
SnapStream should work, since it does nothing more then loading a JPG image and reloading it every second. If your settings are correct, then it indicates that OctoPrint web interface crashed.

Do you perhaps have any non-bundled plugins enabled?

diverp01, 2016-05-18:
Do I have to remove MJPG-Streamer completely? Like I said, if I look at it through a browser on my touchscreen laptop, it will all show fine. Its just when showing locally on the touch display on my printer where the camera does not show. Octoprint is still up and running fine when I access it from the laptop. I can access settings, etc., from the touch display on my printer.

No non-bundled plugins. This was a fresh install based on the steps on the Octoprint web site/wiki for setting up an image from scratch. 

Now, I am using haproxy as is in that step-by-step so that I can access everything under port 80. Not sure if that would introduce any issues.

I'll play with it some more tonight when I get home and see if I can narrow down any more clues.

Thanks

BillyBlaze, 2016-05-18:
No, MJPG is still needed. It also provides the snapshot to snapstream.

You can enable the remote console of Chromium:

1. Edit default file of touchui

    ````
    sudo nano /etc/default/touchui
    ````
2. Change

    ````
    TOUCHUI_FILE="start_chromium.xinit"
    ````

    into

    ````
    TOUCHUI_FILE="start_chromium.xinit --remote-debugging-port=9222"
    ````

3. Save (CTR-O, Enter, CTR-X)

4. Reboot TouchUI

    ````
    sudo service touchui restart
    ````

5. Run
    ```ssh -L 0.0.0.0:9223:localhost:9222 localhost -N```

6. Open on your laptop the following url: ``http://yourpi:9223/``

diverp01, 2016-05-19:
Thank you for the information. I looked at the output and had a look at the other post about no camera availability and what you mentioned here was my issue:

"If that works in the browser then replace /webcam/?action=snapshot with http://192.168.1.250:8080/?action=snapshot and save it."

I had to do the above and the camera works now. Thank you.

As for the login, I have done a little more testing. If I am logged in on my laptop and then boot up the Raspberry Pi, the TouchUI interface will not load on the Pi and none of the buttons on the OctoPrint screen will be active. If I log out of Octoprint on my laptop, then reboot, the touchui interface will load on the RPi screen. I am still looking in to that. Have there been any issues with multiple logins out of curiosity?

Thanks

BillyBlaze, 2016-05-19:
Awesome on the snapstream :+1: 

I never heard about multiple login issues, I always develop on my PC and on a Pi, and I never experienced this issue. It does sounds like a OctoPrint issue.

Any logs of this in your remote console?

diverp01, 2016-05-25:
Hi. I didnt have any other logs from remote console that seemed to show much. I can take another capture of those this week.

After having run my system for a week or so now, the behavior seems to be this:

- TouchUI always loads fine on a PC or tablet using Chrome.
- On startup, it is random as to if TouchUI or the default Chromium Octoprint home page will load. Half the time, Touch UI will not load, half it will.
- If TouchUI does load, I often cannot login using my Octoprint login. I enter credentials, but hitting apply does nothing
- If TouchUI does not load, and the Octoprint default page loads, pressing any tab or hot area does nothing. So far, after many resets, Octoprint has loaded the default page once already logged in.
- If TouchUI does load, sometimes it will already be logged in and everything works fine.

I'm still playing with this (its a busy week). It does seem like there are a couple of things going on. It seems like Octoprint is not handling the display properly. And it also seems like there is a timing issue possibly between when the unit starts and TouchUI will load or not load.

Thanks,
Paul

BillyBlaze, 2016-05-25:
I am afraid you found a bug in OctoPrint, I also encountered once in a while. If you read the octoprint log ```~/.octoprint/logs/octoprint.log``` can see that octoprint is giving on every request a 401 or 404?

BillyBlaze, 2016-05-31:
anything?

diverp01, 2016-06-01:
Hello. Sorry about the delay. I was out of town for work and just got back today.

I am attaching the log you mention. I don't see anything like a 401 or 404 error.

[octoprint.zip](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/292636/octoprint.zip)

lciscon, 2016-06-03:
FYI I am seeing exactly the same behavior Paul described above.  To me it appears that Chrome boots too fast and it tries to load the OctoPrint page before the server is fully on-line.  So you get some random partial website.  And it's in a nonfunctional state.  Moreover if I boot without the --kiosk flag I can refresh the page manually from the Chrome menu and then everything works correctly.

diverp01, 2016-06-05:
Iciscon, I think that makes sense. Whats the best way to delay Chromium from starting so that it starts after Octoprint?

BillyBlaze, 2016-06-06:
Could you guys please enable the ```--debug``` mode in OctoPrint and share with me the OctoPrint log? This is a bug in OctoPrint that we need to get fixed, workarounds are ok, but fixing the source of the issue is the best we can do.

To enable debug run:

```
sudo nano /etc/default/octoprint
```

Change

```
DAEMON_ARGS="--port=$PORT"
```

into 

```
DAEMON_ARGS="--port=$PORT --debug"
```

Reboot and reproduce the problem, after that share the log :+1:

diverp01, 2016-06-06:
I enabled debug in Octoprint and rebooted easily to the issue. Here is a log from it. 

Thank
[OctoprintLog.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/300617/OctoprintLog.txt)
s

BillyBlaze, 2016-06-06:
Sorry I dont see any debug messages, did you upload the correct one?

Your log should start with something like:

```
2016-06-06 16:34:31,298 - octoprint.server - INFO - Starting OctoPrint 1.2.11 (master branch)
2016-06-06 16:34:31,342 - octoprint.server - DEBUG - Starting intermediary server on 0.0.0.0:5000
2016-06-06 16:34:31,348 - octoprint.server - DEBUG - Intermediary server started
```

diverp01, 2016-06-06:
The attached log is what I get when I enable debug as you mention above. Is there a different way to enable debug output?

BillyBlaze, 2016-06-06:
No, above method should work, are you sure you have saved the file correctly, I didn't say that step (CTR-O, Enter, CTR-X)

diverp01, 2016-06-07:
Sorry, uploaded an older log. Here is one with debug info.

[octoprint.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/301637/octoprint.txt)

BillyBlaze, 2016-06-08:
Update:

I was finally able to reproduce the problem today, and we (me and foosel) seem to found a culprit, it seems like chrome is loading OctoPrint from it's cache with an incorrect API key, because the OctoPrint server isn't 100% ready it will leave OctoPrint in a 'stuck state'.

We are currently trying to get a fix for the issue and hopefully it would be fixed in the next octoprint release (coming very soon).

diverp01, 2016-06-08:
That's great thanks for looking in to it. I'll look forward to the next release.

BillyBlaze, 2016-06-08:
FYI; The OctoPrint issue: https://github.com/foosel/OctoPrint/issues/1364

BillyBlaze, 2016-06-08:
This should be fixed in the latest OctoPrint 1.2.12 release

BillyBlaze, 2016-06-09:
Argh, I just had the same issue with my Pi on 1.2.12, I guess I'll reopen this issue for now...

diverp01, 2016-06-09:
I just upgraded and also had the same issue. Attached is a log from my system.

[octoprint.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/307793/octoprint.txt)

BillyBlaze, 2016-06-10:
Ok, I just finished creating a bootloader, this will ensure that TouchUI is loaded in the right order. If any of the steps are wrong an error will appear and you can retry easily by tapping the screen.

Could you try this out, if this works for you? If so, please follow below instructions:

1. Uninstall TouchUI and install TouchUI by url with the following url:

    ```
    https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip
    ```

1. Login to ssh and get the new bootloader:

    ```
    git clone https://github.com/BillyBlaze/OctoPrint-TouchUI.git ~/TouchUI-autostart && cd ~/TouchUI-autostart && git checkout autostart-dev
    ```

1. Register the new paths:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
    sudo update-rc.d touchui defaults
    ````

reboot and the new screen should appear

diverp01, 2016-06-11:
I uninstalled the older version and installed the newer version as you mentioned above with mixed results.

When I start the machine, touch ui shows as loading, then trying to load octoprint, but it cannot load the first time. Then, after pressing the screen, it tries it all again and finally says it is successful.

When it comes up, touch areas are hot now. However, I cannot get the touchui interface to fully show up. It will just show the Octoprint page. Also, the tabs will not get drawn on the Octoprint page. They are hot to touch and if I just try to touch areas where I know they usually are they will show a given tab. But, I cannot get TouchUI to load on the PI itself.

If I try to load on a small touch laptop, it shows up perfectly fine.

I am attaching the octoprint log if it helps.

Thanks.
[octoprint1.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/310092/octoprint1.txt)

BillyBlaze, 2016-06-11:
Thanks for the feedback! Hmm, do you have by any change the customization feature on? And if yes, if turn it off, does it behave the same?

diverp01, 2016-06-11:
I just looked and no, the TouchUI Customization checkbox is not enabled.

BillyBlaze, 2016-06-11:
Ok, can you get me another log? If yes,

Run
````
nano ~/TouchUI-autostart/start_chromium.xinit
````

and add the parameter:
````
 --enable-logging
```` 

Save (CTR-O, Enter, CTR-X)

Restart everything and share the log ```/home/pi/.config/chromium/chrome_debug.log```

diverp01, 2016-06-11:
Here is the log. 
[chrome_debug.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/310345/chrome_debug.txt)

BillyBlaze, 2016-06-11:
Thanks that is really usefull!

Strange errors:
```
[1005:1005:0611/160309:ERROR:gl_surface.cc(56)] Requested GL implementation is not available.
[1005:1005:0611/160309:ERROR:gpu_child_thread.cc(260)] Exiting GPU process due to errors during initialization
[797:981:0611/160310:ERROR:browser_gpu_channel_host_factory.cc(133)] Failed to launch GPU process.
```

Could you edit ``~/TouchUI-autostart/start_chromium.xinit`` and remove ``--use-gl=Desktop``

Can you then reboot TouchUI and share the log again?

diverp01, 2016-06-11:
Ok. Looking promising. I have restarted several times and the TouchUI interface comes up each time.

The "only" issue is that the webcam has been lost. Its looking much better though.

Here is the log:
[chrome_debug_2.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/310366/chrome_debug_2.txt)

BillyBlaze, 2016-06-11:
Yes, bingo!

Have you installed [SnapStream](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Troubleshooting#webcam-image-not-working)?

diverp01, 2016-06-11:
I did. I am using an RPi 3 and went from using ethernet to wifi. I had to go into the Snapshot URL setting and change the IP from my wired IP to the IP for ethernet and it showed up. Peculiar dependency, but it works.

Everything else looks perfect on the display and on a tablet/laptop. Thanks very much for debugging this. It looks great.

One question, for future installs (I have several printers and will be setting up each), will the install for TouchUI integrate these changes or will there be a few one off items I will need to change as in this thread?

Thanks again, I appreciate it.

diverp01, 2016-06-11:
Sorry, I had to change the Snapshot URL from the wired IP to the wireless IP. I mistyped.

BillyBlaze, 2016-06-11:
Awesome! I'm glad we finally tackled the problems :+1: 

I have updated the autostart branch already, so next time if you install TouchUI it doesn't have the --use-gl

We do need to change the development codes once I release the files. I will update you with some instructions during release. (if we don't we will get issue's next release :))

diverp01, 2016-06-11:
Does the autostart branch include the bootloader? Prior to next release, will installing the autostart branch be all I would need to do? (then uninstall and reinstall upon next release)?

BillyBlaze, 2016-06-11:
Yeah some things will change, I already moved the autostart-dev branch to a [new repository](https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/).

And I already made some new changes to the bootloader. So when I release 0.3.4 we need to uninstall TouchUI, remove all old files, clone from new repository and then install TouchUI.

diverp01, 2016-06-11:
Sounds good. Thanks for the Plugin and help.

BillyBlaze, 2016-06-19:
Ok, 0.3.4 has just been released. If you installed the maintenance branch then please follow these instructions to prevent TouchUI not working in the future:

1. Remove TouchUI and install the latest 0.3.4 version through the plugin manager

1. Make sure you're using the right bootloader repository and are up-to-date:

    ```
    rm -rf ~/TouchUI-autostart/
    git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/
    ```

1. Register the new paths:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
    sudo update-rc.d touchui defaults
    ````

reboot and it should work :)

closed 2016-06-19 by BillyBlaze

diverp01, 2016-06-26:
Hi there. I just got home from a trip and attempted to remove and reinstall with new bootloader, etc. TouchUI seems to work fine with browser on laptop, however, on my 7" display it continues to say Connecting to TouchUI and the counter increases. I have verified that the old directory for the bootloader is removed. I then updated the new boot files and then updated both Octoprint as well as Touchui to the latest.

BillyBlaze, 2016-06-27:
ok, I just found an issue reported by @EngineeredEntropy and also a fix.

Could you try this:

1. Update TouchUI bootloader:

    ```
    cd ~/TouchUI-autostart/
    git pull
    ```

diverp01, 2016-06-28:
That worked, thanks.

chrkan, 2016-07-16:
hi,
after bioting the pi or  restarting the touch ui service i get 10 sec a black screen after that it is going back to the command line.
when i change fb1 to fb0 nothing would happened

[   617.311] 
X.Org X Server 1.17.2
Release Date: 2015-06-16
[   617.312] X Protocol Version 11, Revision 0
[   617.312] Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
[   617.313] Current Operating System: Linux octopi 3.12.34+ #1 PREEMPT Mon Dec 8 00:05:41 CET 2014 armv6l
[   617.313] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0xe bcm2708.serial=0xc9b3d20f smsc95xx.macaddr=B8:27:EB:B3:D2:0F bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait fbtft_device.custom fbtft_device.name=fb_ili9340 fbtft_device.gpios=dc:22,reset:27 fbtft_device.bgr=1 fbtft_device.speed=48000000 fbcon=map:10 fbcon=font:VGA8x8 dma.dmachans=0x7f35 console=tty1 consoleblank=0 fbtft_device.fps=50 fbtft_device.rotate=270 logo.nologo
[   617.314] Build Date: 11 January 2016  01:32:44PM
[   617.314] xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support) 
[   617.315] Current version of pixman: 0.33.3
[   617.315] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[   617.315] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   617.317] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Jul 16 11:17:39 2016
[   617.318] (==) Using config directory: "/etc/X11/xorg.conf.d"
[   617.319] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   617.323] (==) No Layout section.  Using the first Screen section.
[   617.323] (==) No screen section available. Using defaults.
[   617.324] (**) |-->Screen "Default Screen Section" (0)
[   617.324] (**) |   |-->Monitor "<default monitor>"
[   617.328] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[   617.328] (**) |   |-->Device "touchscreen"
[   617.328] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[   617.328] (==) Automatically adding devices
[   617.329] (==) Automatically enabling devices
[   617.329] (==) Automatically adding GPU devices
[   617.329] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[   617.329] 	Entry deleted from font path.
[   617.330] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   617.330] 	Entry deleted from font path.
[   617.330] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   617.330] 	Entry deleted from font path.
[   617.331] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   617.331] 	Entry deleted from font path.
[   617.331] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   617.331] 	Entry deleted from font path.
[   617.332] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   617.332] 	Entry deleted from font path.
[   617.332] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   617.332] 	Entry deleted from font path.
[   617.332] (==) FontPath set to:
	built-ins
[   617.333] (==) ModulePath set to "/usr/lib/xorg/modules"
[   617.333] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[   617.333] (II) Loader magic: 0xb6f4df00
[   617.333] (II) Module ABI versions:
[   617.333] 	X.Org ANSI C Emulation: 0.4
[   617.334] 	X.Org Video Driver: 19.0
[   617.334] 	X.Org XInput driver : 21.0
[   617.334] 	X.Org Server Extension : 9.0
[   617.336] (II) no primary bus or device found
[   617.336] (II) LoadModule: "glx"
[   617.339] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[   617.412] (II) Module glx: vendor="X.Org Foundation"
[   617.412] 	compiled for 1.17.2, module version = 1.0.0
[   617.412] 	ABI class: X.Org Server Extension, version 9.0
[   617.412] (==) AIGLX enabled
[   617.413] (II) LoadModule: "fbdev"
[   617.414] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[   617.417] (II) Module fbdev: vendor="X.Org Foundation"
[   617.417] 	compiled for 1.17.2, module version = 0.4.4
[... 48 lines cut ...]
[   617.595] (EE) GLX: could not load software renderer
[   617.595] (II) GLX: no usable GL providers found for screen 0
[   618.031] (II) config/udev: Adding input device Webcam C170 (/dev/input/event0)
[   618.032] (**) Webcam C170: Applying InputClass "evdev keyboard catchall"
[   618.032] (II) LoadModule: "evdev"
[   618.034] (II) Loading /usr/lib/xorg/modules/input/evdev_drv.so
[   618.042] (II) Module evdev: vendor="X.Org Foundation"
[   618.042] 	compiled for 1.17.2, module version = 2.9.2
[   618.042] 	Module class: X.Org XInput Driver
[   618.042] 	ABI class: X.Org XInput driver, version 21.0
[   618.043] (II) Using input driver 'evdev' for 'Webcam C170'
[   618.043] (**) Webcam C170: always reports core events
[   618.044] (**) evdev: Webcam C170: Device: "/dev/input/event0"
[   618.045] (--) evdev: Webcam C170: Vendor 0x46d Product 0x82b
[   618.046] (--) evdev: Webcam C170: Found keys
[   618.046] (II) evdev: Webcam C170: Configuring as keyboard
[... cut ...] tform/bcm2708_usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input0/event0"
[   618.047] (II) XINPUT: Adding extended input device "Webcam C170" (type: KEYBOARD, id 6)
[   618.047] (**) Option "xkb_rules" "evdev"
[   618.047] (**) Option "xkb_model" "pc105"
[   618.048] (**) Option "xkb_layout" "gb"
[   618.479] (EE) FBDEV(0): FBIOBLANK: Invalid argument
[   637.648] (II) evdev: Webcam C170: Close
[   637.649] (II) UnloadModule: "evdev"
[   637.663] (II) Server terminated successfully (0). Closing log file.

BillyBlaze, 2016-07-16:
Hmm yeah fb1 seems to be correct.

However I think your problems is this error:

```
[   617.594] (EE) AIGLX error: dlopen of /usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so: cannot open shared object file: No such file or directory)
```

This is what the line says in my log:

```
[    12.522] (II) AIGLX: Loaded and initialized swrast
```

What OctoPi version are you using?

chrkan, 2016-07-16:
I have the 1.2.13 running, updates today

BillyBlaze, 2016-07-16:
No, not OctoPrint, I mean what version of OctoPi did you install (the OS image you installed onto the microSD card)

chrkan, 2016-07-16:
oh sorry it was this image: 2016-03-18-octopi-jessie-lite-0.13.0

BillyBlaze, 2016-07-16:
Hmm could you download the latest OctoPI and try again? I think this version was a nightly build right?

If this issue persisted in later versions I should have got some more issue reports of it.

chrkan, 2016-07-16:
i have download it today directly  hier: https://octopi.octoprint.org
is there andy difference between the download this morning?

chrkan, 2016-07-24:
i hie now install an older version of octopi with debian Wezzy but it doesn't work

johnprosek, 2017-10-05:
Hey Billy,
just installed a new version of Octpi and touchUI.  I am getting the same issue as described above.  Nothing in the control tab, no video (totally not important), and can't login (Login failed Unknown user and password)  user pi and my password, can login fine from shh!

The latest version of everything!  Downloaded and installed tonight...

Any help?

dikodahan, 2018-08-01:
Hey guys,
I know this is an old thread, however I am experiencing the exact same issue here.
running the latest 1.3.9 OctoPrint. Everything is fully updated to latest as of the date of this post in both the OS and OctoPrint.

Any ideas why this is a lingering problem?