BillyBlaze/OctoPrint-TouchUI #43: Unboot from browser destructions missing? [Request]

gddeen, 2016-05-19:
It would be nice to try this stuff out on working systems, or systems that can get 'reworking.'

I like several options:

1 - Copy old pi sd card image onto a new card and erase it if anything fails.

or

2 - set video drivers back, repair xinit, uninstall packages, repair configuration files.

I think it gives people a better understanding of how the stuff works. But, if a step IS
missed it requires a LOT more skill than the skill of following instructions.

BillyBlaze, 2016-05-19:
Hi gddeen,

You can disable boot to browser by editing ```/etc/default/touchui``` and by changing ``START=yes`` to ``START=no``

But you're absolutely right, that I should documentate that on my Wiki! Does this solve the request for you or would you like to completely remove it?

gddeen, 2016-05-20:
Can I test that first to see what artifacts are laying around? It would drop back to whatever it was
before a console device to a getty terminal, or the graphical desktop with the login window?
Wow. I AM impressed.

Stripping out the installed products isn't a big deal. Getting the system to boot up accepting
a console hdmi video and keyboard is...

closed 2016-05-25 by gddeen