BillyBlaze/OctoPrint-TouchUI #44: Can't get TouchUI to work

Kev7274, 2016-05-24:
I have tried numerous times to get it to work,but I cant figure out as to why.

I have Octoprint installed and it works fine,camera,etc. I see the options for TouchUI.

I have followed this guide.... https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light),but I may be doing something wrong. 

I am using the PI 7" Touchscreen. Not sure as to what I need to do on the following steps?

#4 Does the Touchscreen not already have the lastest drivers? If not,how do I go about updating them?
#6 When I run it and select anybody,I get this "start and stop actions are no longer supported; falling back to defaults"
#8 Do I just copy and paste? or is there a step I'm missing to make it auto boot?

I did get it to boot to debian 8 one time somehow,but I never got the touchUI interface.Was that even a correct install?

Thanks in advance,

BillyBlaze, 2016-05-24:
Thanks for reporting, I have found some issue with the steps and can make them better. I am currently doing a new install to make sure these steps are correct.

Kev7274, 2016-05-24:
Excellent!  I'll wait to see your new instructions before I try another install.

BillyBlaze, 2016-05-24:
Hmm, seems like I cannot reproduce the problem. Due to the driver installing the required files.

Could you try running the following command?

```
sudo apt-get install lxde-core
```

This would install the required X11 config files that are missing in Jessie Lite. Just reboot after it's done.

Also to give an anwser to your previous points
4) Yeah, this is wrong and should be changed in the instructions.
6) You can ignore that warning.
8) The last line will register the copied init file as a debian service, which will auto boot with xinit.

Kev7274, 2016-05-24:
I'm going to do a fresh install. Are there a certain spot I should run "sudo apt-get install lxde-core"

Kev7274, 2016-05-24:
Did another install and included your command and it boots to my pi command screen and I enter my log in info. It then waits for another command,no gui? If i try to boot to desktop,it tells me to install lightdm?

BillyBlaze, 2016-05-24:
No you shouldn't boot to desktop, you should just boot to the console. You also don't have to login, TouchUI should start there.

If you run the following command

```xinit /home/pi/OctoPrint-TouchUI-autostart/start_chromium.xinit```

does that boot up TouchUI? and if no, what is the output?

Kev7274, 2016-05-24:
TouchUI doesn't load. 

xinit /home/pi/OctoPrint-TouchUI-autostart/start_chromium.xinit

X.Org X Server 1.17.2
Release Date: 2015-06-16
X Protocol Version 11, Revision 0
Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
Current Operating System: Linux octopi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l
Kernel command line: 8250.nr_uarts=1 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2709.boardrev=0xa22082 bcm2709.serial=0x3c711c0b smsc95xx.macaddr=B8:27:EB:71:1C:0B bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 11 January 2016  01:32:44PM
xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue May 24 20:33:57 2016
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) no screens found(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

BillyBlaze, 2016-05-25:
Can you edit the following file
````
sudo nano /etc/cmdline.txt
````

and add at the end of the line the following:
````
fbcon=map:10 fbcon=font:ProFont6x11
````

Save (CTR-O, Enter, CTR-X) 

Reboot and manually run the xinit command in previous comment.

Kev7274, 2016-05-25:
I get the same error as my previous post. 

"When you say add at the end of the line the following" fbcon=map:10 fbcon=font:ProFont6x11,there is nothing there until I paste this,is that right. Wasn't sure since you mentioned at end if something should have popped up.

BillyBlaze, 2016-05-25:
Sorry, I made a brainfart...

The command should be:
````
sudo nano /boot/cmdline.txt
````

Kev7274, 2016-05-25:
Still boots to command line and stops.

BillyBlaze, 2016-05-25:
Okay too bad.

Could you provide me the contents of the file ```/var/log/Xorg.0.log```
This should give some more info about what modules and driver it's trying to load (or not)

Kev7274, 2016-05-25:
Is that a command? When  try to run it,I get this....-bash: /var/log/Xorg.0.log: Permission denied

BillyBlaze, 2016-05-25:
no its a file use ```cat /var/log/Xorg.0.log``` to output the contents.

Kev7274, 2016-05-25:
cat /var/log/Xorg.0.log
[     7.461]
X.Org X Server 1.17.2
Release Date: 2015-06-16
[     7.461] X Protocol Version 11, Revision 0
[     7.461] Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
[     7.461] Current Operating System: Linux octopi 4.4.9-v7+ #884 SMP Fri May 6 17:28:59 BST 2016 armv7l
[     7.462] Kernel command line: 8250.nr_uarts=0 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2709.boardrev=0xa22082 bcm2709.serial=0x3c711c0b smsc95xx.macaddr=B8:27:EB:71:1C:0B bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait  fbcon=map:10 fbcon=font:ProFont6x11
[     7.462] Build Date: 11 January 2016  01:32:44PM
[     7.462] xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support)
[     7.462] Current version of pixman: 0.33.3
[     7.462]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[     7.462] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     7.463] (==) Log file: "/var/log/Xorg.0.log", Time: Wed May 25 01:27:11 2016
[     7.468] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[     7.480] (==) No Layout section.  Using the first Screen section.
[     7.480] (==) No screen section available. Using defaults.
[     7.480] (**) |-->Screen "Default Screen Section" (0)
[     7.480] (**) |   |-->Monitor "<default monitor>"
[     7.490] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[     7.490] (**) |   |-->Device "touchscreen"
[     7.490] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[     7.490] (==) Automatically adding devices
[     7.490] (==) Automatically enabling devices
[     7.490] (==) Automatically adding GPU devices
[     7.503] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[     7.503]    Entry deleted from font path.
[     7.503] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[     7.503]    Entry deleted from font path.
[     7.503] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[     7.503]    Entry deleted from font path.
[     7.503] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[     7.503]    Entry deleted from font path.
[     7.503] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[     7.503]    Entry deleted from font path.
[     7.503] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[     7.504]    Entry deleted from font path.
[     7.504] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[     7.504]    Entry deleted from font path.
[     7.504] (==) FontPath set to:
        built-ins
[     7.504] (==) ModulePath set to "/usr/lib/xorg/modules"
[     7.504] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[     7.504] (II) Loader magic: 0x54d2af00
[     7.504] (II) Module ABI versions:
[     7.504]    X.Org ANSI C Emulation: 0.4
[     7.504]    X.Org Video Driver: 19.0
[     7.504]    X.Org XInput driver : 21.0
[     7.504]    X.Org Server Extension : 9.0
[     7.505] (II) no primary bus or device found
[     7.507] (II) LoadModule: "glx"
[     7.511] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     7.624] (II) Module glx: vendor="X.Org Foundation"
[     7.624]    compiled for 1.17.2, module version = 1.0.0
[     7.624]    ABI class: X.Org Server Extension, version 9.0
[     7.624] (==) AIGLX enabled
[     7.624] (II) LoadModule: "fbdev"
[     7.630] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[     7.635] (II) Module fbdev: vendor="X.Org Foundation"
[     7.635]    compiled for 1.17.2, module version = 0.4.4
[     7.636]    Module class: X.Org Video Driver
[     7.636]    ABI class: X.Org Video Driver, version 19.0
[     7.636] (II) FBDEV: driver for framebuffer: fbdev
[     7.636] (--) using VT number 2
[... 84 lines cut ...]
[     7.635]    compiled for 1.17.2, module version = 0.4.4
[     7.636]    Module class: X.Org Video Driver
[     7.636]    ABI class: X.Org Video Driver, version 19.0
[     7.636] (II) FBDEV: driver for framebuffer: fbdev
[     7.636] (--) using VT number 2

[     7.640] (WW) Falling back to old probe method for fbdev
[     7.640] (II) Loading sub module "fbdevhw"
[     7.640] (II) LoadModule: "fbdevhw"
[     7.641] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[     7.647] (II) Module fbdevhw: vendor="X.Org Foundation"
[     7.647]    compiled for 1.17.2, module version = 0.0.2
[     7.647]    ABI class: X.Org Video Driver, version 19.0
[     7.647] (EE) open /dev/fb1: No such file or directory
[     7.647] (EE) No devices detected.
[     7.647] (EE)
Fatal server error:
[     7.648] (EE) no screens found(EE)
[     7.648] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[     7.648] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[     7.648] (EE)
[     7.656] (EE) Server terminated with error (1). Closing log file.

BillyBlaze, 2016-05-25:
Thanks I spotted the error:  ```open /dev/fb1: No such file or directory```

What happens if we change that to zero.

Edit the file:
```
sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf
```

and change ```/fb1``` to ```/fb0```

Save and exit (CTR-O, enter, CTR-X)

Kev7274, 2016-05-25:
BINGO!!!! boots up now. I really appreciate the help.  

Was the last step the main problem? I'm going to make me a cheat sheet just in case I need to install it again. 

Is it safe to say that If I change /fb1 to /fb0 in step #5 of the guide that it would have worked?

BillyBlaze, 2016-05-25:
Awesome :D No, I will update this in the Wiki. But I am still a bit baffled. Your touchscreen works on fb0?? That should preserved for the HDMI port, I assumed, is your touchscreen parhaps connected with HDMI?

Kev7274, 2016-05-25:
No,no hdmi connection.

BillyBlaze, 2016-05-25:
Hmm no, after reading some documentation of the Official Raspbery Pi Touchscreen, it's get add'd to fb0 and not fb1 like other touchscreen modules.

Okay, lesson learned, lets update the wiki :+1:

closed 2016-05-25 by BillyBlaze

Kev7274, 2016-05-25:
Great