BillyBlaze/OctoPrint-TouchUI #46: Cannot Scroll on RPI 7" Touchscreen?

lciscon, 2016-05-28:
I just built an image on a RPI3 with an official RPI touchscreen (7") following the instructions "Boot to Browser (OctoPi or Jessie Light)".  Everything works great except that once it boots up in the TouchUI interface I cannot scroll vertically with my finger.  All of the other touch functions work correctly.  If I turn off the TouchUI interface the standard OctoPrint UI scrolls around correctly.  Any ideas?

Note: I have previously built versions using the older instructions (e.g. Kweb) and it works fine with the same hardware.

BTW great job on the software!!

-larry

BillyBlaze, 2016-05-28:
Thanks for reporting, it sounds like the touch emulator is broken, but it also can be that touchui is half loaded due to some other error. However we can check this with the external console. To enable and view the external console follow this steps:

1. Edit the Chromium startup file

    ````
    nano /home/pi/OctoPrint-TouchUI-autostart/start_chromium.xinit
    ````
2. Add the following before ``http://localhost``

    ````
     --remote-debugging-port=9222
    ````

3. Save (CTR-O, Enter, CTR-X)

4. Reboot TouchUI

    ````
    sudo service touchui restart
    ````

5. Run
    ```ssh -L 0.0.0.0:9223:localhost:9222 localhost -N```

6. Open on your computer the following url: ``http://yourpi:9223/``

lciscon, 2016-05-28:
Ok I did these steps but there's an issue (and a workaround).  When you switch on remote debugging it appears you have to flush the browser cache otherwise the page doesn't reload properly (it is actually frozen and doesn't respond).  And I don't know how to flush the cache manually when you are in Kiosk mode (perhaps you can delete a subdirectory?).  

Instead I just removed the --kiosk flag from the start_chromium.xinit file and then I can flush the queue and access the debugging tools and console directly on the touch screen.  Works great.  And BTW the non-scrolling behavior is still the same in the TouchUI portion of the screen.

So anyways now I have debugging console up-and-running. But I don't see anything specifically related to the touch scrolling behavior.  What should I be looking for?

lciscon, 2016-05-28:
To answer my own question you can manually delete the cache by removing

lciscon, 2016-05-28:
these files:    ~/.cache/google-chrome/[profile]/Cache/

lciscon, 2016-05-28:
Ok it really seems like the TouchUI page size is not correct somehow.  I can't even scroll it on the 7" screen when I connect a regular mouse.  All of the other touch and mouse actions work correctly that I can see.

BillyBlaze, 2016-05-28:
Hmm very strange, can you perhaps make a photo or screenshot of it? Also if you cannot scroll does there not appear any error in the external developer tool?

Also could you confirm me some basic info? Chrome 50? OctoPi 0.13? OctoPrint 1.2.11? TouchUI 0.3.3? and do you have any other plugins installed beside the bundled ones?

lciscon, 2016-05-28:
Here are the versions:  Chrome 50, OctoPi 0.13, OctoPrint 1.2.11, TouchUI 0.3.3.  I also have Custom Control Editor 0.2.1, Email Notifier 0.1.1 and SnapStream 0.2.3 installed.

There's not really anything unusual to show on the screen. On the 7" RPI touch display the bottom of the pages are off the screen and there's no way to get to them.  Under Kweb I was able to drag my finger up vertically to scroll the page up and see the rest.  This is what the screen looks like:

[image]

BillyBlaze, 2016-05-31:
I am absolutely stunned by this issue, I have re-flashed my sd a couple of times now but I am just unable to reproduce this problem.

Could you help me out with some basic questions, if yes, open the external console again and type after the arrow the following commands, just copy paste the feedback you get :+1: 

1. I want to know that no-pointer is not active:
    Run: ``$('.no-pointer').length``

1. I want to know the state of the touch emulator
    1. Run: ``TouchUI.prototype.scroll.iScrolls.body.wrapperHeight``
    1. Run: ``TouchUI.prototype.scroll.iScrolls.body.scrollerHeight``
    1. Run: ``TouchUI.prototype.scroll.iScrolls.body.maxScrollY``
    1. Run: ``TouchUI.prototype.scroll.iScrolls.body.enabled``
    1. Run: ``TouchUI.prototype.scroll.iScrolls.body.directionLocked``

1. I want to know what happens when you refresh the touch emulator
    Run: ``TouchUI.prototype.scroll.iScrolls.body.refresh()`` (can you scroll now?)

lciscon, 2016-06-01:
No problem.  Keep in mind this may very well be a stupid user issue, but I just don't see what I'm doing differently from the standard install instructions.  Here is the information:

no-pointer length: 0
wrapperHeight:  593
scrollerHeight: 811
maxScrollY: -218
enabled: true

Very oddly the directionLocked and refresh() method are undefined??? 

I'm happy to run additional tests to help diagnose the issue.

-larry

lciscon, 2016-06-01:
Actually refresh is defined but somehow I can't call it.

lciscon, 2016-06-01:
One other clue: If I run Chrome without the --kiosk flag, go into the settings and change the font size (for example from medium to large or visa versa) then suddenly the scrolling works correctly.  However once I do a full reboot it goes back to not working.

BillyBlaze, 2016-06-08:
Thanks for all the information, this helps alot. I still haven't found a culprit but this does narrow it down.

It looks like the touch emulator initialized perfectly, it's not blocked by no-pointer and is enabled. If you change the text the window will trigger a resize. I just upgraded the touch emulator to a newer version.

Could you give this a try?

If yes, please uninstall TouchUI and then install TouchUI with this link:

```
https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip
```

BillyBlaze, 2016-06-08:
Also could you be try updating to the latest OctoPrint (1.2.12) and see if the problem still persists?

lciscon, 2016-06-08:
Well this is interesting.  I did those updates and it did not seem to fix the issue.  However on a sudden impulse I went ahead and removed all Chrome settings by deleting the ~/.config/chromium subdirectory and rebooted.  That fixed it.  Now the scrolling works fine.  

Edit: Nope spoke too soon.  I rebooted the machine and the non-scrolling came back.  However this gave me an idea.  I added a line "rm -rf ~/.config/chromium" at the beginning of the start_chromium.xinit script.  This works.  Every time I boot it scrolls correctly now.  However obviously you need to re-login each time, so it's really not a viable long-term solution.

BTW I also added a line "Sleep 20" to the beginning of the start_chromium.xinit file to help delay the startup of the browser till after the OctoPrint server starts. This combined with the 1.2.12 fix seems to make the boot-up work reliably for the touchscreen.

BillyBlaze, 2016-06-10:
Ok, I just finished creating a bootloader, this will ensure that TouchUI is loaded in the right order. If any of the steps are wrong an error will appear and you can retry easily by tapping the screen.

Could you try this out, if this works for you? If so, please follow below instructions:

1. Uninstall TouchUI and install TouchUI by url with the following url:

    ```
    https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip
    ```

1. Login to ssh and get the new bootloader:

    ```
    git clone https://github.com/BillyBlaze/OctoPrint-TouchUI.git ~/TouchUI-autostart && cd ~/TouchUI-autostart && git checkout autostart-dev
    ```

1. Register the new paths:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
    sudo update-rc.d touchui defaults
    ````

reboot and the new screen should appear

lciscon, 2016-06-11:
Thank you for going through all of the effort to create the bootloader.  I think it is a great improvement.  It improves the startup look and makes it much more reliable.  I think this resolves the startup race condition.

Unfortunately it did not fix the scrolling issue.  I still have to add the "rm -rf ~/.config/chromium" to the beginning of the start_chromium.xinit file for the scrolling to work correctly.

BillyBlaze, 2016-06-11:
Your welcome :+1: Yeah it really enhance loading of the interface. I am quiet pleased with it.

But argh, I am absolutely baffled by this scrolling issue, but I think I might have a solution. Could you 'reset' your workarounds and uninstall TouchUI and reinstall it by url with??

```
https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip
```

lciscon, 2016-06-11:
Nope sorry.  That didn't make a difference that I can tell.  Are you using a third-party package for scrolling? This really looks like a weird interaction between the browser and whatever is controlling the scrolling and/or screen extent calculations.

lciscon, 2016-06-11:
I've narrowed down the workaround to just deleting a single file:

rm -f ~/.config/chromium/First\ Run

This resolves the issue and yet keeps the user settings intact.  Now it stays logged in between reboots.  

Oddly this is a zero-length file. My best guess is that when this file is missing Chrome pops up the "You can search from here with Google" popup and this fixes whatever screen extent calculation is incorrect without it.

BillyBlaze, 2016-06-11:
wow thanks for finding out the root of that problem. I have the identical file here.

Can you edit chromium.xinit (the new one in ~/TouchUI-autostart)

and add the parameter: ```--no-first-run```

Does this also solve this issue?

lciscon, 2016-06-11:
No that doesn't fix it.  I don't think the first run is the issue itself. Rather I think what's happening is that something is trying to read the screen extents before anything is on the screen and chromium is returning (0,0,0,0).  Deleting that file just causes something to be on the screen when the call is made and thus the screen extents are valid.  The fix should entail double checking the return value from the screen extents call and re-running it until it returns a correct value.

BillyBlaze, 2016-06-11:
Hmm, but the weird part is that iScroll (the library in use) is responding [with](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/46#issuecomment-222856946):
> wrapperHeight: 593
> scrollerHeight: 811

so it is aware of the window height and scrollable area, I think there is something that will just not trigger the mousedown event until you someway trigger a window resize.

I have just updated all the parameters of Chromium on the autostart-dev branch including a logging parameter, can you get this? if yes, follow below instructions:

1. Uninstall TouchUI and install TouchUI by url with the following url:

    ```
    https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip
    ```

1. Login to ssh and get the new bootloader files:

    ```
    cd ~/TouchUI-autostart
    git pull
    ```
    You might need todo ```git reset --hard``` to get the latest changes

1. Register the new paths:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
    sudo update-rc.d touchui defaults
    ````

Then after you restart TouchUI there should be a file inside ````/home/pi/.config/chromium``` called ```chrome_debug.log``` with alot more information

lciscon, 2016-06-11:
Ok here is the log: 

[chrome_debug.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/310411/chrome_debug.txt)

lciscon, 2016-06-11:
FYI with these latest changes the screen seems to be stuck.  Nothing is responding and even the chromium popup is not going away.  Maybe because of the debugging?

BillyBlaze, 2016-06-11:
Noo, it shouldn't... Anything new in chrome_debug.txt??

Also the log looks perfectly fine, this is getting stranger and stranger....
Could you open the [remote console](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/46#issuecomment-222315131)

and then run this time:
```
window.dispatchEvent(new Event('resize', {}))
```

does that enable scrolling?

BillyBlaze, 2016-06-11:
Oww and for fixing not responding could you retry this step:

1. Login to ssh and get the new bootloader files:

    ```
    cd ~/TouchUI-autostart
    git pull
    ```

BillyBlaze, 2016-06-11:
Sorry. I am sleepy, I already switched to a new repository for the autostart files.

Can you run this instead?
```
rm -rf ~/TouchUI-autostart/
git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ./TouchUI-autostart
```

lciscon, 2016-06-11:
Ok the latest updates fixes the stuckness.  But the lack of scrolling is the same.  Here is the latest log file:
[chrome_debug.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/310461/chrome_debug.txt)

lciscon, 2016-06-11:
And running window.dispatchEvent(new Event('resize', {})) from the remote console doesn't make a difference.

BillyBlaze, 2016-06-11:
Okay thanks for all the patients. Everything still looks ok, I really think there is nothing wrong with TouchUI but that this is something with Chrome. For example; ```--disabled``` didn't do anything for me, but for you it was really disabled..

If you add ```--incognito``` to ```chromium.xinit``` will scrolling then work?

lciscon, 2016-06-11:
Nope. No change.  But for now my workaround is working fine so don't sweat it too much.

BillyBlaze, 2016-06-12:
hehe yeah but I wanna know what the source is of this problem :smile:

rkoens, 2016-06-12:
I am seeing the same issue with the scrolling. I see loads of test versions in this report so I figured id ask what files to use for testing. Wouldn't want to be using an old version and mess up your trouble shooting ;)

btw, I am only seeing this in chromium, when I start kweb scrolling works fine. It just wont let me login from kweb -_-

BillyBlaze, 2016-06-12:
Can you edit ```~/OctoPrint-TouchUI-autostart/start_chromium.xinit``` and add ```--enable-logging --v=100``` to the file (Save with CTR-O, enter, CTR-X)

Reboot TouchUI ```sudo service touchui restart```

And then share ```/home/pi/.config/chromium/chrome_debug.log```

rkoens, 2016-06-12:
Debug log attached as per your request.
[chrome_debug.zip](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/310740/chrome_debug.zip)

BillyBlaze, 2016-06-12:
@lciscon 

We found the issue, can you add ``` --touch-events=enabled``` to your chromium.xinit? and see if you now can scroll?

lciscon, 2016-06-12:
Eureka! I think that fixed it!  Well done!

enrstudio, 2016-06-13:
Hi, the scroll work fine now but I have another small problem, the touch click function work perfectly on all the surface, exluded the menu bar, so i can't change window, can you help me?

BillyBlaze, 2016-06-13:
It seems like this is a X11 and/or Chrome 50 bug, adding ```--touch-event=enabled``` is sadly not the solution.

If you remove ```--touch-events=enabled``` and restart TouchUI, then if you click on the screen after X11 started but before Chrome started (the moment the touchscreen turns black) then you need to press the screen. If you did that before Chrome started, then TouchUI will work perfectly fine.

Can somebody confirm this behavior?

@enrstudio; we need another fix for this; I will update you once I have found a real solution.

rkoens, 2016-06-13:
First of all i cannot replicate enrstudio's issue. When I set --touch-events=enabled my menu bar works fine. So maybe his issue is related to previous troubleshooting steps?

As for the touching of the screen before chrome starts, I cannot seem to find the right time to press the screen. I tried tapping it from the moment I run the xinit script till the moment I get the octoprint interface but it still does not detect touch and TouchUI.prototype.settings.hasTouch remains false.

rkoens, 2016-06-13:
Some more testing shows that Chrome does appear to detect the touch input correctly when using it without touch-events flag. It is just the TouchUI interface that seems to break. When I open the normal octoprint interface I can scroll ok.

BillyBlaze, 2016-06-13:
Thanks for the update, this helps alot, I think I found the bug within TouchUI.

Can you try if this works (remember to remove ```--touch-events``` before testing :+1: )

1. Uninstall TouchUI and install TouchUI by url with the following url:

    ```
    https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip
    ```

lciscon, 2016-06-13:
I am not able to replicate enrstudio's issue either.  The --touch-events=enabled makes everything work correctly for me.  So far I have tried it on two almost-identical RPI3+RPI Touchcreen machines using the latest builds.  I will try the touch-before-chrome-starts test later today.  

The menu issue may be a separate chrome state issue.  You might want to try deleting the subdirectory ~/.config/chrome and then reboot.

FYI rkoens description of the issue sounds identical to my behavior before all of this started, i.e. touching and scrolling works correctly on regular Octoprint and touching (but not scrolling) works correctly in TouchUI.

BillyBlaze, 2016-06-13:
Thanks both for the feedback. I can conclude that you will not have this issue on a Official Raspberry Pi Touchscreen. I hope my bugfix will resolve your issue without adding the parameter.

However my el-cheapo touchscreen does have this problem, and when touch events are enabled I have the same issue as @enrstudio

rkoens, 2016-06-13:
@BillyBlaze the patch does not seem to fix it for me.

BillyBlaze, 2016-06-13:
sorry, I forgot something (e.g. made a brainfart).. I need to run gulp........ Will do that once I'm at home...

rkoens, 2016-06-13:
rgr, ill be on irc tonight if you want to do some quick troubleshooting.

BillyBlaze, 2016-06-13:
Can you retry? I have editted the file manually for now :smile:

enrstudio, 2016-06-17:
I have installed the new version of touchUI and removed --touch-events but the scroll doesn't work, so i've tryed to reinsert it and now, at first look, everything works fine.

Stratrascal, 2016-06-17:
I just wanted to chime in that I am having the exact same problem. If you
would like I could send the info as well

BillyBlaze, 2016-06-17:
@enrstudio, ok so that "Revert bugfix" fixed your problem of not able to click on the tabbar? and yes the graph should resize to +/- 100% of the window size, but it def. shouldn't look like its cut-off.

@Stratrascal what is your problem atm? did adding --touch-event=enabled not fix the issue?

rkoens, 2016-06-18:
@BillyBlaze The latest version of the autostart script seems to work. As you suspected i forgot i had downgraded back to the stable release of touchui

BillyBlaze, 2016-06-18:
Ok. good! I want to release 0.3.4 as soon as possible, therefor @Stratrascal and @enrstudio I need more info if the latest maintenance branch and bootloader is not working for you?

This steps are:

1. Uninstall TouchUI and install TouchUI by url with the following url:

    ```
    https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip
    ```

1. Login to ssh and get the new bootloader files:

    ```
    git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/
    ```
    if you already have the folder then delete it with ``rm -rf ~/TouchUI-autostart/`` and try again

1. Register the new paths:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
    sudo update-rc.d touchui defaults
    ````

Then after you restart TouchUI there should be a file inside ````/home/pi/.config/chromium``` called ```chrome_debug.log``` with alot more information to share.

BillyBlaze, 2016-06-19:
Ok, 0.3.4 has just been released. If you install the maintenance branch then please follow these instructions to prevent TouchUI not working in the future:

1. Remove TouchUI and install the latest 0.3.4 version through the plugin manager

1. Make sure you're using the right bootloader repo:

    ```
    rm -rf ~/TouchUI-autostart/
    git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/
    ```

1. Register the new paths:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
    sudo update-rc.d touchui defaults
    ````

reboot and it should work :)

enrstudio, 2016-06-20:
This new solution does not allow me to going into the GUI, the splash
screen counter going out of range without connecting...

BillyBlaze, 2016-06-20:
If the screen counter keeps going, and doesn't throw an error after 30 tries, then that means you're running an old version of the bootloader where this bug wasn't fixed.

If it does throw an error, then that means you have an old development version of 0.3.4 installed or TouchUI failed to load during startup (or is not installed).

enrstudio, 2016-06-20:
Probably I have an old version of the bootloader, what can I do?

BillyBlaze, 2016-06-20:
Did you follow these steps?

1. Make sure you're using the right bootloader repository and are up-to-date:

    ```
    rm -rf ~/TouchUI-autostart/
    git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/
    ```

1. Register the new paths:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
    sudo update-rc.d touchui defaults
    ````

enrstudio, 2016-06-27:
I have octopi: 0.13.0, OctoPrint: 1.2.13, TouchUI: 0.3.4 and yes, I have done all this...

BillyBlaze, 2016-06-27:
ok, I just found an issue reported by @EngineeredEntropy and also a fix.

Could you try this:

1. Update TouchUI bootloader:

    ```
    cd ~/TouchUI-autostart/
    git pull
    ```

[7 more replies]

commit 4cc3a14a (2016-06-13): :bug: Disable hasTouch false on Touchscreen

Touchscreen devices (e.g. Raspberry Pi Offical) no longer use mouse events and emit touch events; if hasTouch is set to false, then iScroll will not detect touch events and scrolling will not work. Fixes #46
closed 2016-06-27 by BillyBlaze