BillyBlaze/OctoPrint-TouchUI #48: Screen displays common web-frontend after reboot

MaxMihawk, 2016-06-12:
Hi Billy. I´ve installed your script yesterday (octopi + touchui with chromium autoboot in kiosk mode, all with actual versions). All works like a charm (touchUI frontend is working, touch is working, i can login, all is perfectly) until i reboot (putty sudo reboot). At the next start (and the following restarts), the screen is displaying the common (white) web-frontend. Any ideas?

I´m using a Pi3. Maybe it´s possible to force the touchUI frontend with a special URL? :)

BillyBlaze, 2016-06-12:
Hi, Thanks for reporthing, You can force TouchUI by adding #touch at the end of the URL. However I suspect that you're having the same issue's as #42

Could you try this out:
1. Uninstall TouchUI and install TouchUI by url with the following url:

    ```
    https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip
    ```

1. Login to ssh and get the new bootloader:

    ```
    git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/
    ```

1. Register the new paths:

    ````
    sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
    sudo chmod +x /etc/init.d/touchui
    sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
    sudo update-rc.d touchui defaults
    ````

reboot and a new screen should appear before loading OctoPrint.

MaxMihawk, 2016-06-12:
Hi. Thank you very much. It´s better now. The frontend tells "TouchUI not loaded" and you can touch for a new try. Thats much better, thx :) .
PS: I like the color fading :+1: .