BillyBlaze/OctoPrint-TouchUI #50: Latest update not rendering in Epiphany correctly

shanewarner, 2016-06-21:
Controls page is broken up and not rendering correctly in the epiphany browser after the last update. The mouse is also now highlighting text rather than scrolling up and down on a touchscreen interface.

BillyBlaze, 2016-06-22:
Thanks for reporting! I could reproduce the scrolling issue, and pushed a fix to the maintenance branch. Epiphany is reporting that you're using a touch device, and we listen to Touch events, however Epiphany is not giving us touch events, but mouse events....

However the controls tab is working fine for me, and I haven't changed any codes in the last update. Do you have any plugins installed for the controls tab?

You can test the scroll fix by the following method:

1. Uninstall TouchUI and install TouchUI by url with the following url:

    ```
    https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip
    ```

shanewarner, 2016-06-22:
Everything appears to be working with the maintenance branch fix now. Thanks!

BillyBlaze, 2016-06-24:
Sorry forgot to respond! Awesome and thanks for the confirmation!!

I will leave this ticket open for now so if someone else has this issue they can fix it. (It will automatically closed by github in the next release)

commit 4b3bf888 (2016-06-22): :bug: Epiphany wrong input device

Epiphany is reporting that touch is enabled, while this is not true. Therefor disable touch and listen to mouse events on KWEB and Epiphany.

Fixes #50
closed 2016-12-30 by BillyBlaze