BillyBlaze/OctoPrint-TouchUI #52: TouchUI Autostart fails to connect when using USB wireless keyboard

EngineeredEntropy, 2016-06-27:
Have an issue with TouchUI failing to connect to the OctoPrint backend when a Logitich Unifying USB wireless reciever attached to the Raspberry Pi.

**Setup:**
1. Raspberry Pi 3 with a fresh Rasbian (Jessie Full) install.
2. Installed Adafruit PiTFT drivers
3. Installed OctoPrint
4. Installed TouchUI from OctoPrint plugin manager.
5. Installed TouchUI auto-start.

The only things connected to the Raspberry Pi were the Logitech USB receiver and the Adafruit 3.5" PiTFT.

**Problem:**
TouchUI times out on startup with the message "Connecting to TouchUI failed.."  It appears to be caused by the User-Agent string in chromium.xinit file.

OctoPrint logged a malformed HTTP message error to console everytime TouchUI attempted to connect:

OctoPrint Error: 2016-06-26 00:45:14,001 - tornado.general - INFO - Malformed HTTP message from 127.0.0.1: Malformed HTTP headers: '\r\nHost: localhost:5000\r\nConnection: keep-alive\r\nUser-Agent: TouchUI'

Chromium debug log showing user-agent string:

[7142:7176:0626/004520:VERBOSE1:spdy_header_block.cc(125)] Inserting: (user-agent, TouchUI (X11, Chrome 50.0.2661.102-0ubuntu0.14.04.1.1118) (P:5000) (Logitech USB Receiver
Logitech USB Receiver
stmpe-ts))

**Fix:**
Changed chromium.xinit user-agent line from:
--user-agent="TouchUI (X11, Chrome $CHROME_VERSION) (P:$PORT) ($UA_DEVICE)" \
to
--user-agent="TouchUI (X11, Chrome $CHROME_VERSION) (P:$PORT)" \

Resulting chromium log user-agent string:
[2786:2820:0626/214441:VERBOSE1:spdy_header_block.cc(125)] Inserting: (user-agent, TouchUI (X11, Chrome 50.0.2661.102-0ubuntu0.14.04.1.1118) (P:5000))

TouchUI was able to connect to the backend after the fix was applied.

BillyBlaze, 2016-06-27:
Thanks for the report and for the detailed explanation. I guess I will drop the $UA_DEVICE, thought it might be handy in the future, but as far as I can see it will only give more problems.

Again thanks for the report!

closed 2016-06-27 by BillyBlaze

EngineeredEntropy, 2016-06-28:
No problem, it was an interesting one to track down.  Hopefully some future
headaches were prevented.