BillyBlaze/OctoPrint-TouchUI #53: Can I have a piTablet run the touchui exclusively without octoprint installed?

ExclusiveOne, 2016-07-01:
I had a PiTablet running my 3d printer, but i needed to move the printer, so i have a ne pi3 running octoprintwith a pi cam. I have a new pitablet built that I would like to keep at my desk to only run the touchui

BillyBlaze, 2016-07-01:
Hello,

Yes, thats possible, install the bootloader on the device you want TouchUI on, then after doing all the steps, edit this file:

https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/load-screen/script.js#L8

Which can be located under ```/home/pi/TouchUI-autostart/load-screen/```

Then at line 8, change ```localhost``` to the IP of your OctoPrint, you might need to login into your router and assign a static IP address to your OctoPrint-Pi, this so that this IP will not change after X amount of hours.

ExclusiveOne, 2016-07-01:
Do I need to have octoprint installed on the tablet also?

ExclusiveOne, 2016-07-01:
> sudo update-rc.d touchui defaults
> insserv: Service octoprint has to be enabled to start service touchui
> insserv: exiting now!
> update-rc.d: error: insserv rejected the script header
Getting this on the last step of the install

BillyBlaze, 2016-07-01:
No, you don't need octoprint to be installed on the tablet, only the TouchUI bootloader.

Oww yeah I forgot that was a dependency in the daemon script.

Edit ```/home/pi/TouchUI-autostart/touchui.init```

and remove ```$all octoprint``` on line 5

Then you can redo this steps.

ExclusiveOne, 2016-07-01:
for some reason it is still booting to desktop

ExclusiveOne, 2016-07-01:
I acctually dont think I am getting the load-screen file written right. when I try to pull it up it says it is a directory.

BillyBlaze, 2016-07-01:
Booting to desktop? I assume you're running Wheezy or Jessie (Full)?

Did you follow the '[make sure you boot to the command line](https://www.raspberrypi.org/documentation/configuration/raspi-config.md)' step?

ExclusiveOne, 2016-07-01:
I installed noobs. Guess I should start over with Jessie full?

BillyBlaze, 2016-07-01:
No, not necessary, you can still disable it manually but how is uncharted territory for me.

Btw. I thought NOOBS included Raspbian as default, is the command not found?

ExclusiveOne, 2016-07-01:
Well I took a step back, re-wrote my card, and got the touchui to auto boot. now when I try to open `/home/pi/TouchUI-autostart/load-screen/` it says it is a directory, and that is where my knowledge runs out! So all I need to do now is learn how to edit that file, and get my ip in there.

BillyBlaze, 2016-07-01:
yes ```load-screen/``` is the directory, and you're looking to edit ```script.js``` on line 8 :wink:

ExclusiveOne, 2016-07-01:
How fantastic! Ok I got the ip in there, but it still doesn't want to connect on reboot. I might be in left feild, but do I need to creat a user in octoprint to allow the tablet to connect?

BillyBlaze, 2016-07-02:
Did you install the main TouchUI plugin on the OctoPrint-Pi?

ExclusiveOne, 2016-07-02:
Yes i have 0.3.4 installed

     Austin Truelove
     UNITED MECHANICAL INC.
     PHONE *(405)409-4408 <(405)409-4408>*
*     ATruelove@Unitedmech.com <ATruelove@Unitedmech.com>*
[image: United Mechanical Construction_LOGO]

BillyBlaze, 2016-07-02:
Hmm, okay could explain a little bit better where you're stuck on? I might misunderstood something.

ExclusiveOne, 2016-07-02:
This is how I left line 8 `var prefix = "http://192.168.1.3:"+port+"/";` The tablet does it's count to 10 trying to connect, the says connection failed [Screen Image](http://i.imgur.com/ysQBX4k.jpg)

BillyBlaze, 2016-07-02:
Okay thanks, if you try a computer and manually go to ```http://192.168.1.3:5000/``` does OctoPrint load? Perhaps you're using OctoPi and is it blocked?

ExclusiveOne, 2016-07-02:
That has to be it. it is blocked only sometimes i have to give permission to open it though. but it only opens it i use `http://192.168.1.3`

BillyBlaze, 2016-07-02:
okay, so if you change

```var prefix = "http://localhost:"+port+"/";```

into

```var prefix = "http://192.168.1.3/";```

then it works?

ExclusiveOne, 2016-07-02:
My computer connects that way in chrome. The tablet does not

BillyBlaze, 2016-07-02:
and does ```http://192.168.1.3/plugin/touchui/ping``` work on your computer??

ExclusiveOne, 2016-07-02:
just a screen that says 

> OK  

on it

ExclusiveOne, 2016-07-02:
The octopi is on wifi, but that shouldnt affect it should it? I have to tablet connected directly to the router

BillyBlaze, 2016-07-02:
hmm no that shouldn't affect it, if both devices are connected to the same router.

can you edit ```~/TouchUI-autostart/chromium.xinit``` and uncomment the 5th line? Then retry everything and after that share the log ```~/.config/chromium/chrome_debug.log```??

ExclusiveOne, 2016-07-02:
To be honest, Im not sure how to share the log. I have it pulled up in putty, but from there i'm not sure unfortunately

BillyBlaze, 2016-07-02:
You can use Pastebin or just save it as .txt file and drag-and-drop it in github

ExclusiveOne, 2016-07-02:
is there a simple way to copy the file? Sorry, I still have my training wheels on.

ExclusiveOne, 2016-07-02:
I mean copy the file content onto clipboard. i cant simply `ctrl+C` because it is multiple pages long

BillyBlaze, 2016-07-02:
Right click on the title bar (Saying Command Prompt) and click Edit > Mark

ExclusiveOne, 2016-07-02:
phew that was a learning experience. [Log](https://www.dropbox.com/s/tum8tmey0tsh932/YcRd8qnS.txt?dl=0)

BillyBlaze, 2016-07-03:
Everything looks fine, maybe you can connect with ```http://yourprintername.local/```

I just tested it, and everything worked as expected.

You can also be sure that your tablet can connect to your Pi by pinging it with ```ping 192.168...``` or ```ping yourpintername.local```

ExclusiveOne, 2016-07-03:
I can ping it, but It wont connect `octopi.local` either. I tried that last night unfortunately.

BillyBlaze, 2016-07-03:
hmm okay, lets check out the external developers tool. you already have it enabled.

Just run ```ssh -L 0.0.0.0:9223:localhost:9222 localhost -N``` and open on your computer in the browser the following url: http://192.168.1.5:9223

Please share to me the contents of the console.

ExclusiveOne, 2016-07-03:
`pi@raspberrypi:~ $ ssh -L 0.0.0.0:9223:localhost:9222 localhost -N
pi@localhost's password:
` 

That URL wont open either.

BillyBlaze, 2016-07-03:
Try with ```sudo```

ExclusiveOne, 2016-07-03:
`pi@raspberrypi:~ $ sudo ssh -L 0.0.0.0:9223:localhost:9222 localhost -N
The authenticity of host 'localhost (::1)' can't be established.
ECDSA key fingerprint is 1f:9c:0e:a9:07:78:96:5c:71:2e:b6:94:6a:44:91:e8.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'localhost' (ECDSA) to the list of known hosts.
root@localhost's password:
Permission denied, please try again.
` 

 I did another fresh install to make sure I wasnt screwing anything up, and I am back to the same spot in my troubles. feeling doomed

ExclusiveOne, 2016-07-03:
My phone, and other two computers connect to that url no problem, so feeling a bit puzzled now

BillyBlaze, 2016-07-03:
Ok, try this; copy the files inside the directory ```load-screen``` to your computer, then open ```startup.html```in your Chrome browser. Does it still not connect? Can you then open the developer tools and copy the console from there?

ExclusiveOne, 2016-07-03:
I'm hoping this is what you are referring to
 `Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #10, waiting 100s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
http://192.168.1.3/webcam/?action=stream&1467533043240 Failed to load resource: net::ERR_ADDRESS_UNREACHABLE`

BillyBlaze, 2016-07-03:
Can you do it again but this time after you opened the developers tools, can you refresh the page? This so you get all errors / warnings

ExclusiveOne, 2016-07-03:
Starting dependency resolution...
packed_app.js?1ae0b1f2:11624 ... dependency resolution done
packed_app.js?1ae0b1f2:11976 Initial application setup done, connecting to server...
packed_app.js?1ae0b1f2:10675 Connected to the server
packed_app.js?1ae0b1f2:11963 Finalizing application startup
packed_app.js?1ae0b1f2:11868 Going to bind 24 view models...
packed_app.js?1ae0b1f2:11936 ... binding done
packed_app.js?1ae0b1f2:11955 Application startup complete
packed_app.js?1ae0b1f2:10588 Reconnect trial #0, waiting 0s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #1, waiting 1s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #2, waiting 1s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #3, waiting 2s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #4, waiting 3s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #5, waiting 5s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #6, waiting 8s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #7, waiting 13s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #8, waiting 20s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #9, waiting 40s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10588 Reconnect trial #10, waiting 100s
http://192.168.1.3/sockjs/info Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
http://192.168.1.3/webcam/?action=stream&1467533043240 Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
packed_app.js?1ae0b1f2:10675 Connected to the server

BillyBlaze, 2016-07-03:
And what error does TouchUI give you now?

Is this consistent with the tablet-pi?

ExclusiveOne, 2016-07-03:
I dont get an error. it just gotes through the loop of trying to connect. then the logo turns red, and says connection failed

BillyBlaze, 2016-07-03:
yeah the connection failed is the error message I was referring to. So you get error on both devices now?

also the log you just gave me indicates it did got connected, and tries to load OctoPrint. But I guess you have alot of errors connecting to non-TouchUI related stuff:

```
Failed to load resource: net::ERR_ADDRESS_UNREACHABLE
````

ExclusiveOne, 2016-07-03:
Question. If I can ping the octopi from the tablet, but but cant connect to it. Then it is something router side that is causing my hang up? Im starting to feel like it isnt getting an internet connection

BillyBlaze, 2016-07-03:
No, if you can ping it, you reach that end-point successfully. Your router could forward ports to a different port but I highly doubt that.

I think it has todo with OctoPi's permissions and closed ports. I tried OctoPI myself but I quickly went back to installing OctoPrint manually.

ExclusiveOne, 2016-07-03:
I will give that a try when this print is finished. I really appreciate all the help, this has for sure been a learning experience

BillyBlaze, 2016-07-16:
Any luck on this?

ExclusiveOne, 2016-07-16:
Unfortunately no so far, but I set it aside for a while. I'm still hanging up at the same point of trying to connect. i can ping it just fine, but the connection failed screen is still all i get

BillyBlaze, 2016-07-18:
Bummer... What is the output of this command: ```wget http://192.168.1.3/plugin/touchui/ping```?

BillyBlaze, 2016-07-23:
Any information on that command?

BillyBlaze, 2016-08-06:
Closed due to inactivity; feel free to add a comment. I will still be able to read that.

closed 2016-08-06 by BillyBlaze