BillyBlaze/OctoPrint-TouchUI #54: Official Display 7" Touchscreen rotated 90º -> Cant calibrate

bolsoncerrado, 2016-07-01:
Hi!

Im planning a vertical setup of my Pi+Octopi+TouchUI hence I have rotated the display config 90º.

However the touch axis remain horizontal instead of vertical for an unknown reason.

I followed the whole Peter deVries guide and whenever tried the CALIBRATION steps, all I get is a black screen after i touch the 4 points on adafruits calibrator. The system is there, ie not hung.

I also tried to manually inspect the  /etc/X11/xorg.conf.d/ location but that does not seem to exist on my fresh install either, hence creating a .conf (specially when I dont know its contents) wont help either.

Any ideas?

I just need to invert both axis now and then Im done.

Thanks!

bolsoncerrado, 2016-07-01:
Checked adafruit's PDF on how to calibrate, they have a section on their PDF where they comment on how to rotate the monitor 90º (that applies to their 4.3" touchpanel).

They instruct on how to manually create the dir and file, then I went ahead, pasted the following:

`Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "stmpe-ts"
Option "InvertX" "true"
Option "InvertY" "true"
EndSection`

and rebooted.

Nothing.

BillyBlaze, 2016-07-01:
Since you're using the Official Raspberry Pi touchscreen, you should use a different FrameBuffer (e.g. ```fb0``` or ```fb1```) then Adafruit. They use a touchscreen through SPI and get a new number, while yours should be attached on the default one.

That Adafruit calibration is your best shot, that's how I eventually got it to work. I rotated mine 180degrees so I can hook up the power supply from the top.

eventually you end up with a 99-calibration.conf like:
```
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "Calibration"   "176 3928 3928 223"
        Option  "SwapAxes"      "1"
EndSection
```

This is for my 3.5".

bolsoncerrado, 2016-07-01:
So probably my issue is the MatchProduct name?
If i do:
`
pi@:~ $ DISPLAY=:0 xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳   Mini Keyboard                         	id=7	[slave  pointer  (2)]
⎜   ↳ FT5406 memory based driver              	id=8	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳   Mini Keyboard                         	id=6	[slave  keyboard (3)]
`

I assume I should use MatchProduct "FT5406 memory based driver" instead?

BillyBlaze, 2016-07-01:
That could be a possibility, but to be sure you can better view ```/var/log/Xorg.0.log``` for that or run manually ```xinit ~/TouchUI-autostart/chromium.xinit``` to have a better view on how X11 is starting up.

bolsoncerrado, 2016-07-01:
Option  "SwapAxes"      "1"

or

Option "InvertX" "true"
Option "InvertY" "true"

?

BillyBlaze, 2016-07-01:
I think I copy-pasted that with the xinput calibration, and if the first one works, I would do that, less is better :+1:

bolsoncerrado, 2016-07-01:
Hmmm tried InvertX&y first and didnt work.

Then I treid Swap Axes and now I get possitive scroll on up/down movement but inverted on the other motion wtf!?!

BillyBlaze, 2016-07-01:
Wow, weird, but what if you try InvertX again together with SwapAxes?

bolsoncerrado, 2016-07-01:
FINALLY! WOHOO!!

Wow! The whole afternoon dealing with this! Phew!

Thanks ;)

Ps. How much u charge to customize the splash screen? LOL

BillyBlaze, 2016-07-01:
AWESOME!!

Your welcome :) and sorry, I don't have any time at the moment to do that...

closed 2016-07-01 by BillyBlaze

Bigfellas, 2016-08-21:
@bolsoncerrado  Would you be able to post a step by step on how you got this to work. I have been trying with scripts and all sorts. I would love to see how you got it solved.
Thanks
Matt

bolsoncerrado, 2016-08-21:
No scripts needed afaik? Just follow this thread from begin to end and you should be set :??