BillyBlaze/OctoPrint-TouchUI #55: Webcam STILL not working?

bolsoncerrado, 2016-07-01:
So I do have the webcam up and working:

1) flawlesly and streaming thru OctoPrint via a regular PC browser on the CONTROL tab
2) flawlesly by issuing the http://pi-ip-adrress:8080/?action=snapshot as you can see [here](http://i.imgur.com/rxF1jSq.png)
HOWEVER 3) added to snapstream plugin [as adviced](http://i.imgur.com/TAkPHKe.png), delivers [broken image](http://i.imgur.com/FA74IaF.jpg) on the WEBCAM button

any clues?

Thanks!

BillyBlaze, 2016-07-01:
Hmm yeah that troubleshooting is not flawless I see.. (It assumes the user has HAproxy configured)

I think it's looking at port 5000 and not 8080. If you save the whole url ```http://pi-ip-adrress:8080/?action=snapshot``` under Snapstream URL, it should work.

bolsoncerrado, 2016-07-01:
Yep with the whole URL works...

SO may be your guide needs a few reTouchs :D

closed 2016-07-01 by BillyBlaze