BillyBlaze/OctoPrint-TouchUI #56: GUI not saving changes...

bolsoncerrado, 2016-07-01:
...on a fresh install, and fresh reboot, the issue is I can only "update" colors of the TouchUI via a PC. If i try to save color changes via the Touch screen, I get [503 Forbidden error](http://i.imgur.com/e2kRj2Y.jpg).

Any clues?

BillyBlaze, 2016-07-01:
Are you logged in as an admin on the touchscreen?? And if you logout and login again, does it still persists??

I think OctoPrint UAC is denying access, also can you share your ~/log/octoprint.log if it contains anything about the error?

bolsoncerrado, 2016-07-08:
https://www.youtube.com/watch?v=smFs58DUmXU

bolsoncerrado, 2016-07-08:
...and the only weird entry in the .log is

py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.13-py2.7.egg/octoprint/util/__init__.py:564: UnicodeWarning: Unicode unequal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if value_source != value_target:

bolsoncerrado, 2016-07-15:
@BillyBlaze not sure if u saw the above youtube before closing the issue?

BillyBlaze, 2016-12-29:
It's been a very long time, and I am incapable of reproducing this problem every time I try to look into this. Are you still encountering this issue?

bolsoncerrado, 2017-01-02:
Since its been updated to 1.3.0 and u just updated touchui as well, close
it and i'll reopen if anything faces further. thanks!

BillyBlaze, 2017-01-02:
Your welcome and please do if you enocounter any 👍

closed 2017-01-02 by BillyBlaze