BillyBlaze/OctoPrint-TouchUI #57: [REQUEST?] Force refresh button?

bolsoncerrado, 2016-07-15:
Not sure if this is worth? I find out sometimes the GUI wont refresh and the UI becomes unusable making the printer uncontrollable thru the display touch panel on the raspi, hence I need to use my phone (or a computer), log into the web URL and manually refresh, reboot or restart the server so the GUI refreshes as well. May be a force refresh button on the  TouchUI would help out? Seems it happens when connection with the printer is lost (for example i tend to remote cut power using WeMo switches to the printer) and Octoprint looses connection to the printer, but no matter what I do later via TouchUI / Octoprint via the display touch (reboot, refresh, etc.) Octoprint/Pi wont reboot/restart at all.

BillyBlaze, 2016-07-16:
Hi,

You can make a System Command in OctoPrint that restarts TouchUI from there.

Read more about this in the [OctoPrint manual](https://github.com/foosel/OctoPrint/wiki/Cookbook:-System-Commands#reboot-the-server-linux)

The command for restarting TouchUI is: ```sudo service touchui restart```

bolsoncerrado, 2016-07-16:
I will try that but the same as raspi won't restart nor reboot nor
shutdown, not sure if this will make anything different. was thinking of a
refresh like a browser  page refresh.

but will report back.

BillyBlaze, 2016-08-24:
Thanks, any word? (else I will close this issue due to inactivity)

bolsoncerrado, 2016-08-28:
still have to get into creating the button...

BillyBlaze, 2016-11-15:
Closed due to inactivity. Please comment to re-open this issue.

closed 2016-11-15 by BillyBlaze