BillyBlaze/OctoPrint-TouchUI #59: Freeze under Chrome on RasPI with potential fix

lciscon, 2016-07-18:
I'm having fairly frequent freezes of the entire user interface when running TouchUI inside Chrome on a RPI-3 with the RPI touchscreen.  My current theory is that Chrome itself is freezing.  

I'm currently testing the potential fix of turning off hardware acceleration in Chrome.  This supposedly is the root of many Chrome freezes.

http://www.solveyourtech.com/turn-hardware-acceleration-google-chrome/

FYI you can get at the Chrome settings when you're running TouchUI full-screen by hitting Ctrl-N.  This opens up a new window with the menu pulldown.  Then you can select the Settings.

I will update this message in a couple of days once I know whether that fix actually takes care of the problem.

BillyBlaze, 2016-07-18:
Thanks for the info.

Did you try turning on logging of chrome and view the contents of chrome_debug.log?

If not; edit [chromium.xinit](https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/chromium.xinit#L5) and remove the hash infront.

halxinate, 2017-05-17:
TouchUI is **completely** freezing my Android phone (rooted Note 3 with Marshmellow) sometimes for several minutes (like 10-15), sometimes like forever after finishing loading in Chrome. Then system restarts. Uninstalled.