BillyBlaze/OctoPrint-TouchUI #61: Boot to Chromium browser not working

waityn4jc, 2016-08-09:
Paul,

I had been putting off doing any updates until I could have time to reinstall fresh and start using Chromium since that is now the supported path.  I started from scratch with the latest OctoPi image and then followed Adafruits DIY Installer script for the PiTFT 2.8 resistive.  That all seems to have installed fine and I can control Octoprint through a computer just fine.  I then followed the steps to make it boot direct to browser following your instructions here:
https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

I suppose it must not have worked because it still boots to the normal linux log in.  Is there a way to see what might have gone wrong?  Thank you.
[image]

BillyBlaze, 2016-08-09:
It seems TouchUI started, perhaps it's still attached to the wrong 'video port' (e.g. fb0), on step 6, what config did you use?

If that's correct, could you share the log /var/logs/Xorg.0.log

waityn4jc, 2016-08-09:
I used this one:

Section "Device"
 Identifier "touchscreen"
 Driver "fbdev"
 Option "fbdev" "/dev/fb1"
EndSection

http://pastebin.com/dfcF6VR2

Is there a place to make a donation for what you do with TouchUI?  It amazes me the support you provide.

BillyBlaze, 2016-08-11:
hmm, I think it has todo with this error:

```
AIGLX error: dlopen of /usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so: cannot open shared object file: No such file or directory)
```

Could you try the following:

1.  Edit ```sudo nano /etc/default/touchui```

1.  And change ```pi``` into ```root``` on line 4

1. Save and reboot

Also thank you & no I dont accept donations :) it's all free!

BillyBlaze, 2016-08-24:
any word? (else I will close this issue due to inactivity)

waityn4jc, 2016-08-24:
Yes, please close it for now.  I got 2 new printers and have not been able to try this on the old one.  I am currently just controlling it through the web interface.  I super appreciate your help and still have the fix in my inbox to try, but haven’t been able to do it.

BillyBlaze, 2016-08-25:
oki, no problem & your welcome. Just leave a comment if you have any questions

waityn4jc, 2016-10-06:
So, I finally got around to trying this fix and it doesn’t seem to have fixed it.  I got two new printers in the meantime and set them up from scratch and they are working perfectly.  This machine used to work fine months ago but when an update was required it has not worked since.  I reimaged this machine in hopes that would work, but it’s still doing the same thing.  I then attempted this change you suggested and it’s still doing the same thing.

Here is a log file I got using cat /var/log/Xorg.0.log.old  http://pastebin.com/v3PYihgE

My two new machines are Pi 3s as where this one is a Pi2.  The new ones are using the Adafruit 3.2 (which uses the same software as the 2.8)  This machine is the old 2.8.

I’m starting to think I just need to buy a replacement Pi and screen for this machine since it seems to work when I do that…

Any ideas before I do that?

Thanks again for all your help.

BillyBlaze, 2016-10-07:
Heya,

First, your welcome. I also like to know what is the exact cause for this (might help others in the future)

Could you try this, I found this on Adafruit forum:
"Sounds the resistive touch was detected but not the screen. Check your soldering on the Pi headers. also make sure that the display connector is well seated. The flex connector should have both side's closed and the flex connector well seated so that the pins are all the way in."

BillyBlaze, 2016-11-15:
Closed due to inactivity. Please comment to re-open this issue.

closed 2016-11-15 by BillyBlaze