BillyBlaze/OctoPrint-TouchUI #62: Touch Events not working on Waveshare 7" v2.1 LCD

CapnBry, 2016-08-15:
The new Waveshare 7" LCD version 2.1 now uses a HID device for the touchscreen driver, which means it works out of the box with Raspbian. However, using the TouchUI-autostart script, "clicks" work, but not drags so you can't scroll the TouchUI or access the gcode item slide list. Oddly enough, booting to the full standard Raspbian desktop and launching chromium-browser (no command line options), the touch and drag events work as expected.

To work around this, I force touch events in the same way FT5406 touch events are forced and everything works great!
```
diff --git a/chromium.xinit b/chromium.xinit
index 68e8d24..9d5cfe3 100644
--- a/chromium.xinit
+++ b/chromium.xinit
@@ -13,7 +13,7 @@ sed -i 's/"exit_type": "Crashed"/"exit_type": "None"/' ~/.config/chromium/Defaul

 # Force Chrome to render with touch events
 # Bugfix #46
-TOUCHUI_DEVICE=$(egrep -o "FT5406" /proc/bus/input/devices)
+TOUCHUI_DEVICE=$(egrep -o "ZH851|FT5406" /proc/bus/input/devices)
 if [ -z "$TOUCHUI_DEVICE" ]
 then
        echo "TouchUI: touch events automaticlly detected"
```

BillyBlaze, 2016-08-15:
Thanks for reporting :+1: 

I immediately committed this to the master branch.

closed 2016-08-15 by BillyBlaze

BillyBlaze, 2017-01-15:
@CapnBry, I got to find a more solid solution for detecting touch events, could you perhaps share your `cat /proc/bus/input/devices`

I am hoping that `H: Handlers=mouse0 event0` will be different on your machine.

CapnBry, 2017-01-15:
You bet:
```
I: Bus=0003 Vendor=0eef Product=0005 Version=0110
N: Name="BYZHYYZHY By ZH851"
P: Phys=usb-3f980000.usb-1.4/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:0EEF:0005.0001/input/input0
U: Uniq=Â§ÃP39112398
H: Handlers=mouse0 event0
B: PROP=2
B: EV=b
B: KEY=400 0 0 0 0 0 0 0 0 0 0
B: ABS=2608000 3
```

synman, 2017-02-03:
Here's mine:

pi@octopi:~/TouchUI-autostart $ cat /proc/bus/input/devices 
I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="ADS7846 Touchscreen"
P: Phys=spi0.1/input0
S: Sysfs=/devices/platform/soc/20204000.spi/spi_master/spi0/spi0.1/input/input0
U: Uniq=
H: Handlers=mouse0 event0 
B: PROP=0
B: EV=b
B: KEY=400 0 0 0 0 0 0 0 0 0 0
B: ABS=1000003

BillyBlaze, 2017-02-04:
Thank you both very much!1! Sadly this wont help us...

However, after finding a linux device with touch events I was able to find out a way to detect touch events. ```xinput --long``` will report the classes X11 loads and with touch it will be something like XITouchClass.

I have created a maintenance branch and I was wondering if you guys could try out if this also works for you?

If so, just follow these steps:

1. ```sudo service touchui stop```
1. ```cd ~/TouchUI-autostart/```
1. ```git checkout maintenance```
1. ```git pull```
1. ```sudo apt-get install xinput```
1. ```sudo xinit ~/TouchUI-autostart/chromium.xinit```

synman, 2017-02-04:
`pi@octopi:~ $ sudo service touchui stop
pi@octopi:~ $ cd ~/TouchUI-autostart/
pi@octopi:~/TouchUI-autostart $ git checkout maintenance
Branch maintenance set up to track remote branch maintenance from origin.
Switched to a new branch 'maintenance'
pi@octopi:~/TouchUI-autostart $ git pull
Already up-to-date.`

Is this expected?

BillyBlaze, 2017-02-04:
No, argh, I forgot to fetch first. I think its a local branch now..

Run:

1. ```git branch --set-upstream maintenance origin/maintenance```
1. ```git pull```

That should grab the new files.

synman, 2017-02-04:
still up to date

synman, 2017-02-04:
and i can report scrolling works fine

BillyBlaze, 2017-02-04:
ahh, I haven't seen your edited comment.

Yeah depending on your git setup it will pull the entire remote. So you might already have the new changes.

What is the output of this command ```cat ~/TouchUI-autostart/chromium.xinit | grep XITouchClass```??

synman, 2017-02-04:
sorry... we're typing over one another... see above

synman, 2017-02-04:
re:  cat of chromium.xinit

```
pi@octopi:~/TouchUI-autostart $ cat ~/TouchUI-autostart/chromium.xinit | grep XITouchClass
TOUCHUI_DEVICE=$(xinput --list --long | grep XITouchClass)
```

I'll stop editing posts LOL

BillyBlaze, 2017-02-04:
Ok that is awesome 👍 thank you very much for verifying.

If @CapnBry also could verify if this fix is working, then I will release this bugfix.

synman, 2017-02-04:
I have an unrelated question regarding SnapStream.  Do you mind if I open a issue for it?

CapnBry, 2017-02-05:
Works like a champ. All I needed to do was this
```
git pull
git checkout maintenance
```

If it matters at all, this is my xinput
```
â   â³ BYZHYYZHY By ZH851                        id=6    [slave  pointer  (2)]
        Reporting 4 classes:
                Class originated from: 6. Type: XIButtonClass
                Buttons supported: 5
                Button labels: "Button Unknown" "Button Unknown" "Button Unknown" "Button Wheel Up" "Button Wheel Down"
                Button state:
                Class originated from: 6. Type: XIValuatorClass
                Detail for Valuator 0:
                  Label: Abs MT Position X
                  Range: 0.000000 - 1024.000000
                  Resolution: 0 units/m
                  Mode: absolute
                  Current value: 208.000000
                Class originated from: 6. Type: XIValuatorClass
                Detail for Valuator 1:
                  Label: Abs MT Position Y
                  Range: 0.000000 - 600.000000
                  Resolution: 0 units/m
                  Mode: absolute
                  Current value: 328.000000
                Class originated from: 6. Type: XITouchClass
                Touch mode: direct
                Max number of touches: 10
```