BillyBlaze/OctoPrint-TouchUI #64: TouchUI not logging in when launched

ssa3512, 2016-09-02:
Fresh install today - When TouchUI boots, it loads the UI but none of the buttons are responsive. If I go to the menu and disable TouchUI, octoprint shows not logged in, then after a few seconds logs in automatically (as configured in config.yaml)

If I then return to TouchUI the controls work.

Please see this video of the issue. https://www.youtube.com/watch?v=b92cIZTkw0Y
 Octoprint & Chrome debug logs from taking this video are attached.
[chromelog.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/452968/chromelog.txt)
[octoprintlog.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/452969/octoprintlog.txt)

ssa3512, 2016-09-02:
Sorry, should have mentioned: Octoprint 1.2.15, Chromium 51, Rpi 2 B running Raspian Jessie. Screen is the official 7" touchscreen

BillyBlaze, 2016-09-03:
Wow, thanks for reporting, that's really weird.

I cannot see anything obvious, no errors or warnings in any logs, everything looks running fine. Perhaps there is a conflict with a plugin, could you try if there is an issue with Custom Control Editor or History plugin?

ssa3512, 2016-09-04:
I was just able to reproduce running locally in a fresh virtualenv from the latest octoprint master on my windows machine. No plugins except the defaults and TouchUI installed via plugin manager. Getting the 401 UNAUTHORIZED messages in chrome as shown below. Unfortunately, it looks like this is an octoprint and/or browser issue, rather than a TouchUI issue. If I disable TouchUI in plugin manager, and open a new incognito window I get the same 401 errors. If I try the same behavior in edge inprivate or firefox private window I have no issue, with or without TouchUI. Any thoughts on additional troubleshooting I should do, or should I submit a bug on the Octoprint repository?

[octoprintlog.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/453754/octoprintlog.txt)
[configyaml.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/453755/configyaml.txt)

[image]

BillyBlaze, 2016-09-04:
Thanks alot for the detailed explanation, I don't understand why this 401 errors didn't show up in the chromelog.txt. But that aside.

It does sounds like a serious bug in either OctoPrint or Chrome, but I suspect the latter since you couldn't reproduce it with Firefox/Edge.

I think - since TouchUI doesn't do anything on the backend - that posting this in OctoPrint will be more helpful.

ssa3512, 2016-09-06:
I started writing up a bug for Octoprint and now I can't reproduce the issue on my desktop. Go figure. I hooked up a keyboard to the rpi and 'logged out' then logged back in with the username and password to get a cookie. Seems to be ok now when I reboot the pi so I am going to leave it for now. Not sure if there is something weird going on with my laptop but I did all the same steps to set up virtualenv etc. Not a particularly satisfying resolution, as I certainly don't know what was causing the issue in the first place, but I'm not sure what else to do at this point.

closed 2016-09-06 by ssa3512