BillyBlaze/OctoPrint-TouchUI #65: [?????] Any clues what's this about?

bolsoncerrado, 2016-09-30:
I'm getting this since upgrading Octoprint to 1.2.16 Master

bolsoncerrado, 2016-09-30:
Errr.... 
[image]

bolsoncerrado, 2016-10-01:
Tried to uninstall, reboot and reinstall touchi, this is what I got:

Installing plugin...

Installing plugin "TouchUI" from https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/master.zip...
/home/pi/oprint/bin/pip install https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/master.zip
Collecting https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/master.zip
Downloading https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/master.zip
Requirement already satisfied (use --upgrade to upgrade): OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg (from TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): tornado<4.1,>=4.0.1 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.1-py2.7-linux-armv7l.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): sockjs-tornado<1.1,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.1-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.6-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): pyserial<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.17-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.0-py2.7-linux-armv7l.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): requests<2.8,>=2.7.0 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): psutil<3.3,>=3.2.1 in /home/pi/oprint/lib/python2.7/site-packages/psutil-3.2.1-py2.7-linux-armv7l.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): Jinja2>=2.4 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8-py2.7.egg (from flask<0.11,>=0.9->OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): itsdangerous>=0.21 in /home/pi/oprint/lib/python2.7/site-packages/itsdangerous-0.24-py2.7.egg (from flask<0.11,>=0.9->OctoPrint->TouchUI==0.3.4)
[... cut ...]
Requirement already satisfied (use --upgrade to upgrade): MarkupSafe in /home/pi/oprint/lib/python2.7/site-packages/MarkupSafe-0.23-py2.7-linux-armv7l.egg (from Jinja2>=2.4->flask<0.11,>=0.9->OctoPrint->TouchUI==0.3.4)
Requirement already satisfied (use --upgrade to upgrade): pytz>=0a in /home/pi/oprint/lib/python2.7/site-packages/pytz-2016.3-py2.7.egg (from Babel>=1.0->Flask-Babel<0.10,>=0.9->OctoPrint->TouchUI==0.3.4)
Installing collected packages: TouchUI
Running setup.py install for TouchUI: started
Running setup.py install for TouchUI: finished with status 'done'
Successfully installed TouchUI-0.3.4
Done!

But the message persists.

BillyBlaze, 2016-10-01:
What plugins are you using? 

getExtension is not a function TouchUI nor OctoPrint uses. So it looks like another plugin is crippling OctoPrints startup procedure.

However we do disable all extensions in Chrome. If we can rule out that a plugin is causing this issue, then we might need to enable extensions in the startup file.

bolsoncerrado, 2016-10-01:
the weird thing is that I updated 4 raspi with the same exact. image and
only. 1 is failing :???

lemme dig the details on the plug-ins list hmmm

bolsoncerrado, 2016-10-01:
Now this is weirder! I'm able to log into touchui and octoprint remotely but it's locally what's failing?!?! 
[image]

BillyBlaze, 2016-10-02:
I'm pretty sure this problem is caused by the plugin "Slicer plugin". It tries to load WebGL and that is not supported on the RaspberryPi. Hence the function getExtension is not supported, it seems the library Three.js is throwing this error.

Basically there is no check if the browser supports WebGL, what breaks all other features of OctoPrint and TouchUI.

Please confirm by disabling the plugin, and if so, please report the error to that plugin.

bolsoncerrado, 2016-10-02:
thanks!  will do!

bolsoncerrado, 2016-10-03:
Confirmed, works again upon deinstallation of the slicer plugin.

BillyBlaze, 2016-10-03:
Thanks for the confirmation!

closed 2016-10-03 by BillyBlaze