BillyBlaze/OctoPrint-TouchUI #67: unable to start touchUI

Mermalady, 2016-10-12:
Hiya, 
I have followed the instructions but Pi is still rebooting to command line. I have it hooked up to a touch screen which is working fine (well, the screen is fine, haven't gotten to GUI yet). 
In another thread you suggested using
xinit /home/pi/OctoPrint-TouchUI-autostart/start_chromium.xinit
I'm getting 'a no such directory'. 
In /home/pi there's a 'TouchUIautostart' and 'TouchUI-autostart'. In home/pi/Octoprint there is no autostart files.

BillyBlaze, 2016-10-12:
Hi,

I think something went terrible wrong. There should only be one directory 'TouchUI-autostart'.

Delete both directories and try again these steps:
```
git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/     
```

```
sudo cp ~/TouchUI-autostart/touchui.init /etc/init.d/touchui
sudo chmod +x /etc/init.d/touchui
sudo cp ~/TouchUI-autostart/touchui.default /etc/default/touchui
sudo update-rc.d touchui defaults
```

Mermalady, 2016-10-12:
Heya, Thanks for getting back to me. After following those steps, running 
xinit /home/pi/OctoPrint-TouchUI-autostart/start_chromium.xinit
still gets me a 'no such file or directory'. LS in ~home/pi/  returns
mjpg-streamer OctoPiPanel OctoPrint oprint rpi-fbcp scripts TouchUI-autostart
inside /TouchUI-autostart is
chromium.xinit load-screen README.md touchui.default touchui.init
I attempted to run variants on: xinit /home/pi/OctoPrint-TouchUI-autostart/start_chromium.xinit
xinit /home/pi/OctoPrint-TouchUI-autostart/chromium.xinit
xinit /home/pi/OctoPrint/TouchUI-autostart/start_chromium.xinit
all returned 'no such file or directory'

BillyBlaze, 2016-10-12:
Did you try ```xinit /home/pi/TouchUI-autostart/chromium.xinit```?

Mermalady, 2016-10-12:
I'm getting various errors:
`sed: can't read /home/pi.config/Chromium/Defualt/Preferences: No such file or directory`
checking..

BillyBlaze, 2016-10-12:
Did you install Chromium successfully on this [page](https://www.raspberrypi.org/forums/viewtopic.php?t=121195)?

Mermalady, 2016-10-12:
Wondering about that- since it's saying 'this guide is not needed'. I went through the steps though I believe there were issues with the youtube bit.

Mermalady, 2016-10-12:
I remember. I Got:
`invalid option -- 'y'`

BillyBlaze, 2016-10-12:
Hmm, I see, there has been a major change with Chromium.

I need to update a few things. Hold on :)

Mermalady, 2016-10-12:
Yay! Thanks so much, i was worried I was SOL. Let me know when you know more/ETA :) I'm building this for my lab.

BillyBlaze, 2016-10-12:
I think I got it running again.

Let's first remove some old stuff:

1. Run:  
   ```
   sudo apt-get remove chromium-browser rpi-youtube
   sudo apt-get autoremove
   sudo apt-key del Bintray
   ```

2. Run and Place a "#" infront of the line that contains Bintray  
   ```
   sudo nano /etc/apt/sources.list
   ```

  Save and Exit (CTR-O, Enter, CTR-X)

3. Run:  
   ```
   sudo apt-get update
   ```

4. Reinstall Chromium:  
   ```
   sudo apt-get install chromium-browser
   ```

5. Now run:  
   ```
   xinit ~/TouchUI-autostart/chromium.xinit
   ```

If you get the error:
```
/usr/lib/chromium-browser/chromium-browser: symbol lookup error: /usr/lib/chromium-browser/chromium-browser: undefined symbol: mmal_vc_init_fd
```

Then we need to update your kernel (you might need to re-install your driver):

1. Run:  
   ```
   sudo apt-get install rpi-update
   ```

2. Run:  
   ```
   sudo rpi-update
   ```

3. Reboot:  
   ```
   sudo reboot
   ```

Mermalady, 2016-10-12:
could you elaborate on

> Run and Place a "#" infront of the line that contains Bintray
?

BillyBlaze, 2016-10-12:
In that file, change the following line:
````
deb http://dl.bintray.com/kusti8/chromium-rpi jessie main
````

into:
````
#deb http://dl.bintray.com/kusti8/chromium-rpi jessie main
````

Mermalady, 2016-10-12:
Oddly, that line doesn't exist in that file. Shall I continue on with the other steps?

BillyBlaze, 2016-10-12:
Hmm then that means you where already using the latest Chromium, you could skip all steps and only do these:

1. Run:  
   ```
   sudo apt-get upgrade
   sudo apt-get install rpi-update
   ```

2. Run:  
   ```
   sudo rpi-update
   ```

3. Reboot:  
   ```
   sudo reboot
   ```

BillyBlaze, 2016-10-12:
oww wait and ofcourse reinstall Chromium with:
```
sudo apt-get install chromium-browser
```

Mermalady, 2016-10-12:
ok, working on it. Sudo upgrade is taking forever.

Mermalady, 2016-10-12:
Huzzah! Has loaded up. Trying to load files off of SD card now.

BillyBlaze, 2016-10-12:
Awesome 👍 

Are you using OctoPi btw?

Mermalady, 2016-10-12:
ya, I'm using Octopi. We had an astrobox setup in my lab (we have 8 machines for students to use) and it was *awful* so wanted to try yours. I'm having trouble reading SD card, BTW- initialize and refresh files not showing anything.

BillyBlaze, 2016-10-12:
Ok thanks :)

TouchUI just changes the layout, this should be dealt with by OctoPrint, try if you have the same issue on a pc without TouchUI, then you might have found a bug.

Mermalady, 2016-10-12:
I'm hoping to use this as a direct control for individual machines, is that..not the right way to use it?

BillyBlaze, 2016-10-12:
Yes, OctoPrint will do this, TouchUI is just a plugin that will change the layout to a mobile/touchscreen friendly interface, It still relies on the main features provided by OctoPrint :) Also if you grab a laptop (or pc) and connect to the printer, you will get the default interface. It might be easier to setup your initial config on that.

(closing this issue since you confirmed the solution + I have updated the Wiki, please feel free to continue)

closed 2016-10-12 by BillyBlaze

Mermalady, 2016-10-12:
so that connection would be laptop > pi running octopi/touchui > printer? 
one of the weird problems in my particular lab is that I can't connect the pi and the controlling laptop to the same network, so I can't access octopi.local or SSH into it (yet, working on it).

BillyBlaze, 2016-10-12:
1. Yes.
2. OctoPi has some strict security. Are you sure you have bonjour installed on your laptop? You should be able to login on SSH with the IP address, I'm sure you cannot access OctoPi with IP addressed (that's locked). If you install OctoPrint manually, all this security would not be in place and you can access everything easier.

Mermalady, 2016-10-12:
First- thanks for being so super super helpful, i really appreciate it. Then.. how do I get into terminal from the gui? I plugged an SD card into a USB adapter and that into the pi, shouldn't that be showing up?

BillyBlaze, 2016-10-12:
Your welcome.

OctoPrint will store everything on your Pi, the SD card you're referring to, is if your printer support SD cards. Normally when you print with OctoPrint the file will be 'streamed' to the printer. If your printer has an SD card, then OctoPrint can 'upload' to your printer.

Mermalady, 2016-10-12:
hm. My goal is to actually have the whole system be based off of SD cards, but with a graphical interface (that's new-user friendly). I'm running printerbot metal+, btw. 
My hope is to have the pi + touchui + sd card reader for each machine.

BillyBlaze, 2016-10-12:
I see, nice plan. To be honest you're better off on OctoPrint's Community pages. I have absolutely no experience with setups like these. If OctoPrint supports it, then TouchUI should also be able. If not TouchUI wont provide extra functionality.

Mermalady, 2016-10-12:
okeedoke! then one more question- how to I get back into command line from the GUI of touchUI? I uh, didn't write down my ip address.