BillyBlaze/OctoPrint-TouchUI #69: Disable screen timeout?

jstefanop, 2016-10-22:
Hey is there a way to disable the screen from going blank after a certain time? Most answers seem to point to configuring lightdm to disable the timeout, but since this is jessie light there is no lightdm installed.

BillyBlaze, 2016-10-24:
Hi, Please take a look at ```[disable x11 screen blanking](https://www.google.com/#safe=off&q=disable+x11+screen+blanking)```. I haven't tried it myself so I can't point a configuration that works, but it shouldn't be that hard.

BillyBlaze, 2016-11-15:
Closed due to inactivity. Please comment to re-open this issue.

stsfin, 2017-08-05:
Im going to reopen this. 

The cromium init should contain 

xset s off         # don't activate screensaver
xset -dpms         # disable DPMS (Energy Star) features.
xset s noblank     # don't blank the video device

Or atleast when installing the TouchUI autloader give a option of a option to enable these. It has really been a PITA to find the right spot with matchbox to get them to register. Since matchbox does not really care about xinitrc or anything else thats fancy.

stsfin, 2017-08-05:
So the chromium.xinit that will not blank should be changed in the end to this:

stsfin, 2017-08-05:
Or just add a text how to the guide how change the chromium.xinit to fix this to the users that need this feature.

BillyBlaze, 2017-08-13:
Hi @stsfin,

Thanks for your suggestions. I will add a configuration to the default file where you can disable or enable the screensaver functionality.

closed 2017-08-13 by BillyBlaze

BillyBlaze, 2017-08-13:
I have updated the script with a new parameter called "DISABLE_SCREENSAVER". If you [update the bootloader](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Update-your-bootloader) and set the variable ```DISABLE_SCREENSAVER``` to ```true``` inside the file /etc/default/touchui then should work now.

eboston, 2018-03-08:
It's been awhile since this has been discussed and I'm glad I found it because the screen saver was driving me nuts.  I was wondering though, if it would be possible to modify the blanking behaviour to only blank when it is not printing (i.e. Status is Printing)?