BillyBlaze/OctoPrint-TouchUI #71: Stuck at "Connecting to TouchUI" screen for awhile then "Connecting to TouchUI" failed

tanthua, 2016-11-15:
Hi BillyBlaze,

First of all, just want to say thank you. Very awesome program. I have a question regarding to TouchUI. I managed to boot it up, but it stuck on the connecting screen. It was "Connecting to TouchUI" for awhile then turn red "Connecting to TouchUI failed." I'm wondering if I need to do any additional steps to be able to pass that screen. I'm using Rpi3 and octopi.

BillyBlaze, 2016-11-15:
Hi,

Your welcome :) and thanks.

This error is thrown after 30 failed attempts to connect to the TouchUI API, this either means that TouchUI within OctoPrint is not installed, or that the TouchUI install got corrupted, or that you have an older version of TouchUI.

If you're sure that above is not the case, then please share your octoprint log, so we can check for other errors.

tanthua, 2016-11-15:
I got it working. I didn't know I still had to install it. Maybe you can add that into the instruction too, just my two cents :D.

Thanks so much!

BillyBlaze, 2016-11-26:
Awesome, I have added a line to the Guide to prevent this ;)

Your welcome

closed 2016-11-26 by BillyBlaze