BillyBlaze/OctoPrint-TouchUI #72: Flowrate and Feedrate Not Sending Commands

felixleong, 2016-11-18:
**Versions:** TouchUI 0.3.4 and Octoprint 1.2.7

**Browser Tested:** Google Chrome 54, Firefox 49 and Microsoft Edge mobile

**Problem:** The flow rate and feed rate text boxes is not sending any commands back to the printer upon confirmation. No error pop-up is shown.

**Reason:** Have bring it up using Google Chrome and it appears that it is due to the `factor` parameter is being sent as a string instead of an int.

This causes Octoprint to respond with a HTTP 400.

BillyBlaze, 2016-11-26:
Thanks for reporting, I'll look into this issue very soon and if needed a fix in the upcoming 0.3.5 release.

BillyBlaze, 2016-12-05:
I just pushed a fix into the maintenance branch and will release this in 0.3.5 very soon.

jstefanop, 2016-12-28:
Hi, any update on 0.3.5 release? I have this issue as well and can only control feedrate through computer browser.

BillyBlaze, 2016-12-29:
Lol, euhhh, yeah, 24 days ago is not 'very soon' indeed 😄 

I'll try to fix one more open issue and release 0.3.5 directly afterwards.

closed 2016-12-30 by BillyBlaze

BillyBlaze, 2016-12-30:
Thanks for your patience. I have released 0.3.5 👍 

In advance, happy new year everybody!

jstefanop, 2016-12-30:
@BillyBlaze thanks! I dont see the update being pushed to octoprint software update...is there a way to update manually?

jstefanop, 2016-12-30:
nevermind...needed to do a force update on the octoprint software update UI

BillyBlaze, 2016-12-30:
Awesome I was just about to type that 👍

commit 734569c7 (2016-12-05): :bug: Slider send number - Fixes #72